slipx06/sunsynk-power-flow-card #118: Add temperature to show aux

Legnava89, 2023-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

This feature is currently unavailable

### Expected behaviour

1.Please add a function to the sensors aux_load1 and aux_load2 to insert temperature sensors, if possible.

2. Fix a graphical problem with energy_cost_buy and energy_cost_sell in the card in the full version. see my screenshot

Thank you very much and I wish you and your family all the best.

### Possible Solutions

[image]

### Mode

full

### Context / Reason

Just an idea. This function would be used a lot

slipx06, 2023-11-04:
Your card is looking great 😄 
I'll take a look at both of those

closed 2023-11-05 by slipx06

Legnava89, 2023-11-05:
Perfect job. You could move the spot price entity a little to the left so that it does not cover the entities from the grid

slipx06, 2023-11-05:
I thought there was enough space when I tested it. Can you share a screenshot?

Legnava89, 2023-11-05:
[image]

Legnava89, 2023-11-05:
A little higher and it will be perfect :) 
Thank you very much.
[image]

slipx06, 2023-11-05:
Should be fixed in 3.9.0

commit 6bdcf4ca (2023-11-05): feat: closes #118,  fix: closes #119 & other fixes