slipx06/sunsynk-power-flow-card #119: change the inverter status readings for Goodwe inverters

paulv1999, 2023-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

The inverter_status_59 for Goodwe is implemented as follows:

- The input entity is (based on your examples) - sensor.on_grid_mode_code
- expected values - 0,1,2 or idle, importing, exporting

It tells the user whether the inverter is either exporting or importing power.

see the configuration page:
[image]

### Expected behaviour

should be consistent with Sunsynk implementation:

0, 1, 2, 3, 4 or standby, selftest, normal, alarm, fault

### Possible Solutions

The inverter_status_59 for Goodwe should be implemented as follows:
- The input entity is - sensor.work_mode_code
- expected values:

0: "Wait Mode"
1: "Normal (On-Grid)"
2: "Normal (Off-Grid)"
3: "Fault Mode"
4: "Flash Mode"
5: "Check Mode"

For more information please refer to the link below. The author has reverse engineered the UDP for Goodwe inverters EH series but according to SolarGo app for my ET series inverter the codes are same.  https://gist.github.com/fizcris/a35b1ea1a4883c2145b2ae79c933b19b#udp-protocol-reverse-engineering-for-inverter-goodwe-eh-series. 

### Mode

Both

### Context / Reason

The fact that inverter is exporting/importing power is fairly obvious from the actual values of grid sensors. The inverter status should provide more information about the actual status of the inverter itself (if possible).

Note: Great job in making this flow card happen. I have been using it for more than 2 moths and I really appreciate the effort you put in. Thank you very much !!

slipx06, 2023-11-05:
```
    goodwe: {
        standby: { states: ['0', 'Wait Mode'], color: 'blue', message: localize('common.standby') },
        normal: { states: ['1', '2', 'Normal (On-Grid)', 'Normal (Off-Grid)'], color: 'green', message: localize('common.normal') },
        fault: { states: ['3', 'Fault Mode'], color: 'red', message: localize('common.fault') },
        flash: { states: ['4', 'Flash Mode'], color: 'yellow', message: localize('common.flash') },
        check: { states: ['5', 'Check Mode'], color: 'orange', message: localize('common.check') },
    }    
```

closed 2023-11-05 by slipx06

paulv1999, 2023-11-05:
Great, thank you! It is fantastic how fast you respond even over the weekend. Respect.

mamo-nick, 2023-11-05:
Hi,

I would like to explain why `inverter_status_59: sensor.on_grid_mode_code` is more useful.

If I use `sensor.work_mode_code`, I only get the useless "Normal" information. It is state in 99,999 %
But I can get better information about grid connection using `no_grid_color:` and `grid_connected_status_194: sensor.grid_mode_code`.

With this change, users lost much more useful information about power flow.

I propose the possibility of choosing which entity will be displayed at the inverter, similar to 
```
solar:
  display_mode: 2
```

It could by
```
inverer:
  display_mode: 2
```
1 - use `work_mode_code`
2 - use `on_grid_mode_code`

Thank you for considering this option.

paulv1999, 2023-11-05:
> Hi,
>
> I would like to explain why `inverter_status_59: sensor.on_grid_mode_code` is more useful.
>
> If I use `sensor.work_mode_code`, I only get the useless "Normal" information. It is state in 99,999 [...]

Hi, just my private comment on this subject:
- some people disconnect their battery for winter altogether. In such a case, the inverter changes its state quite often between Check mode -> Standby -> Normal
-  the sensor.on_grid_mode_code provides information that the user can easily read, e.g. from grid sensors. The inverter status sensor should IMHO provide more details about the actual operating mode of the inverter itself. It is the same way as Sunsynk inverter (the original implementation) uses this codes
- your proposal makes sense to me, it depends on @slipx06 to decide how to implement it

Legnava89, 2023-11-05:
The sensor.on_grid_mode text entity can also be used for goodwe.

slipx06, 2023-11-05:
Hi. 
The `inverter_status_59` is better represented by the `work_mode_code`. My Sunsynk inverter status is `normal` 99.9% of the time. It means the system is running as expected.

The `on_grid_mode_code`  (idle, exporting or importing) state is meant to be represented by the power flow (grid_in_out) on the card (animated dot). `grid_connected_status_194` can be used to show status of the grid connection.  

Am I correct to assume:

- Idle means the grid is disconnected?
- Importing you are consuming energy from the grid?
- Exporting would assume you are sending energy back into the grid?

Legnava89, 2023-11-05:
Idle means that the inverter does not export or import. The energy comes, for example, from the battery (the grid is always connected)
Importing means that it importts energy to the grid
Exporting means that it exports energy to the grid

slipx06, 2023-11-05:
Ok. If your grid power is positive it shows that you are importing. The animated dot should reflect power flow
You can also use the grid attribute `export_colour:` to change the colour of the grid when you are exporting to make it more visible.

Legnava89, 2023-11-05:
I think a lot of people would be happy if it went back to how it was in version 3.7.0

slipx06, 2023-11-05:
Ok I have a solution that will allow both options

slipx06, 2023-11-05:
In v3.9.0 Goodwe users can now select which inverter model to use: `goodwe` or `goodwe_gridmode`. If you are using `sensor.work_mode_code` to display inverter status use `goodwe` as the model. If you are using `sensor.grid_mode_code` to display inverter status use `goodwe_gridmode` as the model.

You guys owe me a coffee 😉 I hope this meets everyone's needs now

Legnava89, 2023-11-05:
We owe you a big coffee. Thank you very much

mamo-nick, 2023-11-06:
As you know, I already bought you ☕ few days ago.   😉
Thanks - Good job.👍🏼

commit 6bdcf4ca (2023-11-05): feat: closes #118,  fix: closes #119 & other fixes