slipx06/sunsynk-power-flow-card #120: Goodwe inverter status shows values related to different inverter

paulv1999, 2023-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

In v3.8.0 the new Goodwe inverter status was introduced. However, It now returns strings based on Sunsynk codes. the code 0 returns string 'Standby' instead of 'Wait mode'.  Example:
[image]

### Steps To Reproduce

1. change to inverter model to Goodwe
2. set inverter_status_59 to sensor.work_mode_code

### Expected behaviour

the code 0 returns string 'Wait mode'

### Card Version

v3.8.0

### Home Assistant Version

11.1

### Configuration

```shell
# Loads default set of integrations. Do not remove.
default_config:

# Load frontend themes from the themes folder
frontend:
  themes: !include_dir_merge_named themes
  extra_module_url:
    - /hacsfiles/lovelace-card-mod/card-mod.js

lovelace:
  resources:
    - url: /hacsfiles/lovelace-multiple-entity-row/multiple-entity-row.js
      type: module
      
# Text to speech
tts:
  - platform: google_translate

automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml
rest: !include rest.yaml

homeassistant:
  name: Home
  unit_system: metric
  customize:
    sensor.brandt_washer_energy:
      device_class: energy
      unit_of_measurement: kWh
sensor:
  - platform: time_date
    display_options:
      - 'time'
      - 'date'
      - 'date_time'
      - 'date_time_utc'
      - 'date_time_iso'
      - 'time_date'
      - 'time_utc'
      - 'beat'
  - platform: average
    name: 'inverter power avg'
    duration: 10
    precision: 0
    entities:
      - sensor.goodwe_consumption
```

slipx06, 2023-11-05:
Is standby not equivalent to wait-mode?

paulv1999, 2023-11-05:
> Is standby not equivalent to wait-mode?

yes, it is equivalent. just that your doc says ´wait-mode' for goodwe, which is does not match the current state

slipx06, 2023-11-05:
I think the confusion there is the expected sensor input i.e. `0,1,2,3,4,5` or `Wait mode, Normal (On-Grid), Normal (Off-Grid), Fault Mode, Flash Mode, Check Mode` vs how it is displayed no the card. For simplicity I mapped the sensor states to display:

states: ['0', 'Wait Mode'] --> Standby
states: ['1', '2', 'Normal (On-Grid)', 'Normal (Off-Grid)'] --> Normal
states: ['3', 'Fault Mode'] --> Fault
states: ['4', 'Flash Mode'] --> Flash
states: ['5', 'Check Mode'] --> Check

I think this is a good representation. I'll try make the docs clearer

paulv1999, 2023-11-05:
it makes sense. thanks for creating this excellent card !!

closed 2023-11-05 by slipx06