slipx06/sunsynk-power-flow-card #121: Additional Loads add autoscale to _extra + Define units

bodawui, 2023-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

The autoscale doesn't seems to apply for essential_load1_extra and I got it all as full Wh.
[image]

Also as Additional I would like to define that for the essential_load2 I want to use kWh as units.
I do not have realtime consumption, but still would like to use dailly usage there.

### Expected behaviour

1. the _extra would be in kWh
2. the essential_load2 would be displayed in kWh

### Possible Solutions

1. apply parent load: auto_scale:true on essential_load1_extra as well
2. so something like essential_load2_unit:kWh to override default.

### Mode

lite

### Context / Reason

The total consumption is more likely to be in kWh so Wh is just a huge number.

_load2 is not actual consumption, but it is daily kWh sensor.  and displaying it as 0 and using _extra for the daily consumption is possible solution...yet kind of not great solution.

slipx06, 2023-11-06:
Hi. Auto scale is not applied to the extra sensor as this field is designed to show anything i.e temperature, humidity, and not only power or Wh. I would recommend that you create a template sensor to convert to kWh i.e. just divide by 1000 as its not something I plan to implement.

I'll see what I can do about using the essential loads to display energy values. It wont work with auto scale though.

closed 2023-11-06 by slipx06

bodawui, 2023-11-07:
I see you have deployed it to display sensor units. I had in mind to completely overwrite the units there by some custom unit. 

  load2_name: Tepelko
  load2_icon: mdi:heat-pump
  load2_unit: bullshit

and it would display like: 10 bullshit 

just to completely overwrite. with that it would still work with auto_scale.

slipx06, 2023-11-07:
You can manipulate the units through templates. 
The auto_scale function only works for power (W) as the units are also coded into the function

commit bc5e5d7d (2023-11-06): feat: closes #121, closes #122, fix: closes #123