slipx06/sunsynk-power-flow-card #122: Battery Energy as a sensor and French translation update

adnovea, 2023-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Battery / energy shall be a number. It does not currently support a sensor.

[image]

```

### Expected behaviour

Could the feature to support battery capacity sensor be added ? 
Thanks for your help.

### Possible Solutions

In addition, some French translation updates are needed to avoid text overwriting and keep consistency :
```
xxxxx DU JOUR     ---> xxxxx (JOUR)
xxxxx CE JOUR     ---> xxxxx (JOUR)
AVANT D'ATTEINDRE ---> DÉCHARGÉE À

### Mode

Both

### Context / Reason

As the number of additional features increases, the graph becomes more cluttered. 
Some wordy languages like French will require more concise wording.

Thanks for maintaining constantly this very helpful add-on.

slipx06, 2023-11-06:
Will include this in the next release. Thanks

closed 2023-11-06 by slipx06

adnovea, 2023-11-07:
THX, could you also do the same for "**shutdown_soc**" ?

I really appreciate your efforts to maintain and improve this Add-ons.
Unfortunately, I was not able to solve a big issue with the kellerza/sunsynk Add-ons (_data makes pauses every 20 to 30 mins for about 15 mins with no reason_).
I run a _Deye inverter_ with the _Solarman from Stephane Joubert_ that I had to tune (_LSE-3 Ethernet sticker and customized yaml configuration_)

slipx06, 2023-11-07:
Its already possible to use a sensor for shutdown_soc. Give it a try.

commit bc5e5d7d (2023-11-06): feat: closes #121, closes #122, fix: closes #123