slipx06/sunsynk-power-flow-card #125: hide grid

Privatecoder, 2023-11-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

cannot see an option to hide the grid section, i.e. `show_grid: false` does not exist.

### Expected behaviour

there should be an option to hide the grid section with `show_grid: false`

### Possible Solutions

add `show_grid` option to hide the grid section

### Mode

lite

### Context / Reason

for off-grid installations, showing the grid section is useless

slipx06, 2023-11-08:
Thanks for the suggestion. Will look into it

closed 2023-11-11 by slipx06

Privatecoder, 2023-11-11:
@slipx06 almost perfect, however you would also want to hide `Autarky` and `Ratio` and make `grid_power_169` optional :)

Privatecoder, 2023-11-11:
or leave `Autarky` (will be 100%) and `Ratio` (not sure how this is calculated) and make only `grid_power_169` either optional or settable to `0` (I currently just use a custom sensor which returns 0)

slipx06, 2023-11-11:
You can hide Autarky by setting it to `no`. I'll fix the logic for grid_power when show_gird: false

[image]

slipx06, 2023-11-11:
Have you tried setting `grid_power_169: none` ?

Privatecoder, 2023-11-11:
yes – works fine. should update the docs regarding this parameter, i.e. `none` is a valid value

slipx06, 2023-11-11:
Yes its optional so you should just delete. It says that in the docs

commit 50bb3cfc (2023-11-11): feat: opt to hide grid closes #125, fix: ess label