slipx06/sunsynk-power-flow-card #126: Essential load displayed wrongly

spopinski, 2023-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

[image]

It's different when opened

[image]

### Expected behaviour

Should display the same as entity

### Card Version

V3.10.0

### Home Assistant Version

2023.9.1

slipx06, 2023-11-09:
Looks like it might be an issue with the supplied sensor value. Please post a screenshot of your sensor from developer tools. It should be kWh

[image]

spopinski, 2023-11-10:
It was working for months until now

[image]

slipx06, 2023-11-10:
According to your screenshot the sensor value is 18932kWh. Which is what the card is displaying. I suspect you are capturing Wh and you need to convert to kWh. 18.9kWh

spopinski, 2023-11-10:
Yes, what puzzles me that previously it's not a problem. Ok then consider case closed. Thanks a lot for this wonderful plugin!

closed 2023-11-10 by spopinski