slipx06/sunsynk-power-flow-card #127: Add non_essential_load3 to the grid

Legnava89, 2023-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

This function is not available

### Expected behaviour

Please add non_essential_load3: to the grid. I know there is limited space on the card, but I think it would work :)
Thanks in advance for your great work and patience with annoying users like me.

For information on how load3 would look on the card, see screenshot.

### Possible Solutions

[image]

### Mode

full

### Context / Reason

The possibility of using more home appliances and having a better overview of energy.

slipx06, 2023-11-09:
Let me think about it 😄

slipx06, 2023-11-11:
This will be a tight fit if you a or using all the card option

[image]

Legnava89, 2023-11-11:
I use a different style of battery layout than yours. It works very well for me.
[image]

slipx06, 2023-11-11:
Its because you dont define this entity `switch.sunsynk_toggle_system_timer`. I also need to think about all the other users of the card. It will only work if there is a conditional check to only show the 3rd load if this is false.

Legnava89, 2023-11-11:
I understand. Unfortunately, I didn't realize that.
The only option is to move the cells to the right a little.

slipx06, 2023-11-11:
I figured out how to do this. Will push in the next release. It will only display the 3rd load if you are not using the inverter timer schedules.

[image]

Legnava89, 2023-11-11:
I don't use the inverter timer, I solve it with node red.
Thank you very much.

commit ceb3eb4a (2023-11-13): feat: add third non_essental load closes #127
closed 2023-11-13 by slipx06