slipx06/sunsynk-power-flow-card #128: Load.additional_loads = 4 not working

fruscic, 2023-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When I set Loads / additional_loads = 4 it doesn't do anything, despite the documentation indicating that it should

### Steps To Reproduce

load:
  show_daily: true
  show_aux: false
  additional_loads: 4
  load1_name: Geyser1
  load2_name: Geyser2
  load3_name: Pool_Pump
  load4_name: Borehole

### Expected behaviour

[image]

### Card Version

v3.10

### Home Assistant Version

11.1

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
large_font: true
battery:
  energy: 5320
  shutdown_soc: 20
  show_daily: true
solar:
  show_daily: true
  mppts: 2
  pv1_name: Office
  pv2_name: House
  auto_scale: false
  display_mode: 1
load:
  show_daily: true
  show_aux: false
  additional_loads: 2
  load1_name: Geyser1
  load2_name: Geyser2
  load3_name: Pool_Pump
  load4_name: Borehole
  load1_icon: boiler
  load2_icon: boiler
  load3_icon: mdi:pool
  load4_icon: pump
grid:
  show_daily_buy: true
  show_daily_sell: false
  show_nonessential: false
entities:
  remaining_solar: sensor.solcast_forecast_remaining_today
  inverter_voltage_154: sensor.sunsynk_output_voltage_154
  load_frequency_192: sensor.sunsynk_output_frequency_192
  inverter_current_164: sensor.sunsynk_output_current_164
  inverter_power_175: sensor.sunsynk_output_total_power_175
  grid_power_169: sensor.sunsynk_grid_power_169
  essential_load1: sensor.geyser_1_200l_power
  essential_load2: sensor.geyser_2_150l_power
  essential_load3: sensor.pool_pump_power
  essential_load4: sensor.borehole_pump_power
  essential_load1_extra: sensor.geyser_1_200l_voltage
  essential_load2_extra: sensor.geyser_2_150l_voltage
  day_grid_import_76: sensor.sunsynk_day_grid_import
  day_grid_export_77: sensor.sunsynk_day_grid_export
  use_timer_248: sensor.current_use_timer_248
  day_pv_energy_108: sensor.sunsynk_daily_pv_108
  priority_load_243: sensor.energy_mode_243
  grid_ct_power_172: sensor.sunsynk_grid_ct_power
  day_load_energy_84: sensor.sunsynk_daily_load_84
  inverter_grid_voltage_154: sensor.sunsynk_output_voltage_154
  inverter_load_freq_192: sensor.sunsynk_output_frequency_192
  inverter_out_164: sensor.sunsynk_output_current_164
  inverter_out_175: sensor.sunsynk_output_current_164
  grid_status_194: sensor.sunsynk_grid_online_194
  inverter_status_59: sensor.sunsynk_inverter_status_59
  day_battery_charge_70: sensor.sunsynk_battery_daily_charge_70
  day_battery_discharge_71: sensor.sunsynk_battery_daily_discharge_71
  battery_voltage_183: sensor.sunsynk_battery_voltage_183
  battery_power_190: sensor.sunsynk_battery_power_190
  battery_soc_184: sensor.sunsynk_battery_soc_184
  battery_out_190: sensor.sunsynk_battery_power_190
  battery_current_191: sensor.sunsynk_battery_current_191
  inverter_load_grid_169: sensor.sunsynk_grid_power_169
  grid_buy_day_76: sensor.sunsynk_daily_grid_buy_76
  grid_sell_day_77: sensor.sunsynk_daily_grid_sell_77
  grid_external_power_172: sensor.sunsynk_grid_power_172
  loadday_84: sensor.sunsynk_daily_load_84
  essential_power: sensor.sunsynk_load_power_ess
  nonessential_power: sensor.sunsynk_grid_power_169
  aux_power_166: sensor.sunsynk_output_aux_power_166
  solarday_108: sensor.sunsynk_daily_pv_108
  pv1_power_186: sensor.sunsynk_pv1_186
  pv2_power_187: sensor.sunsynk_pv2_187
  pv3_power_188: none
  pv4_power_189: none
  pv1_voltage_109: sensor.sunsynk_pv1_voltage_109
  pv1_current_110: sensor.sunsynk_pv1_current_110
  pv2_voltage_111: sensor.sunsynk_pv2_voltage_111
  pv2_current_112: sensor.sunsynk_pv2_current_112
```

slipx06, 2023-11-09:
Additional loads 3 and 4 are only displayed in the lite version of the card due to limited space. I mentioned it in the release notes but did not make it clear in the documentation.  Will update it.

fruscic, 2023-11-10:
Ah, ok

Thanks so much for the prompt reply. Works as you say wth the 'lite'
version which is perfect for me

As an aside, have you considered allowing for logic to change the colours?
Eg depending on the level of the battery, or usage of the inverter?

I appreciate the work on the integration - I am new to home assistant, and
I'm finding this very useful

closed 2023-11-10 by fruscic