slipx06/sunsynk-power-flow-card #130: Historical Base64 image

brucehat, 2023-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Current historical base64 image is in the javascript application, could it please be configurable as an entity or the PNG put in the www folder so it can be swapped to images of other brands like:
[image]

### Expected behaviour

Image based on model and modern

### Possible Solutions

1. Use model & modern: false combination to present image for other inverters
2. provide a script include file for base64 image with true / false driver
3. externalise png to folder on HA so it can be replaced but not overwritten during updates.

### Mode

full

### Context / Reason

Just to make your card more cool - it's part of a lot of our lives!

slipx06, 2023-11-11:
I can change the modern attribute to `image:` with the options of; `modern`, `sunysnk`, `lux`, `solis` and  `goodwe`. If you can share with me the images/svg's ?

Legnava89, 2023-11-11:
Image for goodwe
[image]

brucehat, 2023-11-11:
Hi,Here is the image for LuxPower Would you prefer base64?ThanksBruce

brucehat, 2023-11-11:
Hi,

Here is Lux, not sure what is better or do you want bas64?

[image]

Thanks
Bruce

brucehat, 2023-11-11:
Here is Solis
[image]

brucehat, 2023-11-11:
Sorry I focused on quality not size --> tell me what you need and I will happily convert / resize etc.

Pho3niX90, 2023-11-11:
@slipx06 I did this on my fork. The base64 is moved to constants, and then selected based on the model. I haven't created a PR as yet as I am trying to do some refactoring.

CaptainMidnight, 2023-11-11:
> I can change the modern attribute to `image:` with the options of; `modern`, `sunysnk`, `lux`, `solis` and `goodwe`. If you can share with me the images/svg's ?

Hi @slipx06 I'd love it if you could make that change - I've just been looking at the code myself (although I'm not a coder more hacker/scripter) and created my own base64 code string from a Lux inverter image that I resized to 72x54 which I think is your modern image size, replaced it in your code and uploaded to HA just as a way of learning just a little of how this stuff works.....

Anyway, your suggestion would be fantastic as you can see with now supporting 4x different inverters, there's now a much bigger potential user base for your card.

I did look a changing a other card to try and integrate some of your functionality but why reinvent the wheel......

slipx06, 2023-11-11:
> @slipx06 I did this on my fork. The base64 is moved to constants, and then selected based on the model. I haven't created a PR as yet as I am trying to do some refactoring.

Thanks. I'll wait for you to finish before I do anything

slipx06, 2023-11-11:
> Hi @slipx06 I'd love it if you could make that change - I've just been looking at the code myself (although I'm not a coder more hacker/scripter) and created my own base64 code string from a Lux [...]

Great. Can you share your encoded image

CaptainMidnight, 2023-11-11:
Absolutely here it is, on this image I've reduced the icon to 52x72 as 54x72 was a bit too close to the inverter temps.

[image]

Image code:
`
iVBORw0KGgoAAAANSUhEUgAAADQAAABICAYAAAC5mNZRAAABb2lDQ1BpY2MAACiRdZG9S8NAGMaftkrFVoroIOIQoYpDC0VBHLUOXYqUWsGqS3JNWiFJwyVFiqvg4lBwEF38GvwPdBVcFQRBEUTc3P1apMT3mkKLtHdc3h9P7nm5ew7wp3Vm2D0JwDAdnk0lpdX8mhR8R4jmIMYRkZltLWQyaXQdP4/wifoQF7267+s4QgXVZoCvj3iWWdwhnidObzmW4D3iYVaSC8QnxDFOByS+Fbri8Zvgosdfgnkuuwj4RU+p2MZKG7MSN4iniKOGXmHN84ibhFVzZZnqKK0x2MgihSQkKKhgEzocxKmalFlnX6LhW0KZPIy+Fqrg5CiiRN4YqRXqqlLVSFdp6qiK3P/naWsz0173cBLofXXdzwkguA/Ua677e+q69TMg8AJcmy1/mXKa+ya91tKix0BkB7i8aWnKAXC1C4w8WzKXG1KAll/TgI8LYCAPDN0D/eteVs3/OH8Cctv0RHfA4REwSfsjG3/MEWfyDgOuPwAAAAlwSFlzAAALEwAACxMBAJqcGAAABXBJREFUaEPdW11u20YQ/kiuJMuynRgoYCBogz4ELeIaRU5QFO27z9ID+MVnaI7RHKJHcIAWaAyrbpACLprKon9l8a8zS1OiKFLalZakorUNG9TuzvfNzM4Od8fW69c//2RZiPDJtASqlUFswaZmuW4vol/FdDIfFU2XN8GMWZdQXzLr9OyOIyB6vR5sx9YWUA7YNAx9CZJQGIUAfee2KLaHRT6ZbVkfjWb0nRyrClR/Fdh2CMHT54mw6Gmn2QYDvfEH1IsE0HeWHH/Oz4QQEncYhvQTP6ujxSgYLgFrNZsyOjQIXOgFePuxC9uy8c1nX8JqOogILBuC1p4c4/u+dNcgCNB3r+A4DlqNRh08RjJHhBrNBn7/4x3arQ1cu9f4p3GLbquPZiTw4eIC3vsr+E4EQX7KQaRBwIMgxP39PXa2t6SFzv96j/2XX+PVtwfwPF8qqepGhNg1bAhb4N+PPdzd3sLtX+P+eQMbz3fgDCN0P/yN7V4IzwkRRAGGD0M8DIZ4+vQJXLLMF58/ww/ff4evXryQxJhM7HZV0yE23e4phW1b+vwDAe27LjbJSk6niV8v3sKObPz47BU2Gk381+/BJ6s82dmBNxzCvbqGIxxsdzpot9toktV8cj/P82pZQ0I0xoRYl0yM10FEXxZFvpbF6yHCQ+QjoFBoW47UOruSzX88miAkEkykupZvevYOkXaLOEKNY7gHfwIjfVoiZjP+KXjdVuvrZoAXaXYU5cyqvlzQhVhJrAahGkAuIDKH0AKzmDCvIbGP+5Ch2VSIlSxKw+Vy0JYMLl8/+ULjnJRySuUItyLg58GYDtvzRqi4lVafYo1rTfPYOXY5oyTMAtQlJZYjUy/4PLJC/XhkGrxRw+qaoqA/WWj8zrqKAHV5CotzOd1RtfcvQkxhe7XZ6Ktahu3qmj5AXWzqG6vWzOUDH8FJi+JsezkL1QR8hnJzLFQRyJLEkIXGYVvLq1Q7lwR8WnzMw9waqhR4sTaLM4XKAJpNJkXZHjepy/K1RC5XhpAy5sxfNemnLJVcjndWFQAqfVQjRXG/ZaQwFTGZKCwznTqZsqSws9H9ULlhe1Hw8b1TVknjB5MHpMlzTk412qLgNETIc3M+o+Zz9llZDK99vpdKH13LsXwwn9eqAB/LnZTE905MiIHOi1etVkveTyW3iPImkV2ufPDzJbA1tra20O2e4ejoCCcnJ9jd3Z2wQFrxLl37HB4e4vj4WD7mKxw2jrxj1W+LjZolRy5oumu6vLzE2dkZLujWkH9mtdN3p+BLb74CSqySiXLFi1CftMaI8ZqGRdedqnuj7Jvxr1RQMK/1OhbnI6EFySw4TMN2il2TrUcnbK8M+Nkcp/ehyoDnCVp+k49PfYyQMDKJoosVd9PKFMbTrAb4PFpz9qHVBV5kI9qHjPnc0u6y6ATxbhR/6RfKLSq1lHHTHrTgGjKHLg2J/+aKL86iVRr3yxZIGSVUvOLU1qLvB1QQtYuDgwNZ8bW52Sms6Lq7u8P+/kvqR6U8qfcM68/z80gmd5otH6Ia8FmiGIvnDWX2XISLX3gCqtXjgqn0yw/3V7JQWeCzxDj9D8IALaoGa7c32QHHm2QOCHa5bOVXqkRzee1qGTlHXJI5e76nPFU2My+/8KJCPck3VmVVzOtYIfCijIXP6emwnjy36DRCB6Sxi7MlhNLbq3VzcxtHOZ155llL+XOzQmWUe/PmlworAM0SyOqNzySsvb29SPUdXlnxNXSUdbB0lic6VNGbFJbXgMOYyFGFP/vdOliINcM8PvFse9rAa0Mo2XrWhlBiq7UhlMSBtSG0ti4nk9Pkv7OMbQo1TJRYiAn9NhgMgjr+eck0bzJM83+TYoOvfV17wgAAAABJRU5ErkJggg==
`

CaptainMidnight, 2023-11-11:
Your adjusted sunsynk-power-flow-card.js.gz

[sunsynk-power-flow-card.js.gz](https://github.com/slipx06/sunsynk-power-flow-card/files/13326160/sunsynk-power-flow-card.js.gz)

closed 2023-11-12 by slipx06

Legnava89, 2023-11-12:
Beauty, beauty ... slipx06 is a god.
[image]

brucehat, 2023-11-12:
Lux also good - awesome work @slipx06 [image]

CaptainMidnight, 2023-11-12:
What a great update, which I'm sure will make many users very happy, thank you!

slipx06, 2023-11-12:
@Pho3niX90  did all the heavy lifting for this change. He deserves all of the credit. 👏

PR #131 (2023-11-12): Feature/refactoring
- There are some code refactoring happening, to try and reduce duplication, redundant checks. The goal is to reduce the overall size and line count.
- Fixed some status code from the solis
- Change common strings to interfaces and enums for type safety
- Lint warning fixes
- Move dev packages to dev, to reduce bundle size

## Description

## Related issues
Resolves #130 

## Motivation and Context
The bulk of the PR is related to reducing complexity, and reducing the dom size.

## How has this been tested
I have been running this for the last couple days. 

## Type of chang [... cut]