slipx06/sunsynk-power-flow-card #133: Unavailable entity

mamo-nick, 2023-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

After click  I get error message

[image]

### Expected behaviour

... may be disabled?

### Card Version

3.12

### Home Assistant Version

2023.11

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: lite
panel_mode: false
large_font: true
title: Aktuální výroba - spotřeba
title_colour: grey
title_size: 24px
show_solar: true
show_battery: true
inverter:
  modern: true
  colour: grey
  autarky: energy
  three_phase: true
  model: goodwe_gridmode
battery:
  energy: 10650
  shutdown_soc: 20
  invert_power: false
  colour: var(--energy-battery-in-color)
  show_daily: true
  max_power: 5400
  full_capacity: 100
  empty_capacity: 20
  show_absolute: false
  auto_scale: true
solar:
  show_daily: true
  display_mode: 2
  mppts: 2
  animation_speed: 9
  max_power: 5400
  pv1_name: VÝCHOD
  pv2_name: ZÁPAD
load:
  show_daily: true
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: true
  nonessential_name: Doma
  invert_grid: false
  export_colour: green
entities:
  use_timer_248: 'no'
  priority_load_243: 'no'
  load_frequency_192: sensor.meter_frequency
  inverter_voltage_154: sensor.on_grid_l1_voltage
  inverter_voltage_L2: sensor.on_grid_l2_voltage
  inverter_voltage_L3: sensor.on_grid_l3_voltage
  inverter_current_164: sensor.on_grid_l1_current
  inverter_current_L2: sensor.on_grid_l2_current
  inverter_current_L3: sensor.on_grid_l3_current
  inverter_power_175: zero
  grid_connected_status_194: sensor.grid_mode_code
  day_battery_charge_70: sensor.today_battery_charge
  day_battery_discharge_71: sensor.today_battery_discharge
  battery_voltage_183: sensor.battery_voltage
  battery_soc_184: sensor.battery_state_of_charge
  battery_power_190: sensor.battery_power
  battery_current_191: sensor.battery_current
  grid_power_169: sensor.house_consumption
  day_grid_import_76: sensor.energy_buy_daily
  day_grid_export_77: sensor.energy_sell_daily
  grid_ct_power_172: sensor.active_power_l1
  grid_ct_power_L2: sensor.active_power_l2
  grid_ct_power_L3: sensor.active_power_l3
  day_load_energy_84: sensor.today_load
  essential_power: none
  nonessential_power: none
  aux_power_166: null
  day_pv_energy_108: sensor.today_s_pv_generation
  pv1_power_186: sensor.pv2_power
  pv2_power_187: sensor.pv1_power
  pv1_voltage_109: sensor.pv2_voltage
  pv1_current_110: sensor.pv2_current
  pv2_voltage_111: sensor.pv1_voltage
  pv2_current_112: sensor.pv1_current
  battery_temp_182: sensor.battery_temperature
  radiator_temp_91: sensor.inverter_temperature_radiator
  dc_transformer_temp_90: sensor.inverter_temperature_air
  energy_cost_buy: sensor.current_spot_electricity_price
  energy_cost_sell: sensor.current_spot_electricity_price
  pv_total: sensor.pv_power
  inverter_status_59: sensor.on_grid_mode_code
  remaining_solar: sensor.energy_production_today_total
  load_power_L1: sensor.load_l1
  load_power_L2: sensor.load_l2
  load_power_L3: sensor.load_l3
  battery_status: sensor.battery_mode_code
  total_pv_generation: sensor.total_pv_generation
```

slipx06, 2023-11-12:
That entity is calculated in one of two ways. You can either provide a sensor using the `essential_power:` or the card calculates this based on the formula `essential = inverter_power_175 + grid_power_169 - aux_power_166`. If you do not provide a sensor then there will be no popup or historical state information. Since you have set `essential_power: none` you are using the cards internal logic. You should supply your own sensor as you are using 3 phase power and to enable the more-info popup i.e. `essential_power: sensor.total_essential_load`

[image]

mamo-nick, 2023-11-12:
Yes, that's clear, but I'm getting an error popup. I would expect similar behavior to `pv_total`, where the entity is unclickable if calculated.

slipx06, 2023-11-12:
OK thanks. I found the problem. For now just delete the entry essential_power: none

mamo-nick, 2023-11-12:
Yep! That's OK. Thanks.

slipx06, 2023-11-13:
Fixed in 3.13.2

closed 2023-11-13 by slipx06