slipx06/sunsynk-power-flow-card #134: Battery Time to reach % is incorrect

adnovea, 2023-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

The computed charge time is incorrect.
Since a couple of releases I have notice something strange in the battery time computation.
I thought I did broke something but it does not seems so.
[image]

### Steps To Reproduce

Charge the battery 

### Expected behaviour

The time should be 50 mins
(80% - 69%) x 3500 kWh / 464 W from batt.

### Card Version

3.12.0

### Home Assistant Version

2023.11.2

slipx06, 2023-11-12:
Please test 3.13.0

adnovea, 2023-11-12:
Sorry, no change
[image]

slipx06, 2023-11-12:
I cant reproduce this. Please share your card config

adnovea, 2023-11-12:
Hi Slipx and THX for your efficient support.

Here is my config. I use a Deye Sun 5K with a fork of Solarman to support LSE-3 Stick Logger

```
type: custom:sunsynk-power-flow-card
cardstyle: full
lang: fr
panel_mode: false
card_width: 50%
card_height: 485px
large_font: false
title: Système photovoltaïque
title_colour: '#CCCCCC'
title_size: 19px
show_solar: true
show_battery: true
decimal_places: 2
inverter:
  modern: true
  colour: lightgrey
  autarky: power
  model: sunsynk
  auto_scale: false
battery:
  energy: sensor.deye_battery_capacity
  shutdown_soc: sensor.deye_battery_shutdown
  invert_power: false
  colour: '#f06292'
  show_daily: true
  animation_speed: 6
  max_power: 2000
  full_capacity: 100
  empty_capacity: 20
  show_absolute: false
  auto_scale: true
  three_phase: false
solar:
  colour: orange
  show_daily: true
  mppts: 2
  animation_speed: 9
  max_power: 4920
  pv1_name: Ouest
  pv2_name: Est
  auto_scale: false
  display_mode: 2
load:
  colour: '#5fb6ad'
  show_daily: true
  show_daily_aux: true
  show_aux: true
  invert_aux: false
  animation_speed: 8
  max_power: 5000
  aux_name: Surplus (V.E.)
  show_absolute_aux: true
  aux_type: default
  aux_colour: lightblue
  aux_off_colour: lightgrey
  aux_loads: 0
  aux_load1_name: Véhicule
  aux_load1_icon: mdi:car-electric-outline
  aux_load2_name: Piscine
  aux_load2_icon: mdi:pool
  additional_loads: 2
  load1_name: E.C.S.
  load1_icon: mdi:water-boiler
  load2_name: Autres
  load2_icon: mdi:power-plug-battery-outline
  auto_scale: false
grid:
  colour: '#488fc2'
  no_grid_colour: rgb(192,0,0)
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: true
  nonessential_icon: default
  nonessential_name: Non essentielles
  essential_name: Essentielles
  additional_loads: 2
  load1_name: P.A.C.
  load1_icon: mdi:heat-pump-outline
  load2_name: Autres
  load2_icon: mdi:washing-machine
  invert_grid: false
  animation_speed: 8
  max_power: 6000
  auto_scale: false
  energy_cost_decimals: 0
entities:
  use_timer_248: switch.deye_time_of_use_switch
  priority_load_243: switch.deye_priority_mode_switch
  inverter_voltage_154: sensor.sensor.deye_grid_voltage_l1
  load_frequency_192: sensor.deye_load_frequency
  inverter_current_164: sensor.deye_current_l1
  inverter_power_175: sensor.deye_inverter_l1_power
  grid_connected_status_194: sensor.deye_grid_connected_status
  inverter_status_59: sensor.deye_running_status
  day_battery_charge_70: sensor.deye_daily_battery_charge
  day_battery_discharge_71: sensor.deye_daily_battery_discharge
  battery_voltage_183: sensor.deye_battery_voltage
  battery_soc_184: sensor.deye_battery_soc
  battery_power_190: sensor.deye_battery_power
  battery_current_191: sensor.deye_battery_current
  grid_power_169: sensor.deye_internal_ct_l1_power
  day_grid_import_76: sensor.deye_daily_energy_bought
  day_grid_export_77: sensor.deye_daily_energy_sold
  grid_ct_power_172: sensor.deye_external_ct_l1_power
  day_load_energy_84: sensor.deye_daily_load_consumption
  essential_power: sensor.deye_load_l1_backup_power
  essential_load1: sensor.ecs_backup_power
  essential_load2: sensor.deye_nonessential_others
  nonessential_power: sensor.deye_load_l1_home_power
  non_essential_load1: sensor.aquarea_energy_consumption
  non_essential_load2: sensor.deye_essential_others
  aux_power_166: sensor.deye_gen_power
  aux_load1: none
  aux_load2: none
  aux_connected_status: sensor.deye_aux_connected_status
  day_aux_energy: none
  day_pv_energy_108: sensor.deye_daily_production
  pv1_power_186: sensor.deye_pv1_power
  pv2_power_187: sensor.deye_pv2_power
  pv3_power_188: none
  pv4_power_189: none
  pv_total: null
  pv1_voltage_109: sensor.deye_pv1_voltage
  pv1_current_110: sensor.deye_pv1_current
  pv2_voltage_111: sensor.deye_pv2_voltage
  pv2_current_112: sensor.deye_pv2_current
  pv3_voltage_113: none
  pv3_current_114: none
  pv4_voltage_115: none
  pv4_current_116: none
  remaining_solar: sensor.energy_production_today_remaining
  battery_temp_182: sensor.deye_battery_temperature
  radiator_temp_91: sensor.deye_ac_temperature
  dc_transformer_temp_90: sensor.deye_dc_temperature
  prog1_time: sensor.deye_mode_point_1
  prog1_capacity: sensor.deye_mode_soc_point_1
  prog1_charge: sensor.deye_mode_grid_point_1
  prog2_time: sensor.deye_mode_point_2
  prog2_capacity: sensor.deye_mode_soc_point_2
  prog2_charge: sensor.deye_mode_grid_point_2
  prog3_time: sensor.deye_mode_point_3
  prog3_capacity: sensor.deye_mode_soc_point_3
  prog3_charge: sensor.deye_mode_grid_point_3
  prog4_time: sensor.deye_mode_point_4
  prog4_capacity: sensor.deye_mode_soc_point_4
  prog4_charge: sensor.deye_mode_grid_point_4
  prog5_time: sensor.deye_mode_point_5
  prog5_capacity: sensor.deye_mode_soc_point_5
  prog5_charge: sensor.deye_mode_grid_point_5
  prog6_time: sensor.deye_mode_point_6
  prog6_capacity: sensor.deye_mode_soc_point_6
  prog6_charge: select.deye_mode_grid_point_6
  energy_cost_buy: sensor.linky_papp
  energy_cost_sell: none
  solar_sell_247: switch.deye_solar_sell_switch

```

slipx06, 2023-11-12:
Can you please test what happens when you substitute the sensor values with numbers i.e. 

```
  energy: sensor.deye_battery_capacity
  shutdown_soc: sensor.deye_battery_shutdown
```

e.g.

```
  energy: 15000
  shutdown_soc: 10
```

adnovea, 2023-11-12:
THX I edited my post because I confused 2 sensors.

slipx06, 2023-11-12:
Glad its fixed

closed 2023-11-12 by slipx06