slipx06/sunsynk-power-flow-card #135: Native Huawei Support

Roving-Ronin, 2023-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

No native Huawei support

### Expected behaviour

Native Huawei support would be nice.

### Possible Solutions

If you could give some pointers as to what you require to add support for Huawei, I would be able to look at it to add support to determine the sensors that could be utilised from the Huawei Solar Integration by WLCRS ( https://github.com/wlcrs/huawei_solar ).

### Mode

Both

### Context / Reason

Provide added support for Huawei users.

Pho3niX90, 2023-11-13:
> ### Is there an existing issue for this?
> * [x]  I have searched the existing issues
>
> ### Current Behavior
> No native Huawei support
>
> ### Expected behaviour
> Native Huawei support would be nice.
>
> ### [...]

What features are you missing, or hoping to see for huawei?

slipx06, 2023-11-18:
Hi. That integration will give you most of the required sensors for the card to work. If you can share your card configuration I can add it as an example. If there are specific Huawei Inverter status codes and/or messages please share them.

Roving-Ronin, 2023-11-19:
Hi,  apologies on slow reply, just busy as atm. Yes I've been working on 'translating' WLCRS entities into the card, however there's some items on the card that don't map to WLCRS / Huawei inverters and also wondering if/how we can add a picture of the Huawei inverters into the card, same as like GoodWe etc have been added?

If I can map out the corresponding entities for Huawei in the card, is there somewhere to publish them or would this be added into the code so if a user has something like this in the yaml, it would show the Huawei inverter image ?

```
inverter:
  modern: false
  colour: grey
  autarky: energy
  model: huawei_gridmode
  three_phase: true
```

Also wondering if anyone has looked at supporting multiple inverters in the card? May not be space for 2 inverter images, but more instead of 4 MPTT on 1 inverter in the card, if you could have 2 MPTT per inverter (maybe option to different colour per inverter data) and additional entries for 2nd inverter internal temperature / status / autarky / inverter Amps and Active Power etc.

slipx06, 2023-11-19:
Hi. If you share an image of the inverter I can add together with the model tag. What is the significance of the appended "gridmode". I would have assumed just `model: huawei` . You will also need to share the inverter_status messages for mapping. 

I don't think support for multiple inverters will be possible without a complete redesign. Maybe there is a way for you to have two separate cards to depict this.

Roving-Ronin, 2023-11-20:
Hi @slipx06, th

The 'gridmode' was just from mymicing the GoodWe inverter config that has that, ignore/delete it in regards Huawei. For the residential sizes Huawei inverters, the current range looks like:

[image]

or if you need a dark background:
[image]

Will revert re the mappings, need to get a chance to do some more testing.

kurtdefeyter, 2023-11-24:
Very interested in this ,

my setup
SUN2000-8KTL-M1
luna 2000 15 kw
22 optimizers

Please contact me and let me know how i can help

Roving-Ronin, 2023-11-24:
@kurtdefeyter 
Can you send me your LUNA ESS, to allow testing ? I'll send you a pre-paid mailbag ;-)

a) Do you have WLCRS integration for Huawei installed ? If not see: https://github.com/wlcrs/huawei_solar

b) If/once you have WLCRS installed, could you lookup ALL the battery entities that is creates ? Open up the entity and copy the raw e.g. 'sensor.battery_state_of_capacity_1' (or does it not add the _1 if there's only 1 ESS?) and all the other sensor for the LUNA ESS ? Below is the entities its supposed to have... but there might be more that are not enabled in HA by default (handy to list them also).

[image]

c) I assume you have a single inverter, could you also confirm the entity naming structure WLCRS uses by default, as I have a couple of inverters and have changed the format of mine to make more sense.

iirc I think your inverter Input for the panels would have:   sensor.inverter_pv_1_voltage  or is it sensor.inverter_pv_voltage_1 ?

kurtdefeyter, 2023-11-24:
the inventor looks like this

[image]

The input sensors 

[image]
[image]

Roving-Ronin, 2023-11-24:
@slipx06 is this to type of information that you require, to allow the card to understand the Huawei status / state and grid status codes? 

https://github.com/wlcrs/huawei-solar-lib/blob/master/src/huawei_solar/register_values.py

i.e. sensor.inverter_device_status appears to have any of these for status message:

Name:	Device Status
Entity: sensor.inverter1_device_status

"Standby: initializing",
"Standby: detecting insulation resistance",
"Standby: detecting irradiation",
"Standby: grid detecting",
"Starting",
"On-grid",
"Grid Connection: power limited",
"Grid Connection: self-derating",
"Off-grid mode: running",
"Shutdown: fault",
"Shutdown: command",
"Shutdown: OVGR",
"Shutdown: communication disconnected",
"Shutdown: power limited",
"Shutdown: manual startup required",
"Shutdown: DC switches disconnected",
"Shutdown: rapid cutoff",
"Shutdown: input underpowered",
"Grid scheduling: cosphi-P curve",
"Grid scheduling: Q-U curve",
"Grid scheduling: PF-U curve",
"Grid scheduling: dry contact",
"Grid scheduling: Q-P curve",
"Spot-check ready",
"Spot-checking",
"Inspecting",
"AFCI self check",
"I-V scanning",
"DC input detection",
"Running: off-grid charging",
"Standby: no irradiation",

=============================

Name: 	Inverter State
Entity:	sensor.inverter1_inverter_state

0b0000_0000_0000_0001: "Standby",
0b0000_0000_0000_0010: "Grid-Connected",
0b0000_0000_0000_0100: "Grid-Connected normally",
0b0000_0000_0000_1000: "Grid connection with derating due to power rationing",
0b0000_0000_0001_0000: "Grid connection with derating due to internalcauses of the solar inverter",
0b0000_0000_0010_0000: "Normal stop",
0b0000_0000_0100_0000: "Stop due to faults",
0b0000_0000_1000_0000: "Stop due to power rationing",
0b0000_0001_0000_0000: "Shutdown",
0b0000_0010_0000_0000: "Spot check",

Is what is in the py code, but looking at my inverter its only had 'Standby' (currently as its 3am here) and during the day 'Grid-Connected, Grid-Connected normally'.  Based upon that and the grouping of the options above by the 0000_0000 pattern in front it looks like the options are:

"Standby",
"Grid-Connected",
"Grid-Connected, Grid-Connected normally",
"Grid-Connected, Grid connection with derating due to power rationing",
"Grid connection with derating due to internalcauses of the solar inverter",
"Normal stop",
"Stop due to faults",
"Stop due to power rationing",
"Shutdown",
"Spot check",

=============================

Name:	Battery Status
Entity: sensor.battery_status

Options: 

OFFLINE
STANDBY
RUNNING
FAULT
SLEEP_MODE

=============================

Not sure about this, if I could find out the sensor entiry it is published as AND if it equates to being able to use this as the current working mode (flow direction) of the battery.

class StorageForcibleChargeDischarge(IntEnum):
    """Storage Product Model."""

Options:
    STOP
    CHARGE 
    DISCHARGE

kurtdefeyter, 2023-11-24:
We do have a dynamic contract , so the price we pay is based on the Belpex 
the calculation is this part of Belgium is actually
{% set s = { "bijz_accijns": 0.014416, "energiebijdrage": 0.0020417, "groenestroom_wkk": 0.02184, "kwh_tarief": 0.0374193, "VAT": 1.06, "marge": 0.002040 } %} {{((current_price + s.marge) * s.VAT + s.bijz_accijns + s.energiebijdrage + s.groenestroom_wkk + s.kwh_tarief) | float}}

the daily result looks like 

[image]

Yes we do get paid for the injection , this is also based on the Belpex price 

https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show?name=&defaultValue=true&viewType=GRAPH&areaType=BZN&atch=false&dateTime.dateTime=25.11.2023+00:00|CET|DAY&biddingZone.values=CTY|10YBE----------2!BZN|10YBE----------2&resolution.values=PT15M&resolution.values=PT30M&resolution.values=PT60M&dateTime.timezone=CET_CEST&dateTime.timezone_input=CET+(UTC+1)+/+CEST+(UTC+2)

Roving-Ronin, 2023-11-24:
Umm yeah ok, talk about overly complicated. Has someone already released an integration for HA, that provides you with an entity of the current price to buy and sell power ?

No idea who the Belgium power retailers are but UK example iirc is Octopus, in Australia there is Amber ... look them up in HACS and you'll see they have e an integration that allows pulling 5min update prices as they resell wholesale market prices.

kurtdefeyter, 2023-11-24:
sure this is the calculation template based on the entsoe intagration however i have many costumers that are using it with nordpool to .

We post our templates on https://github.com/JaccoR/hass-entso-e/discussions/categories/price-modifyer-templates

Roving-Ronin, 2023-11-25:
@slipx06 @kurtdefeyter 

OK I think I've worked out most of the mappings / Huawei equivalents, I've just got to work out the best way of defining the extra sensors that WLCRS doesn't create by default (i.e. the Daily Export / Import totals) and try and ensure they align with other addons/integrations.. oh and also the p/kWh import costs / export sell price (I use a sensor that tracks day and time and switches automatically as per our Time of Use plan in Australia, but Kurt provides an example of way complex plans that EU has.... thinking it might just be a case of BYO for that sensor).

Here's my current setup, ignore the reference to IoTaWatt, its just providing some of the sensors (lights, HVAC, GPOs etc) whilst the servers / network gear is provided by Athom ESP plugs.

**Exporting KWh back to the Grid** - correct export price also shows.
[image]

**Importing kWh from the Grid** - correct import price also shows.
[image]

Only thing missing at the moment, apart from the battery specific sensors that @kurtdefeyter will need to test (since I don't have a battery) is the inverter status information, that currently the card doesn't understand hence no green/red next to the inverter status and the Grid icon showing as Off Grid.

Here's the yaml as it stands currently (but my servers etc sensors reverted back to the geyser sensor etc). 

```
type: custom:sunsynk-power-flow-card
cardstyle: full
panel_mode: false
large_font: false
title: Huawei Power Flows
title_colour: White
title_size: 18px
show_solar: true
show_battery: true
decimal_places: 3
inverter:
  modern: true
  colour: grey
  autarky: power
  auto_scale: true
  model: huawei
  three_phase: true
battery:
  energy: 15000
  shutdown_soc: 20
  invert_power: false
  colour: pink
  show_daily: true
  animation_speed: 6
  max_power: 5000
  full_capacity: 80
  empty_capacity: 10
  show_absolute: false
  auto_scale: true
solar:
  colour: orange
  show_daily: true
  mppts: 2
  animation_speed: 9
  max_power: 13200
  pv1_name: North
  pv2_name: East
  pv3_name: West
  pv4_name: South
  auto_scale: true
  display_mode: 2
load:
  colour: '#5fb6ad'
  show_daily: true
  show_daily_aux: true
  show_aux: true
  invert_aux: false
  show_absolute_aux: false
  aux_name: Generator
  aux_type: gen
  aux_colour: green
  aux_off_colour: red
  aux_loads: 2
[... 110 lines cut ...]
  prog4_capacity: number.sunsynk_system_mode_soc_time4
  prog4_charge: switch.sunsynk_system_mode_grid_charge_time4
  prog5_time: sensor.sunsynk_time_slot_5
  prog5_capacity: number.sunsynk_system_mode_soc_time5
  prog5_charge: switch.sunsynk_system_mode_grid_charge_time5
  prog6_time: sensor.sunsynk_time_slot_6
  prog6_capacity: number.sunsynk_system_mode_soc_time6
  prog6_charge: switch.sunsynk_system_mode_grid_charge_time6
  energy_cost_buy: sensor.electricity_price
  energy_cost_sell: input_number.electricity_export_rate
  solar_sell_247: switch.sunsynk_toggle_solar_sell

```
@kurtdefeyter could you check the sensors here and 'convert' the sensors that I have based upon multiple inverters, into your single inverter sensor names? I've tried to enter them as I think a single inverter sensor naming format is by default, but they need checking. For example here's what my config has, 2 x 6kW M1 inverters, each with MPTT #1 having a single 19 panel string attached (nothing on MPTT @2), hence my config I use pv1 to represent the primary inverter string and pv2 to represent the secondary inverter string.

  pv1_power_186: sensor.inverter1_input_power
  pv2_power_187: sensor.inverter2_input_power

  total_pv_generation: sensor.inverter1_total_yield
  pv1_voltage_109: sensor.inverter1_pv_1_voltage
  pv1_current_110: sensor.inverter1_pv_1_current
  pv2_voltage_111: sensor.inverter2_pv_1_voltage
  pv2_current_112: sensor.inverter2_pv_1_current

Hopefully this is a good start.

slipx06, 2023-11-25:
@Roving-Ronin  looking good. We can use the entity below for the inverter status. Does the sensor return the string, the number or both?

```
Name: Inverter State
Entity: sensor.inverter1_inverter_state

0b0000_0000_0000_0001: "Standby",
0b0000_0000_0000_0010: "Grid-Connected",
0b0000_0000_0000_0100: "Grid-Connected normally",
0b0000_0000_0000_1000: "Grid connection with derating due to power rationing",
0b0000_0000_0001_0000: "Grid connection with derating due to internalcauses of the solar inverter",
0b0000_0000_0010_0000: "Normal stop",
0b0000_0000_0100_0000: "Stop due to faults",
0b0000_0000_1000_0000: "Stop due to power rationing",
0b0000_0001_0000_0000: "Shutdown",
0b0000_0010_0000_0000: "Spot check",
```

Roving-Ronin, 2023-11-25:
@slipx06 the sensor returns the text string. 

[image]

Here's the messages mine has for 'Device Status':

[image]

So I'd say this is the full list of possible messages for that sensor (not that I've seen many... and hopefully never will).

Name: Device Status
Entity: sensor.inverter1_device_status

"Standby: initializing",
"Standby: detecting insulation resistance",
"Standby: detecting irradiation",
"Standby: grid detecting",
"Starting",
"On-grid",
"Grid Connection: power limited",
"Grid Connection: self-derating",
"Off-grid mode: running",
"Shutdown: fault",
"Shutdown: command",
"Shutdown: OVGR",
"Shutdown: communication disconnected",
"Shutdown: power limited",
"Shutdown: manual startup required",
"Shutdown: DC switches disconnected",
"Shutdown: rapid cutoff",
"Shutdown: input underpowered",
"Grid scheduling: cosphi-P curve",
"Grid scheduling: Q-U curve",
"Grid scheduling: PF-U curve",
"Grid scheduling: dry contact",
"Grid scheduling: Q-P curve",
"Spot-check ready",
"Spot-checking",
"Inspecting",
"AFCI self check",
"I-V scanning",
"DC input detection",
"Running: off-grid charging",
"Standby: no irradiation",
______________________________________________________

The 'Inverter State' you posted above

[image]

Note that currently mine is operating and saying "Grid-Connected, Grid-Connected normally", so it looks like those messages can be grouped together based upon the first 3 number groups being the same i.e. 0b0000_0000_0000_xxxx .

So the options appear to be:

"Standby",
"Grid-Connected, Grid-Connected normally",
"Grid-Connected, Grid connection with derating due to power rationing",
"Grid-Connected, Grid connection with derating due to internalcauses of the solar inverter",
"Normal stop",
"Stop due to faults",
"Stop due to power rationing",
"Shutdown",
"Spot check",

_______________________________________

@kurtdefeyter could you screenshot your battery diagnostic section and also click on the status sensor to confirm the options that is shows are as below ?

Name: Battery Status
Entity: sensor.battery_status

Options:

OFFLINE
STANDBY
RUNNING
FAULT
SLEEP_MODE

slipx06, 2023-11-25:
I'll update the grid connected status logic to be case-insensitive so that your Off-Grid Status is interpreted correctly

Roving-Ronin, 2023-11-25:
@slipx06 ahould we be using this to denote it's Huawei and bring up the correct inverter image and also the status ?

inverter:
  modern: true
  colour: grey
  autarky: power
  auto_scale: true
  model: huawei
  three_phase: true

slipx06, 2023-11-25:
Yes like this once I've updated the code
```
inverter:
  modern: false
  auto_scale: true
  model: huawei
```

Roving-Ronin, 2023-11-25:
@slipx06 D'oh I forgot about that modern : false requirement.

I can confirm though that the grid off symbol no longer appears / grid is correctly showing as online now.

Let me know when to update and I'll check the inverter image / status and advise if it works.

Thanks !

slipx06, 2023-11-25:
I propose the following mapping using the Inverter State. Please check.

```
huawei: {
        standby: {states: ['Standby'], color: 'blue', message: localize('common.standby')},
        selftest: {states: ['Spot check'], color: 'yellow', message: localize('common.selftest')},
        normal: {states: ['Grid-Connected, Grid-Connected normally'], color: 'green', message: localize('common.normal')},
        shutdown: {states: ['Shutdown'], color: 'red', message: localize('common.shutdown')},
        normalstop: {states: ['Normal stop'], color: 'yellow', message: localize('common.normalstop')},
        alarm: {
            states: ['Grid-Connected, Grid connection with derating due to power rationing', `Grid-Connected, Grid connection with derating due to internalcauses of the solar inverter` ], 
            color: 'orange', message: localize('common.alarm')
        },
        fault: {states: ['Stop due to faults', 'Stop due to power rationing'], color: 'red', message: localize('common.fault')},
    },
```

Roving-Ronin, 2023-11-25:
@slipx06 

It looks good apart from the typo ("internalcauses" instead of "internal causes", thats from the WLCRS Huawei integration, I've just submitted a change for this to be fixed: https://github.com/wlcrs/huawei-solar-lib/pull/5

I'd suggest going with the status message with the typo corrected, pending the merge being approved by WLCRS I would think the chances of anyone having that alarm message `Grid-Connected, Grid connection with derating due to internal causes of the solar inverter` appear are non-existant.

```
huawei: {
        standby: {states: ['Standby'], color: 'blue', message: localize('common.standby')},
        selftest: {states: ['Spot check'], color: 'yellow', message: localize('common.selftest')},
        normal: {states: ['Grid-Connected, Grid-Connected normally'], color: 'green', message: localize('common.normal')},
        shutdown: {states: ['Shutdown'], color: 'red', message: localize('common.shutdown')},
        normalstop: {states: ['Normal stop'], color: 'yellow', message: localize('common.normalstop')},
        alarm: {
            states: ['Grid-Connected, Grid connection with derating due to power rationing', `Grid-Connected, Grid connection with derating due to internal causes of the solar inverter` ], 
            color: 'orange', message: localize('common.alarm')
        },
        fault: {states: ['Stop due to faults', 'Stop due to power rationing'], color: 'red', message: localize('common.fault')},
    },
```

slipx06, 2023-11-25:
@Roving-Ronin 
Added to [3.18.0](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v3.18.0). Please test

Roving-Ronin, 2023-11-25:
@slipx06 3am here, so in the dark and in 'Standby' atm. But card now shows:

[image]
Using the 'default' / demo card setup.

[image]
Or here's the version I have with all my other sensors feeding AUX / Essential / Non-Essential... OMG I have no skills for visual arts and selecting colours !

I think for the other battery related sections, we'll need to wait for @kurtdefeyter to reply to confirm the sensor outputs the Huawei Luna + WLCRS integration provides.

Roving-Ronin, 2023-11-25:
@slipx06 on a seperate note, I'm not a software skilled person, but I thought that this might be of interest to you.
https://community.home-assistant.io/t/integration-solar-inverter-huawei-2000l/132350/2753?u=jbhobson

Interesting part is section in which he's using the https://github.com/elax46/custom-brand-icons to allow adding inverter icons  ( phu:huawei-solar-inverter ) versus having to develop your own and then he's also worked out how to make the inverter change colour based upon if its online of not and then blink/flash. Thought this might give some ideas and allow this kind of functionality to highlight status / draw attention to issues.  If not, please ignore.

```
        icon: |
          {% set state=states('sensor.sun2000_10ktl_device_status') %}
          {% if state=='On-grid' %}
          phu:huawei-solar-inverter
          {% elif state!='Standby:no irradiation' %}
          phu:huawei-solar-inverter
          {% endif %}
        icon_color: |
          {% set state=states('sensor.sun2000_10ktl_device_status') %}
          {% if state=='On-grid' %}
          #41bdf5
          {% elif state!='Standby:no irradiation' %}
          disabled
          {% endif %}
        secondary_info: state
        multiline_secondary: false
        badge_icon: ''
        badge_color: ''
        tap_action:
          action: more-info
        card_mod:
          style:
            mushroom-shape-icon$: |
              .shape {
                
                --shape-color: none !important;
                
                {% if is_state('sensor.sun2000_10ktl_device_status', 'On-grid') %}
                --shape-animation: blink 5s linear infinite;
                {% endif %}
              }
              @keyframes blink {
                  30% {opacity: 0;}
              }
            .: |
              ha-card {
                margin-top: 10px;
                --icon-size: 50px;
                --icon-symbol-size: 30px;
                --card-secondary-font-size: 10px;
                --card-primary-font-size: 12px;
                padding-top: 5px !important;
                padding-bottom: 5px !important;
              }
```

slipx06, 2023-11-25:
Thanks. The card is currently rendered through SVG elements so not something we can apply here with that method

kurtdefeyter, 2023-11-26:
Wow you people did a great job so far , regarding the battery
[image]
[image]

Is this the information needed ?

Roving-Ronin, 2023-11-26:
@kurtdefeyter thank, also need to keep an eye on that on the diagram to see it changes (sleep etc).
As well as confirm the battery data on the diagram is displaying correctly also ... charge / discharge/ SoC etc.

kurtdefeyter, 2023-11-26:
@Roving-Ronin    i keep an eye on it ,
Question for you
[image]

The inventor thus put out the V and Amps on each pv group , however we do not have a W output , did you made a template sensor that get this value ?

Roving-Ronin, 2023-11-26:
@kurtdefeyter that is the INPUT POWER for PV 1 and PV2 of your inverter.

kurtdefeyter, 2023-11-26:
@Roving-Ronin 

Weird i do not have this 

[image]

Roving-Ronin, 2023-11-26:
@kurtdefeyter sorry I was referring to the Input Power (next to Efficiency in the picture).

As you have one inverter I'd assume you will be wanting to get the diagram to show one string / MPPT of panels pre PV pane icon.

In this case you'll need to make 2 sensors up. Each sensor will be a state formula that calls the PV1 amps x voltage (to get watts). Repeat for the PV2.

Suggest keeping naming from WLCRS ie 

unique_id: inverter_pv1_input_power and repeat for PV2.

Assume your Ok writing these up and adding state class / availability / type class / measurement units etc?

slipx06, 2023-11-26:
I have not implemented the battery status yet. Was waiting to confirm how the sensor reports these states.

kurtdefeyter, 2023-11-26:
@Roving-Ronin 

Should be like this , is that correct ?

[image]

Roving-Ronin, 2023-11-26:
@kurtdefeyter   fyi to post code into here, you use the key to left of the 1 key,  type that symbol 3 times to open the section and enter the code, then on the next / closing line type 3 more to close the code section.

```
template:

#############################
#   
#   WLCRS - Huawei Solar Integrattion - Custom Sensors for SunSynk Power Flow Card
#   See: https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-daisy-chained-inverters
#          https://github.com/slipx06/sunsynk-power-flow-card
#
#   If your HA instance has a /config/packages directory configured, this may be deployed by saving this text into a file
#   in that location. i.e.   /config/packages/huawei_sunsynk_derived_sensors.yaml
#

  # Shows the inverter PV1 Input Power in Watts, use for with sunsynk:  pv1_power_186:
  - sensor:
    - name: "Inverter PV 1 Input Power"
      unique_id: inverter_pv_1_input_power
      icon: mdi:solar-panel
      unit_of_measurement: "W"
      device_class: "power"
      state_class: measurement
      state: >
        {% set AMPS = states('sensor.inverter_pv_1_current') | float %}
        {% set VOLTS = states('sensor.inverter_pv_1_voltage') | float %}
        {{ ( AMPS * VOLTS ) | float }}
      availability: >-
        {{ (states('sensor.inverter_pv_1_current')|is_number)
            and (states('sensor.inverter_pv_1_voltage')|is_number) }}

  # Shows the inverter PV2 Input Power in Watts, use for with sunsynk:  pv2_power_187:
    - name: "Inverter PV 2 Input Power"
      unique_id: inverter_pv_2_input_power
      icon: mdi:solar-panel
      unit_of_measurement: "W"
      device_class: "power"
      state_class: measurement
      state: >
        {% set AMPS = states('sensor.inverter_pv_2_current') | float %}
        {% set VOLTS = states('sensor.inverter_pv_2_voltage') | float %
        {{ ( AMPS * VOLTS ) | float }}
      availability: >-
        {{ (states('sensor.inverter_pv_2_current')|is_number)
            and (states('sensor.inverter_pv_2_voltage')|is_number) }}

```

Could you check and see that this works, entering them against pv1_power_186  and  pv2_power_187 ?
Also if you could please advise if there are any other sensors that don't work for you, apart from the aux and essential (that you'll need to work out for your specific setup what you want to monitor / can monitor if you have the gear to pull readings from outside of Huawei (i.e. Shelly EM / IoTaWatt / Athom plugs etc).

Also can you keep an eye and validate all the options for the battery, as per @slipx06 examples, like:

[image]

such as the runtime, charges, %'s etc... not sure exactly that the Priority Load and Time On/Off equate to or actually do ... ???

@slipx06 could you advise re that and also these entities, as I don't really follow what they do....hence a bit stuck translating to Huawei version or if Huawei has something the same... I was assuming something to do with setting times to charge / discharge the battery ?

prog1_time: sensor.sunsynk_time_slot_1
prog1_capacity: number.sunsynk_system_mode_soc_time1
prog1_charge: switch.sunsynk_system_mode_grid_charge_time1

slipx06, 2023-11-26:
The Sunsynk allows you to prioritise the load or the battery.  So PV goes to load and excess to battery etc This can be toggled.
`Timer on` refers to the the Sunsynk system timer. If enabled the inverter can be configured with specific SOC, Power and charge conditions i.e  grid/generator/solar etc. If disabled the inverter acts like a UPS. See settings below. 

[image]

slipx06, 2023-11-26:
Please test battery status messages in 3.20.1

You can use the entity `battery_status:` e.g.  `battery_status:sensor.battery_status`

kurtdefeyter, 2023-11-27:
@Roving-Ronin 

All battery params are ok 
[image]

However the flow is wrong , now as we speak the battery discharge something like 1300 w , however the flow is showing like the inventor is putting current in the battery

I have done a test last night charging a 5000w and then it shows the other way , like the battery is transferring to the inventor

On battery temperature, that sensor thus not exist , only the inventor has a  temperature sensor

Roving-Ronin, 2023-11-27:
@kurtdefeyter 

a) add   invert_power: true  so the top part of the card yaml reads:

```
battery:
  energy: 14850
  shutdown_soc: 5
  invert_power: true
  colour: pink
  show_daily: true
  animation_speed: 6
  max_power: 4500
  full_capacity: 95
  empty_capacity: 5
  show_absolute: false
  auto_scale: true
```

Can you also send that 'whole of card' picture and advise re the above data requested and what sensors you had to enable (i.e. power meter Phase A/B/C Active Power and also Phase A/B/C Voltage, Frequency and Power Factor (that aren't enabled by default with WLCRS iirc) ?
Really need to work out what sensors need to be enabled and what sensors we need to define, so we can give @slipx06 a mapping of the sensors to use in the card, and then also a block of yaml code that new users can add into their HA to have any custom sensors created automatically..

Thanks.

slipx06, 2023-11-27:
The battery energy can be provided in one of 3 ways:

1. Specified directly in the battery card config using the energy attribute as a numeric value i..e 15960
2. Provided through a sensor i.e sensor.battery_energy in the battery card config using the energy attribute
3. The previous two are ignored if you provide a sensor in the entities section with the  battery_rated_capacity: in Ah and it will calculate battery energy by multiplying  battery Voltage x battery_rated_capacity

I would recommend option 1 or 2 as the 3rd option has not been tested thoroughly

Roving-Ronin, 2023-11-27:
@slipx06 OK so having a sensor.battery_rated_capacity is required essentially. Looking and it seems that Huawei LUNA ESS (battery) have a modbus register for this.

[image]

But checking the python library for WLCRS it looks like this isn't currently supported / queried. 
https://github.com/wlcrs/huawei-solar-lib/blob/master/src/huawei_solar/register_names.py
A request to add it could be raised though...

I would presume all batteries report their Rated Capacity in Amp Hours... ???
Or else @kurtdefeyter are you able to use something like https://qmodbus.sourceforge.net/ to query your primary inverters IP for this info?

slipx06, 2023-11-27:
sensor.battery_rated_capacity is not required. If you have a sensor that reports battery capacity in Wh then this is preferred and can be used in the config as follows:

```
battery:
  energy: sensor.battery_capacity_wh
```

Roving-Ronin, 2023-12-05:
@kurtdefeyter we need to wrap this up for @slipx06 and provide him the ' FULL' list of entities that work for Huawei, to allow publishing in the documentation this list for new users to copy/paste and be up and (mostly) running with Huawei inverters. Once you reply with your yaml / sensor list, I can double check against mine (and add the battery sensors in) and provide Slipx06 with the final list.

As requested above, could you please send a screenshot of the entire card you have and the card configuration (yaml) your using, noting an custom sensors you had to make?

If you could you also please provide the information (sensors name / ID) for the battery data from:
https://github.com/slipx06/sunsynk-power-flow-card/issues/135#issuecomment-1826833223

that should correspond to the data from WLCRS (here):
[image]

@slipx06 
Side note...I note in your updated 'Full' list of sensors, you are missing a few, that appear mostly to be those related to 3 phase inverters. Wondering if it would be a good idea to include them in the card example, even with 'NULL' against them, so at least people know of their existance and can hunt down the sensor they should use with it? i.e.

 inverter_voltage_L2: null
 inverter_voltage_L3: null
 inverter_current_L2: null
 inverter_current_L3: null
 total_pv_generation: null
 battery_status: null
 load_power_L1: null
 load_power_L2: null
 load_power_L3: null
 grid_ct_power_L2: null
 grid_ct_power_L3: null

slipx06, 2023-12-05:
Thanks I'll add those to the example

Roving-Ronin, 2023-12-29:
@slipx06 
Could you please advise, for the battery_current_direction what is the card looking for as the output/status of that sensor ? 
i.e. Charge or Discharge or a negative or positive number of watts ?

Roving-Ronin, 2023-12-29:
Hi Folken,

With multiple inverters ....

a) note that the card can only display 1 inverter, so I've just gone with using the primary inverters Phase A/B/C Current, but then using the combined "Inverters Active Power" that shows the total Active Power of both inverters.  See next item...
Same goes for the shown 'internal temp' and inverter status, just showing the primary inverter (though if really desired a derived sensor could be created, that would be outputting 'if both inverters are OK, then say Normal', if one inverter is not OK then say 'Fault'..

b) do you have the derived sensors ( https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-daisy-chained-inverters#combining-multiple-inverter-values ) setup, especially the combined "Inverters Active Power" (note the plural)  and the  "House Consumption Power" sensor?

Thinking about it, but would the 'house consumption power' need to also factor in the LUNA discharging into the house or charging from the grid? Can you confirm from looking at your loads and power meter outputs?

c) Given the card can only show 1 inverter, but it can show 4 PV strings, its best to just these for pv1 = inverter 1 string 1, pv2 = inverter 1 string 2, pv3 = inverter 2 string 1,  pv4 = inverter 2 string 2.

  pv1_name: IV1.S1
  pv2_name: IV1.S2
  pv3_name: IV2.S1
  pv4_name: IV2.S2

  pv1_power_186: sensor.inverter_input_power_1
  pv2_power_187: sensor.inverter_input_power_2
  pv3_power_188: none
  pv4_power_189: none

Comment: Huawei / WLCRS input power is the total for each inverter, if you wanted to get the input power for each PV string, you could define a derived sensor that is voltage x current = power (w), and use that for 186 - 189.

This would be:
```
template:
  - sensor:
      # ########## Start: PV Strings - Power ##########
      #
      - name: inverter_pv_1_power_1
        unique_id: inverter_pv_1_power_1
        #friendly_name: 'Inverter 1 PV 1 Power'
        state_class: "measurement"
        device_class: power
        unit_of_measurement: "W"
        icon: mdi:solar-power
        state: >-
          {% set pv_current = states('sensor.inverter_pv_1_current_1')|float(0) %}
          {% set pv_voltage = states('sensor.inverter_pv_1_voltage_1')|float(0) %}
          {{ (pv_current * pv_voltage)|float(0) }}
        availability: >-
          {{ (states('sensor.inverter_pv_1_current_1')|is_number)
              and (states('sensor.inverter_pv_1_voltage_1')|is_number) }}

      - name: inverter_pv_1_power_2
        unique_id: inverter_pv_1_power_2
        #friendly_name: "Inverter 1 PV 2 Power"
        state_class: "measurement"
        device_class: power
        unit_of_measurement: "W"
        icon: mdi:solar-power
        state: >-
          {% set pv_current = states('sensor.inverter_pv_1_current_2')|float(0) %}
          {% set pv_voltage = states('sensor.inverter_pv_1_voltage_2')|float(0) %}
          {{ (pv_current * pv_voltage)|float(0) }}
        availability: >-
          {{ (states('sensor.inverter_pv_1_current_2')|is_number)
              and (states('sensor.inverter_pv_1_voltage_2')|is_number) }}

      - name: inverter_pv_2_power_1
        unique_id: inverter_pv_2_power_1
        #friendly_name: 'Inverter 2 PV 1 Power'
        state_class: "measurement"
        device_class: power
        unit_of_measurement: "W"
        icon: mdi:solar-power
        state: >-
          {% set pv_current = states('sensor.inverter_pv_2_current_1')|float(0) %}
[... 31 lines cut ...]
  pv1_voltage_109: sensor.inverter_pv_1_voltage_1
  pv1_current_110: sensor.inverter_pv_1_current_1
  pv2_voltage_111: sensor.inverter_pv_1_voltage_2
  pv2_current_112: sensor.inverter_pv_1_current_2
  pv3_voltage_113: sensor.inverter_pv_2_voltage_1
  pv3_current_114: sensor.inverter_pv_2_current_1
  pv4_voltage_115: sensor.inverter_pv_2_voltage_2
  pv4_current_116: sensor.inverter_pv_2_current_2

d) I note this new entity in the card 'battery_current_direction', checking and there are these registers

Reg.37000 = STORAGE_UNIT_1_CHARGE_DISCHARGE_POWER = "storage_unit_1_charge_discharge_power"  
Reg.37743 = STORAGE_UNIT_2_CHARGE_DISCHARGE_POWER = "storage_unit_2_charge_discharge_power"

Register Output is:     > 0: charging   or...   < 0:discharging

Folken could you:

- Confirm that output of the battery when its discharging / charging ? ( I assume 600W output means charging, whilst -600W means discharging.
- Keep a track of all the sensors you DON'T have from WLCRS/Huawei and that we need to manually create to get the card working, also noting if that is just because you have 2 inverters and/or battery. 
- Keep a clean 'default' copy of the card yaml (with the geyser / pool etc) with the Huawei sensors we define, that we can then publish? i.e. like this one / no entities specific to your install such as pool pump / UPS etc:

[image]

End goal is to get a list of those sensors that are there with WLCRS by default and those we need to create (for a single inverter vs multiple inverter site, and for those with/without a battery), and then map/populate those default (and extra) entities into the Sunsynk card, so we end up with a couple of example configurations for single inverter / two inverters / inverter with battery that @slipx06  can the publish on his site for new Huawei users to simply copy and paste to setup a HA card (like the examples for Solis / Goodwe etc at: https://slipx06.github.io/sunsynk-power-flow-card/index.html )

Pho3niX90, 2023-12-29:
> @slipx06 
> Could you please advise for the battery_current_direction what is the card looking for as the output/status of that sensor ? 
> i.e. Charge or Discharge or a negative or positive number of watts ?

This is mainly used for solis inverters. Their battery power is always positive, with an extra sensor for direction. This is a bool, 1 for charging, 0 for discharging.

Roving-Ronin, 2024-01-01:
@slipx06 given @Pho3niX90 's reply above, does this mean to get this battery_current_direction working for Huawei users would require adding a derived sensor for Huawei + WCLRS integration users, that says 

'If STORAGE_UNIT_#_CHARGE_DISCHARGE_POWER is > 0 (charging) then report sensor = 1;
elseif  STORAGE_UNIT_#_CHARGE_DISCHARGE_POWER  < 0 (discharging) then report sensor = 0;

Or does the SunSynk card already have logic in to the recognise other inverters that show a postive or negative W output and recognise the power direction from the + vs - ?

slipx06, 2024-01-01:
@Roving-Ronin If battery power is both positive and negative to indicate direction then this sensor is not needed and can be ignored.

Roving-Ronin, 2024-01-27:
Hi @slipx06 

Just working with another user to get this all tested with his setup that includes a LUNA ESS battery (and 2 x 5kW L1 1p inverters).
Whilst operating normally his constantly has Inverter Status = 'Grid-Connected, Grid connection with derating due to power rationing'. Is there any chance of updating the card to see this as normal operation instead of currently it being an alarm?

ie. CHANGE FROM:
```
huawei: {
        standby: {states: ['Standby'], color: 'blue', message: localize('common.standby')},
        selftest: {states: ['Spot check'], color: 'yellow', message: localize('common.selftest')},
        normal: {states: ['Grid-Connected, Grid-Connected normally'], color: 'green', message: localize('common.normal')},
        shutdown: {states: ['Shutdown'], color: 'red', message: localize('common.shutdown')},
        normalstop: {states: ['Normal stop'], color: 'yellow', message: localize('common.normalstop')},
        alarm: {
            states: ['Grid-Connected, Grid connection with derating due to power rationing', `Grid-Connected, Grid connection with derating due to internal causes of the solar inverter` ], 
            color: 'orange', message: localize('common.alarm')
        },
        fault: {states: ['Stop due to faults', 'Stop due to power rationing'], color: 'red', message: localize('common.fault')},
    },
```

TO:

```
huawei: {
        standby: {states: ['Standby'], color: 'blue', message: localize('common.standby')},
        selftest: {states: ['Spot check'], color: 'yellow', message: localize('common.selftest')},
        normal: {states: ['Grid-Connected, Grid-Connected normally', 'Grid-Connected, Grid connection with derating due to power rationing'], color: 'green', message: localize('common.normal')},
        shutdown: {states: ['Shutdown'], color: 'red', message: localize('common.shutdown')},
        normalstop: {states: ['Normal stop'], color: 'yellow', message: localize('common.normalstop')},
        alarm: {
            states: [`Grid-Connected, Grid connection with derating due to internal causes of the solar inverter` ], 
            color: 'orange', message: localize('common.alarm')
        },
        fault: {states: ['Stop due to faults', 'Stop due to power rationing'], color: 'red', message: localize('common.fault')},
    },
```

slipx06, 2024-01-27:
Sure I can make that change for you

Roving-Ronin, 2024-01-27:
@slipx06 
D'Oh that seems to have broken it all :-(
Trying to work out what is wrong and all I can see is, is the ' supposed to be a `  ?

As in when that status mess was in: 

```
        alarm: {
            states: ['Grid-Connected, Grid connection with derating due to power rationing', `Grid-Connected, Grid connection with derating due to internal causes of the solar inverter` ], 
            color: 'orange', message: localize('common.alarm')
```

the first error message is defined by ' at the start and end, and the second status message has ` at the start and end.

Where as in the 4.10.3 we have updated it to:

```
    huawei: {
        standby: {states: ['Standby'], color: 'blue', message: localize('common.standby')},
        selftest: {states: ['Spot check'], color: 'yellow', message: localize('common.selftest')},
        normal: {states: ['Grid-Connected, Grid-Connected normally', 'Grid-Connected, Grid connection with derating due to power rationing'], color: 'green', message: localize('common.normal')},
        shutdown: {states: ['Shutdown'], color: 'red', message: localize('common.shutdown')},
        normalstop: {states: ['Normal stop'], color: 'yellow', message: localize('common.normalstop')},
        alarm: {
            states: [`Grid-Connected, Grid connection with derating due to internal causes of the solar inverter` ], 
            color: 'orange', message: localize('common.alarm')
        },
        fault: {states: ['Stop due to faults', 'Stop due to power rationing'], color: 'red', message: localize('common.fault')},
    },
```

That has both the status messages for the 'normal definition, being found with ' at the start/end of each... should the second message have ` instead ?
Also the alarm message has ` at the start/end, whereas all the other definitions with only one status message as an option use ' at the start/end of the message?

slipx06, 2024-01-27:
I've pushed a fix.  Please test

Roving-Ronin, 2024-01-27:
@slipx06 no that doesn't appear to work either. 

4.10.2 worked fine, just that we're trying to move `Grid-Connected, Grid connection with derating due to internal causes of the solar inverter` from being an alarm type message, to being a normal type status message.

i.e. so normal can be either of 'Grid-Connected, Grid-Connected normally'  or `Grid-Connected, Grid connection with derating due to internal causes of the solar inverter`.  Wilst the alarm message options shrinks to only have the one message option, that being `Grid-Connected, Grid connection with derating due to internal causes of the solar inverter`

PS. Apologies for all the hassle.

slipx06, 2024-01-27:
OK I see the issue.  Pushed another version to test

Roving-Ronin, 2024-01-27:
@slipx06 Thanks version is working now. Appreciated. Hopefully will have mapping list (with battery) and extra items to allow copy/paste for Huawei users to setup.

Roving-Ronin, 2024-02-03:
@slipx06 Apologies but I'm not having any luck working out how to create examples in the documentation for Huawei. Would you be able to setup a page for this, probably looking at examples for:

a) 1 x L1 1phase inverter with 15kWh LUNA ESS battery - 2 PV strings.
b) 2 x L1 1phase inverter with 15kWh LUNA ESS battery - 4 PV strings (2 per inverter) and Primary inverter used for temp etc.
c) 1 x M1 3phase inverter with 15kWh LUNA ESS battery - 2 PV strings.
d) 2 x M1 3phase inverter with 15kWh LUNA ESS battery - 4 PV strings (2 per inverter) and Primary inverter used for temp etc.

I would assume users should be able to work out from that if they don't have a battery, how to disable it in the card etc.

... then the additional derived sensors that users need to create in HA, from the standard WLCRS integration from Huawel, to cater for mulitple inverters combined values, daily kW/h usage/export totals, AUX and Non-Essential Power/Energy totals etc.

For 'solar_sell_247' trying to work out if there's a call for this for Huawei and if so, if/how to setup a switch that calls a WLCRS added service to change the Power Feed to Grid to Maximum (i.e. Sell) or Zero (i.e. Don't sell)

[image]

Debating including examples of sensors to define the current electricity costs (or compensation from exporting kWh), that are then used in the card, or (most probably) including a link to the WLCRS integration Wiki ( https://github.com/wlcrs/huawei_solar/wiki  - new page to add under here).

To start, here's the 2 x Huawei L1 (1phase) with 1 x 15kWh LUNA ESS battery - 4 PV strings:

```
type: custom:sunsynk-power-flow-card
cardstyle: full
panel_mode: false
large_font: false
title: Huawei Power Monitor
title_colour: White
title_size: 18px
show_solar: true
show_grid: true
show_battery: true
decimal_places: 2
inverter:
  modern: false
  colour: grey
  autarky: power
  auto_scale: true
  model: huawei
  three_phase: false
battery:
  energy: 14850
  shutdown_soc: 15
  invert_power: true
  colour: pink
  show_daily: true
  animation_speed: 5
  max_power: 5000
  show_absolute: false
  auto_scale: true
  hide_soc: false
  show_remaining_energy: false
solar:
  colour: '#5fb6ad'
  show_daily: true
  mppts: 4
  animation_speed: 8
  max_power: 13200
  pv1_name: Inv.1 PV1
  pv2_name: Inv.1 PV2
  pv3_name: Inv.2 PV1
  pv4_name: Inv.2 PV2
  display_mode: 2
load:
  colour: magenta
  show_daily: true
  show_daily_aux: true
  show_aux: true
  invert_aux: false
  show_absolute_aux: false
  aux_name: Generator
  aux_type: gen
[... 95 lines cut ...]
  grid_connected_status_194: sensor.inverter_off_grid_status
  aux_power_166: sensor.sunsynk_power_flow_aux_devices_active_power
  aux_connected_status: binary_sensor.sunsynk_aux_connected_status
  energy_cost_buy: sensor.electricity_price
  energy_cost_sell: sensor.electricity_fit
  solar_sell_247: switch.null
  aux_load1: sensor.it_hardware_network_active_power
  aux_load2: sensor.it_hardware_servers_active_power
  aux_load1_extra: sensor.env_network_rack_bme280_temperature
  aux_load2_extra: sensor.env_server_rack_bme280_temperature
  grid_voltage: sensor.power_meter_voltage
```

Note: I haven't defined the:
```
  prog1_time: sensor.sunsynk_time_slot_1
  prog1_capacity: number.sunsynk_prog1_capacity
  prog1_charge: switch.sunsynk_prog1_grid_charge
```
as whilst Huawei has the ability to define (and read via Modbus / WLCRS integration) 'Time of Use Periods'  and 'Fixed Charging Periods' these are made available as attributes in the sensors, where as I assume the card / for Sunsynk inverters expects this to be  individual senors for each time period?

[image]

[image]
(Based upon the times I'm guessing the + vs - at the end of these periods means charge vs discharge).

slipx06, 2024-02-05:
@Roving-Ronin Ive added the example section to populate 

https://slipx06.github.io/sunsynk-power-flow-card/examples/huawei.html

Roving-Ronin, 2024-02-26:
@slipx06 I've just submitted a pull request for this. Still just needing to add some example tariff plan yaml though. I've created a sub folder in the examples to hold the huawei example yaml files.  

As I'm working off my Fork of your repository (so I can make edits and only merge when ready), added a discussion section (in addition to the 'Issues' section is has by default), thinking that instead of clogging up your discussions with Huawei / iStore (fyi in Australia Huawei is now being resold via a partner as iStore) 'can you fix this' or 'can you add that' that people can do that there. Which will also allow them to then use the example yamls from there to test (as they'll be the BETA versions), prior to them merging with your repository (i.e. GA versions).

Thinking as I type... would it be a good idea to add a 'Section' in your Discussion area for "Specific Vendor Support" and then creating 'Categories' within that for the various inverter brands now supported? i.e. Sunsynk, Deye, Solis, Lux, FoxESS, Goodwe, Huawei / iStore, Victron etc?  Then in the Huawei / iStore just put a comment for issues/changes/requests etc refer to https://github.com/Roving-Ronin/sunsynk-power-flow-card/discussions ?

slipx06, 2024-02-26:
Thanks for the support suggestion. I've added the new section and categories.

Roving-Ronin, 2024-02-27:
@slipx06 
Just noticed that there is an entity in the card for the battery called 'shutdown_soc_offgrid:' 

"Only applies for Goodwe inverters. The offgrid battery shutdown percentage used to calculate remaining runtime. Numeric value or sensor i.e. sensor.offgrid_battery_capacity_shutdown" 

Just wondering if this is something that can be extended to Hauwei users also, as looking at the Huawei battery settings and there are two sensors for a Huawei LUNA battery:

1) number.battery_end_of_discharge_soc    = This is the 'normal' operation SoC % at which the battery stops discharging. Default 15%.

2) battery_backup_power_soc = When the grid is off-line and the battery operating, this is the SoC% the battery goes to. Default 0%.

Also looking and wondering if like these entities, could we use sensors instead of numbers in the card config for:

1) max_power:  as Huawei has the sensor number.battery_maximum_discharging_power that would allow pulling this figure directly from the battery.

I don't think the card has an entity for it, but if there was something like...
2) max_discharge_power:   Hauwei / WLCRS also publish this as a sensor called  number.battery_maximum_charging_power

Not sure if this could assist also with allowing a limit to be dynamically added to the cards formula when calculating recharge times etc?

Thanks. Justin