slipx06/sunsynk-power-flow-card #137: option to remove "Battery additional information rectangel"

MonsterD1, 2023-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

on the lite version the battery part has a rectangle to implement additionl information on W/A etc. there is no configuration available to remove the rectangle if not needed. 

[https://community.home-assistant.io/t/sunsynk-deye-inverter-power-flow-card/562933/234?u=monster_d](url)

### Expected behaviour

the additional information rectangel should be hideble on option. 

### Possible Solutions

option to remove the rectangle

### Mode

Both

### Context / Reason

the look and feel of the view is much better without empty rectangel.

slipx06, 2023-11-16:
Hi. The card is designed to work with a minimum set of data. This includes battery power, current and volts so that rectangle should not be empty. For now you should set the attributes to none to hide them.

[image]

[image]

slipx06, 2023-11-30:
Busy working on a new layout see #156

closed 2023-11-30 by slipx06