slipx06/sunsynk-power-flow-card #138: New entity grid_ct_power_total

mamo-nick, 2023-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

In case of `three_phase: true` the grid_ct_power total value is calculated
After click on it, the  history of grid_ct_power_172 entity is displayed.

[image]

### Expected behaviour

The rectangle is non-clickable like other calculated entities.

### Possible Solutions

1. The rectangle is non-clickable like other calculated entities.

and

2. Create a new entity for the sum of these entities. E.g. grid_ct_power_total - this entity is provided by goodwe integration and possibly others, or can be calculated in HA. Rectangle is clickable in this case. 

### Mode

Both

### Context / Reason

Correct display of values.

slipx06, 2023-11-16:
Will fix this in the next release

closed 2023-11-17 by slipx06

mamo-nick, 2023-11-17:
Retest OK. Thanks👍🏼

commit 6c58289a (2023-11-17): feat: new entitiy `grid_ct_power_total` close #138