slipx06/sunsynk-power-flow-card #143: add charge/discharge information for battery

MonsterD1, 2023-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

no information for current charge/discharge load avaiable

### Expected behaviour

create a rectangel for the (chare/dischrge "W") for the battery
[image]

### Mode

lite

### Context / Reason

to be able to see the battery charge/discharge

FajnB, 2023-11-22:
I think you might have miss-configured you card.
The charge/discharge power is in the rectangle next to the battery icon. (in yours it is that 0 W)

In my config and inverter when the battery is charging, the power is in negative value and vice versa.
Also there is an text what the battery is doing.

[image]

You might look into your yaml and configure `battery_power_190:` as shown [here on wiki](https://slipx06.github.io/sunsynk-power-flow-card/configuration.html#entities).

MonsterD1, 2023-11-22:
thanks to your hint. I changed my config accordingly.
I just wondered why the battery (w) is not in an separat rectangle like on the grid power side, solar power side, load power side ... 
[image]

slipx06, 2023-11-22:
Hi. I prefer to keep all the battery information together. There is also not enough space when all the card options are enabled

[image]

closed 2023-11-22 by slipx06