slipx06/sunsynk-power-flow-card #145: Change color on load

perjmolsen, 2023-11-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I have looked in the documentation, but I am unable to find the setting.

### Expected behaviour

I want the house to change color to the solor where the power comes from. 
Ex.. my solar panel color are orange, so if my solar panels are supplying the house load the color of the house has to be orange.
And also if I draw my house current from grid then, the house color changes to the grid color.
And last if I drwa my load from solar battery, then the house color need to change to the battery color.

It would be great if it where possible to mix the colors, so the hose has the % of color from the loads.

### Possible Solutions

Make some color settings :)

### Mode

lite

### Context / Reason

It would look great.
The card power-flow-card-plus has this option to change color, but it is not as great as sunsynk-power-flow-card

slipx06, 2023-11-23:
Nice idea. Let me see what I can do

perjmolsen, 2023-11-24:
Great, I am looking forward to try it, if it's possible :)

slipx06, 2023-11-24:
I've added this to [3.17.0](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v3.17.0). Give it a try 😄

perjmolsen, 2023-11-24:
Is working GREAT! :) 
[image]
Thanks !

closed 2023-11-25 by slipx06

bazcurtis, 2023-11-28:
I have just seen this in my dashboard. What a great idea.