slipx06/sunsynk-power-flow-card #146: after upgrading to new version (Ver 3.16.1) the grid_ct_power value shows negative, but value is positive on HA

roanvn, 2023-11-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

grid_ct_power_172: "The value on HA is positive (on DB), but it shows negative on the flow card. This happened after upgrading to the latest version

### Expected behaviour

the value need to be positive and show trickle feed value when not importing from the grid

### Card Version

3.16.1

### Home Assistant Version

2023.11.3

slipx06, 2023-11-23:
Please share your card config. 
Are you using three phase?
Have you set invert_grid: true?

roanvn, 2023-11-24:
type: custom:sunsynk-power-flow-card
cardstyle: full
card_height: 450
panel_mode: false
show_solar: true
Large_font: false
decimal_places: 2
solar:
  mppts: 2
  show_daily: true
  max_power: 4100
  pv1_name: North
  pv2_name: North
  auto_scale: true
inverter:
  modern: true
  colour: black
  auto_scale: true
  model: sunsynk
battery:
  energy: 15960
  shutdown_soc: 20
  show_daily: true
  battery_temp: true
  inverter_ac_temp: true
  inverter_dc_temp: true
  auto_scale: true
load:
  show_daily: true
  show_aux: 'no'
  invert_aux: 'no'
  aux_name: Generator
  aux_type: gen
grid:
  show_daily_buy: true
  show_nonessential: true
  invert_grid: 'no'
  nonessential_dual: 'no'
  nonessential_name: Non Essential
  nonessential_icon: oven
  nonessential_power: sensor.niewie_inverter_sunsynk_nonessential_power
  additional_loads: 0
  load1_name: GeyserBo
  load2_name: GyeserFlat
  non_essential_load1: sensor.power
  non_essential_load2: sensor.power_2
  max_power: 5000
  auto_scale: true
  three_phase: false
entities:
  inverter_voltage_154: sensor.niewie_inverter_sunsynk_inverter_voltage
  load_frequency_192: sensor.niewie_inverter_sunsynk_inverter_frequency
  inverter_current_164: sensor.niewie_inverter_sunsynk_inverter_current
  inverter_power_175: sensor.niewie_inverter_sunsynk_inverter_power
  grid_power_169: sensor.niewie_inverter_sunsynk_grid_ld_power_167
  battery_voltage_183: sensor.niewie_inverter_sunsynk_battery_voltage
  battery_soc_184: sensor.niewie_inverter_sunsynk_battery_soc
  battery_power_190: sensor.niewie_inverter_sunsynk_battery_power
  battery_current_191: sensor.niewie_inverter_sunsynk_battery_current
  grid_ct_power_172: sensor.niewie_inverter_sunsynk_grid_ct_power
  grid_connected_status_194: binary_sensor.niewie_inverter_sunsynk_grid_connected_status
  inverter_status_59: sensor.niewie_inverter_sunsynk_overall_state
  pv1_power_186: sensor.niewie_inverter_sunsynk_pv1_power
  pv2_power_187: sensor.niewie_inverter_sunsynk_pv2_power
  pv1_voltage_109: sensor.niewie_inverter_sunsynk_pv1_voltage
  pv1_current_110: sensor.niewie_inverter_sunsynk_pv1_current
  pv2_voltage_111: sensor.niewie_inverter_sunsynk_pv2_voltage
  pv2_current_112: sensor.niewie_inverter_sunsynk_pv2_current
  day_pv_energy_108: sensor.niewie_inverter_sunsynk_day_pv_energy
  day_battery_charge_70: sensor.niewie_inverter_sunsynk_day_battery_charge
  day_battery_discharge_71: sensor.niewie_inverter_sunsynk_day_battery_discharge
  day_load_energy_84: sensor.niewie_inverter_sunsynk_day_load_energy
  day_grid_import_76: sensor.niewie_inverter_sunsynk_day_grid_import
  use_timer_248: sensor.current_use_timer_248
  priority_load_243: sensor.energy_mode_243
  essential_power: sensor.niewie_inverter_sunsynk_essential_power
  aux_power_166: none

I don't use three phase.
the card do not accept the invert_grid: true configuration value.
[image]

roanvn, 2023-11-24:
[image]

slipx06, 2023-11-24:
Hi

You have some mistakes in your card configuration. For a full list of options please read the documentation https://slipx06.github.io/sunsynk-power-flow-card/configuration.html

Just quickly looking the following is wrong:

```
show_aux: 'no'
invert_aux: 'no'
invert_grid: 'no'
nonessential_dual: 'no'
```

Should be changed to:

```
show_aux: false
invert_aux: false
invert_grid: false
additional_loads: 0
```

Legnava89, 2023-11-24:
A negative value is a withdrawal from the distribution network.
If I set invert_grid: false , the value is displayed correctly, but the dot animation works in reverse.

[image]

`type: custom:sunsynk-power-flow-card
cardstyle: full
panel_mode: true
card_width: 100%
card_height: 376px
large_font: false
title: Goodwe
title_colour: '#fffacd'
title_size: 35px
show_solar: true
show_battery: true
show_grid: true
inverter:
  modern: false
  three_phase: true
  colour: '#fffacd'
  autarky: power
  auto_scale: true
  model: goodwe_gridmode
battery:
  energy: 10650
  shutdown_soc: 40
  invert_power: false
  colour: '#00ff00'
  show_daily: true
  auto_scale: true
  animation_speed: 6
  max_power: 4500
  show_absolute: false
  full_capacity: 100
  empty_capacity: 20
solar:
  colour: '#ffd700'
  show_daily: true
  auto_scale: true
  display_mode: 3
  mppts: 2
  animation_speed: 9
  max_power: 8100
  pv1_name: JIH
  pv2_name: ZÁPAD
  pv3_name: East
  pv4_name: West
load:
  colour: '#00bfff'
  show_daily: true
  show_daily_aux: true
  show_aux: true
  invert_aux: false
  show_absolute_aux: false
  auto_scale: true
  aux_name: Generator
  aux_type: gen
  aux_colour: '#00bfff'
  aux_off_colour: red
  aux_loads: 2
  aux_load1_name: BOJLER
  aux_load2_name: KLIMATIZACE
  aux_load1_icon: mdi:water-boiler
  aux_load2_icon: mdi:air-conditioner
  animation_speed: 8
  max_power: 8000
  essential_name: null
  additional_loads: 2
  load1_name: Kotel
  load2_name: Kuchyň
  load3_name: null
  load4_name: null
  load1_icon: mdi:gas-burner
  load2_icon: oven
  load3_icon: null
  load4_icon: null
grid:
  colour: '#00bfff'
  export_colour: '#fffacd'
  show_daily_buy: true
  show_daily_sell: true
  no_grid_colour: '#a40013'
  show_nonessential: true
  invert_grid: true
  auto_scale: true
  nonessential_name: null
  essential_name: null
  nonessential_icon: oven
  additional_loads: 3
  load1_name: Myčka
  load2_name: Lednice
  load3_name: Pračka
  load1_icon: mdi:dishwasher
  load2_icon: mdi:fridge
  load3_icon: mdi:washing-machine
  animation_speed: 8
  max_power: 8000
  energy_cost_decimals: 1
entities:
  use_timer_248: 'no'
  priority_load_243: input_boolean.nabit_baterii_ze_site
  day_battery_charge_70: sensor.today_battery_charge
  day_battery_discharge_71: sensor.today_battery_discharge
  day_load_energy_84: sensor.today_load
  day_grid_import_76: sensor.fve_nakup_daily
  day_grid_export_77: sensor.fve_prodej_daily
  day_pv_energy_108: sensor.today_s_pv_generation
  day_aux_energy: sensor.denni_vstup
  inverter_voltage_154: sensor.on_grid_l1_voltage
  inverter_voltage_L2: sensor.on_grid_l2_voltage
  inverter_voltage_L3: sensor.on_grid_l3_voltage
  load_frequency_192: sensor.meter_frequency
  inverter_current_164: sensor.on_grid_l1_current
  inverter_current_L2: sensor.on_grid_l2_current
  inverter_current_L3: sensor.on_grid_l3_current
  inverter_power_175: sensor.house_consumption
  grid_power_169: sensor.house_consumption
  pv1_power_186: sensor.pv2_power
  pv2_power_187: sensor.pv1_power
  pv3_power_188: none
  pv4_power_189: none
  pv_total: sensor.pv_power
  pv1_voltage_109: sensor.pv2_voltage
  pv1_current_110: sensor.pv2_current
  pv2_voltage_111: sensor.pv1_voltage
  pv2_current_112: sensor.pv1_current
  pv3_voltage_113: none
  pv3_current_114: none
  pv4_voltage_115: none
  pv4_current_116: none
  battery_voltage_183: sensor.battery_voltage
  battery_soc_184: sensor.battery_state_of_charge
  battery_power_190: sensor.battery_power
  battery_current_191: sensor.battery_current
  essential_power: sensor.house_consumption
  essential_load1: sensor.zasuvka_plyn_kotel_power
  essential_load1_extra: sensor.bojler_teplota_teplota_bojler
  essential_load2: sensor.load_l1
  essential_load2_extra: sensor.teplomer_obyvak_temperature
  essential_load3: null
  essential_load4: null
  load_power_L1: sensor.load_l1
  load_power_L2: sensor.load_l2
  load_power_L3: sensor.load_l3
  nonessential_power: sensor.mycka_pracka_lednice
  non_essential_load1: sensor.mycka_energy_power
  non_essential_load2: sensor.lednice_energy_power
  non_essential_load3: sensor.pracka_energy_power
  grid_ct_power_total: sensor.active_power_total
  grid_ct_power_172: sensor.active_power_l1
  grid_ct_power_L2: sensor.active_power_l2
  grid_ct_power_L3: sensor.active_power_l3
  grid_connected_status_194: sensor.grid_mode_code
  inverter_status_59: sensor.on_grid_mode_code
  aux_power_166: sensor.bojler_klima
  aux_load1: sensor.load_l3
  aux_load1_extra: sensor.bojler_teplota_teplota_bojler
  aux_load2: sensor.back_up_l2_power
  aux_load2_extra: sensor.teplomer_obyvak_temperature
  aux_connected_status: binary_sensor.sunsynk_aux_connected_status
  remaining_solar: sensor.energy_production_today
  total_pv_generation: sensor.total_pv_generation
  battery_temp_182: sensor.battery_temperature
  radiator_temp_91: sensor.inverter_temperature_radiator
  dc_transformer_temp_90: sensor.inverter_temperature_air
  prog1_time: sensor.sunsynk_time_slot_1
  prog1_capacity: number.sunsynk_system_mode_soc_time1
  prog1_charge: switch.sunsynk_system_mode_grid_charge_time1
  energy_cost_buy: sensor.current_spot_electricity_price
  energy_cost_sell: sensor.current_spot_electricity_price
  solar_sell_247: binary_sensor.indikator_hdo_hdo
  battery_status: sensor.battery_mode_code`

slipx06, 2023-11-24:
No. By default a positive value is expected for grid import. If your sensor provides a negative value you need to set `invert_grid: true` See below

invert_grid: | Optional | false | Set to true if your sensor provides a negative number for grid import and positive number for grid export

Legnava89, 2023-11-24:
I have invert_grid: true set
But the value on the card shows a positive value
On the active_power_total sensor it shows a negative value

it's alright?
[image]

slipx06, 2023-11-24:
Yes that's by design and what the invert setting does. The card is working as expected.

roanvn, 2023-11-24:
> Hi
> 
> You have some mistakes in your card configuration. For a full list of options please read the documentation https://slipx06.github.io/sunsynk-power-flow-card/configuration.html
> 
> Just quickly looking the following is wrong:
> 
> ```
> show_aux: 'no'
> invert_aux: 'no'
> invert_grid: 'no'
> nonessential_dual: 'no'
> ```
> 
> Should be changed to:
> 
> ```
> show_aux: false
> invert_aux: false
> invert_grid: false
> additional_loads: 0
> ```
> 
> Also `three_phase: false` should be moved to the inverter section

Thank you slipx06. It is now working nicely. 

Thank you for your help.

closed 2023-11-24 by slipx06