slipx06/sunsynk-power-flow-card #147: grid_ct_power_172

neilc1970, 2023-11-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Seeing a negative value for grid power when importing, flow towards the grid.

### Expected behaviour

Flow toward the load

### Possible Solutions

Give the ability to invert grid_ct_power_172

### Mode

lite

### Context / Reason

Best power flow card out there and have everything working perfectly except for grid_ct_power_172. Looks like I'm exporting when I'm importing.

slipx06, 2023-11-23:
Please share your card config. 
Are you using three phase?
Have you set invert_grid: true?

neilc1970, 2023-11-23:
type: custom:sunsynk-power-flow-card
cardstyle: lite
show_solar: true
battery:
  energy: 0
  shutdown_soc: 20
  show_daily: true
solar:
  show_daily: true
  mppts: 2
load:
  show_daily: true
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: true
entities:
  use_timer_248: switch.sunsynk_toggle_system_timer
  priority_load_243: switch.sunsynk_toggle_priority_load
  inverter_voltage_154: sensor.solax_inverter_voltage
  load_frequency_192: sensor.solax_inverter_frequency
  inverter_current_164: sensor.solax_inverter_current
  inverter_power_175: sensor.sunsynk_inverter_power
  grid_connected_status_194: binary_sensor.sunsynk_grid_connected_status
  inverter_status_59: sensor.solax_inverter_status
  day_battery_charge_70: sensor.solax_battery_charge_today
  day_battery_discharge_71: sensor.solax_battery_discharge_today
  battery_voltage_183: sensor.solax_battery_voltage
  battery_soc_184: sensor.solax_battery_soc
  battery_power_190: sensor.solax_battery_output_energy
  battery_current_191: sensor.solax_battery_current
  grid_power_169: sensor.solax_house_load
  day_grid_import_76: sensor.solax_grid_import_today
  day_grid_export_77: sensor.solax_grid_export_today
  grid_ct_power_172: sensor.solax_meter_active_power
  day_load_energy_84: sensor.solax_house_load_today
  essential_power: none
  nonessential_power: none
  aux_power_166: sunsynk_aux_power
  day_pv_energy_108: sensor.solax_power_generation_today
  pv1_power_186: sensor.string_1_output
  pv2_power_187: sensor.string_2_output
  pv1_voltage_109: sensor.solax_pv_voltage_1
  pv1_current_110: sensor.solax_pv_current_1
  pv2_voltage_111: sensor.solax_pv_voltage_2
  pv2_current_112: sensor.solax_pv_current_2

slipx06, 2023-11-23:
Please add 

`invert_grid: true` under the grid section. Does it solve it?
If not please share the output from this sensor `sensor.solax_meter_active_power`

neilc1970, 2023-11-23:
Solved

closed 2023-11-23 by neilc1970