slipx06/sunsynk-power-flow-card #148: Auto calculation of essential

omenek159, 2023-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

In previous version when there was no essential_power set it was calculated using `essential = inverter_power_175 + grid_power_169 - aux_power_166` formula and it worked good. 
Now when I set three_phase and don't set essential_power or load_power_L1/2/3 essential is always 0. I don't have any meter where I could get those values from. I could create template sensor but I think it's not perfect solution. Maybe you would add another check and if both sensors are not set then fallback to previous formula?

### Expected behaviour

It should calculate essential power even when essential_power or load_power_L1/2/3 is not set.

### Card Version

3.16.1

### Home Assistant Version

2023.11.3

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
show_battery: false
panel_mode: false
solar:
  show_daily: true
  mppts: 2
  pv1_name: Garaż
  pv2_name: Szczyt
  auto_scale: true
  display_mode: 2
load:
  show_daily: true
  additional_loads: 1
  load1_name: Nasze
  auto_scale: true
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  auto_scale: true
inverter:
  three_phase: true
  autarky: power
  modern: true
  colour: white
  auto_scale: true
entities:
  inverter_voltage_154: sensor.zamel_mew_01_enea_0_electricity_meter_voltage_phase_1
  inverter_voltage_L2: sensor.zamel_mew_01_enea_0_electricity_meter_voltage_phase_2
  inverter_voltage_L3: sensor.zamel_mew_01_enea_0_electricity_meter_voltage_phase_3
  load_frequency_192: sensor.zamel_mew_01_enea_0_electricity_meter_frequency_phase_1
  inverter_current_164: sensor.solarman_grid_current_l1
  inverter_current_L2: sensor.solarman_grid_current_l2
  inverter_current_L3: sensor.solarman_grid_current_l3
  inverter_power_175: sensor.inverter_active_power_w
  inverter_status_59: sensor.solarman_sofar_status
  day_battery_charge_70: none
  day_battery_discharge_71: none
  battery_voltage_183: none
  battery_soc_184: none
  battery_power_190: none
  battery_current_191: none
  grid_power_169: sensor.enea_power_active_total
  day_grid_import_76: sensor.enea_daily_forward
  day_grid_export_77: sensor.enea_daily_reverse
  grid_ct_power_172: sensor.zamel_mew_01_enea_0_electricity_meter_power_active_phase_1
  grid_ct_power_L2: sensor.zamel_mew_01_enea_0_electricity_meter_power_active_phase_2
  grid_ct_power_L3: sensor.zamel_mew_01_enea_0_electricity_meter_power_active_phase_3
  day_load_energy_84: sensor.daily_energy_consumption
  nonessential_power: none
  essential_power: sensor.solarman_power_used
  aux_power_166: none
  remaining_solar: sensor.solcast_pv_forecast_pozostala_prognoza_na_dzis
  day_pv_energy_108: sensor.solarman_today
  pv1_power_186: sensor.solarman_pv1_power
  pv2_power_187: sensor.solarman_pv2_power
  pv1_voltage_109: sensor.solarman_pv1_voltage
  pv1_current_110: sensor.solarman_pv1_current
  pv2_voltage_111: sensor.solarman_pv2_voltage
  pv2_current_112: sensor.solarman_pv2_current
  essential_load1: sensor.nasze_power_active_total
  essential_load1_extra: sensor.nasze_daily_forward
```

slipx06, 2023-11-24:
I just assumed that those users with 2 phase would have the L1,L2 and L3 sensors. I'll update the logic to fall back to the original calc if those are not present in the config

closed 2023-11-24 by slipx06

omenek159, 2024-02-28:
Hi,
Problem came back on 4.16.5. This update made essential power not calculating again when essential_power or load_power_L1/2/3 are not set.

slipx06, 2024-02-28:
Forgot about this. I'll fix it

commit ae98f201 (2023-11-24): feat: new load attribute `dynamic_colour` fix #148