slipx06/sunsynk-power-flow-card #149: Energy flow animation runs fast for NaN energy

herrfrei, 2023-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

In the evening my OpenDTU devices show NaN as power. That results in fast animation on the card.

### Expected behaviour

No animated point

[image]

### Card Version

3.17.0

### Home Assistant Version

2023.11.3

slipx06, 2023-11-24:
What do those two sensors show as their state in HA?

herrfrei, 2023-11-24:
The say 'Not available' ('Nicht verfügbar') in German.

commit a4288e28 (2023-11-25): fix: return 0 when sensor input is NaN. fixes #149
closed 2023-11-25 by slipx06