slipx06/sunsynk-power-flow-card #150: When solar production is higher than house load 0 is displayed in load

perjmolsen, 2023-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

HI

I have a strange issue, with the new release...
When my solar production is higher that my house load, then a 0 is displayed in the box to the house. The color on my house has the correct color, yellow from only solar consumption...but why is the draw not displayed in the box?

[image]

I only see draw in the house when I use power from battery or from grid.

### Steps To Reproduce

None

### Expected behaviour

I would like the house box to display my draw from the solar panels

### Card Version

3.17.0

### Home Assistant Version

2023.11.3

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: lite
card_height: 420px
card_width: 100%
panel_mode: false
show_solar: true
large_font: true
title: null
title_colour: grey
title_size: 0px
show_battery: true
show_grid: true
inverter:
  model: lux
  modern: true
  colour: orange
  autarky: 'no'
  three_phase: false
  auto_scale: true
battery:
  energy: 13000
  shutdown_soc: 10
  invert_power: true
  colour: lightgreen
  show_daily: true
  animation_speed: 6
  max_power: 3000
  full_capacity: 100
  empty_capacity: 10
  show_absolute: true
  auto_scale: true
solar:
  colour: yellow
  show_daily: true
  display_mode: 2
  mppts: 2
  animation_speed: 9
  max_power: 6500
  pv1_name: SPH3600
  pv2_name: MIC3000
  auto_scale: true
load:
  dynamic_colour: true
  show_daily: true
  additional_loads: 2
  load1_name: Tesla ladeboks
  load1_icon: mdi:ev-station
  load2_name: Blackstar +
  load2_icon: mdi:fire
  auto_scale: true
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: true
  invert_grid: false
  export_colour: green
  colour: white
entities:
  use_timer_248: switch.sunsynk_toggle_system_timer
  priority_load_243: switch.sunsynk_toggle_priority_load
  inverter_voltage_154: sensor.sph3600_grid_voltage
  load_frequency_192: sensor.sph3600_grid_frequency
  inverter_current_164: sensor.sph3600_grid_current
  inverter_power_175: sensor.inverter_combined_house_load
  grid_connected_status_194: binary_sensor.sunsynk_grid_connected_status
  inverter_status_59: sensor.sunsynk_overall_state
  day_battery_charge_70: sensor.sph3600_today_s_battery_input_energy
  day_battery_discharge_71: sensor.sph3600_today_s_battery_output_energy
  battery_voltage_183: sensor.sph3600_battery_voltage
  battery_soc_184: sensor.sph3600_battery_soc
  battery_power_190: sensor.sph3600_battery_combined_power
  battery_current_191: sensor.sph3600_batteri_amp
  grid_power_169: sensor.sunsynk_grid_power
  day_grid_import_76: sensor.sph3600_today_s_grid_import
  day_grid_export_77: sensor.sph3600_today_s_grid_export
  grid_ct_power_172: sensor.power_consumed_watts
  day_load_energy_84: sensor.sph3600_today_s_load
  day_pv_energy_108: sensor.sph3600_today_s_power_generation
  pv1_power_186: sensor.sph3600_pv_power_1
  pv2_power_187: sensor.mic3000tl_xe_pv_power_1
  pv1_voltage_109: sensor.sph3600_pv_voltage_1
  pv1_current_110: sensor.sph3600_pv_current_1
  pv2_voltage_111: sensor.mic3000tl_xe_pv_voltage_1
  pv2_current_112: sensor.mic3000tl_xe_pv_current_1
  battery_temp_182: sensor.sph3600_battery_temperature
  remaining_solar: sensor.energy_production_today_remaining
  energy_cost_buy: sensor.norlys_pulje_el
  energy_cost_sell: sensor.nordpool_kwh_dk1_dkk_2_10_025
  solar_sell_247: switch.sunsynk_toggle_solar_sell
  essential_load1: null
  essential_load1_extra: null
  essential_load2: sensor.fyret_power
  essential_load2_extra: sensor.fyr_kg_idag
```

### Relevant log output

```shell
None
```

perjmolsen, 2023-11-25:
it might not be a bug in the card - can this issue be change to a feature request :)

I have figured out why it 0 when my polar panels are producing more than the house load my sensor
sensor.inverter_combined_house_load is 0.

Is there a way to display the house load from the other sensors...

slipx06, 2023-11-25:
Hi. Since you have not defined an essential_power  sensor the card is calculating the value based on internal card logic 

`essential = inverter_power_175 + grid_power_169 - aux_power_166`

You can set your own sensor i.e. essential_power: sensor.house.load

closed 2023-11-27 by slipx06