slipx06/sunsynk-power-flow-card #153: BMS / Smart Shunt Integration

Pho3niX90, 2023-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Currently the battery SOC is hardcoded

### Expected behaviour

More dynamic approach.

### Possible Solutions

Use sensors to calculate both remaining, and max pack watts. Seplos is one of the most used BMS's currently it seems, so there are integrations available (see https://github.com/syssi/esphome-seplos-bms);
[image];

In the above example, it would be 
`seplos-bms total voltage X seplos-bms rated capacity` = total capacity
`seplos-bms total voltage X seplos-bms residual capacity` = remaining capacity

1. Nominal Voltage Method:
- LiFePO4 batteries typically have a nominal voltage of 3.2 volts per cell, making the nominal voltage for a 4-cell pack 12.8 volts or 51.2 volts for a 16-cell pack.
- You can estimate the remaining capacity by monitoring the voltage and comparing it to the nominal voltage. As the battery discharges, the voltage will decrease, and you can use this drop to estimate the remaining capacity. However, this method might not be highly accurate, as voltage can vary based on factors such as load and temperature.

2. BMS Reported Pack Voltage Method:
- Using the voltage reported by the Battery Management System (BMS) can provide a more accurate estimate of the remaining capacity. The BMS takes into account various factors, including cell voltage imbalances, temperature, and discharge rate.
- Multiply the reported pack voltage by the pack's capacity in ampere-hours (aH) to get an estimate of the remaining capacity. This method considers the actual voltage under the current operating conditions.

The best would probably be to have a new configuration attribs:
`
bms:
   pack_voltage: sensor.xxxx
   rated_capacity: sensor.xxxx
   residual_capacity: sensor.xxxx
`

This is just to get the feel of the implementation, and tracking the conversation. I will be more than happy to do the implementation. 

### Mode

Both

### Context / Reason

Provides accurate remaining capacity.

Pho3niX90, 2023-11-26:
It seems I over thinked this. Just addition 1 additional sensor seems to be enough for now.

closed 2023-11-26 by slipx06

slipx06, 2023-11-27:
@Pho3niX90  Just a comment. If you use battery voltage the available battery energy is going to fluctuate as the voltage changes? The original intention was to gave the rated energy and use the SOC to calculate rate of charge/discharge. I'm not sure the card logic will work the same?

PR #154 (2023-11-26): feat: add battery rated capacity
feat: add battery rated capacity
chore(deps): update lit version

## Description

## Related issues
closes #153 

## Motivation and Context
additional sensor to calculate the battery capacity, instead of hardcoding it. 

## Type of change

Please delete options that are not relevant.

- [X] New feature (non-breaking change which adds functionality)
- [X] This change requires a documentation update

## Checklist

- [X] My code follows the style guidelines of this project
- [X] I have performed a self-review of my code
- [X] I have commented my code, particularly [... cut]