slipx06/sunsynk-power-flow-card #155: Space for a Prepaid kWh sensor

Sboshoff76, 2023-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

N/A

### Expected behaviour

N/A

### Possible Solutions

I use the lite card but I guess it could work on both. For those that keep track of their prepaid units within HA, would it be possible to add an extra entity either above or below the current grid indicator that shows your current unit balance? 

### Mode

Both

### Context / Reason

Thanks

slipx06, 2023-11-26:
Hi. You should be able to use this sensor. 

`energy_cost_buy:`

Sboshoff76, 2023-11-26:
But that's the cost per kwh, not so?

I'm talking about an entity that keeps track of how many kWh units you have
on your prepaid meter / account.

slipx06, 2023-11-26:
Ok I've added it to both cards. The new entity is `prepaid_units:` 😄 

[image] 
[image]

slipx06, 2023-11-27:
Added to [3.21.0](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v3.21.0)

closed 2023-11-27 by slipx06