slipx06/sunsynk-power-flow-card #156: Consistent layout for battery instant power with solar, grid and home

github-daniel-stoian, 2023-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Solar, Grid and Home have a horizontal box on the line of the energy flow to the inverter, where instant power is displayed.
Battery has a bigger box on the left side of the battery icon, bellow current and voltage, not on the line of the energy flow to the inverter.

### Expected behaviour

Battery should have a horizontal box (same size as for Solar, Inverter, Grid, Home) on the line of the energy flow to the inverter, displaying the charge/discharge power.
Battery should have an optional second vertical box on the left side of the icon displaying only current and voltage.
An option should allow to display or not the second vertical box.

### Mode

lite

### Context / Reason

Having all the instant power values (PV, Grid, Home, Battery) in a simetrical layout, surrounding the Inverter, is:
-more consistent visually
-makes the reading easy

slipx06, 2023-11-29:
A few people have asked for this. Working on it.

slipx06, 2023-11-30:
Ready for next release :smile"

[image]

mamo-nick, 2023-12-01:
@slipx06 
Does this mean that the battery status description will not be displayed in `cardstyle: lite` ?

slipx06, 2023-12-01:
> @slipx06 
> Does this mean that the battery status description will not be displayed in `cardstyle: lite` ?

No. This will be a new compact card style. There will be 3 styles, compact, lite and full

commit b2f54279 (2023-12-02): feat:card style `compact` closes #156, closes #157
closed 2023-12-02 by slipx06