slipx06/sunsynk-power-flow-card #158: FoxESS Inverter support

lamixer, 2023-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Card can be pretty well adapted to Fox ESS inverters, but room for improvement if you are interested.

### Expected behaviour

Issues mainly are the inverter statuses, which can be:

Self Test, Waiting, Checking, On Grid, Off Grid / EPS, Recoverable Fault, Unrecoverable Fault

Also, I have to have Home Assistant convert many power readings from kW to Watts. I wonder if the card could have optional flags to accept input of kW instead of the Watts it expects so external conversion (via HA Template entries) is not necessary.

### Possible Solutions

maybe I can help, not sure!

### Mode

Both

### Context / Reason

FoxESS inverters are popular in South Africa (elsewhere perhaps also) and are becoming more popular in the UK where I am based.

slipx06, 2023-12-02:
Hi. Happy to add this. What integration are you using? Re the watts/kilowatts if you have any improvements to the code or suggestions I will take a  look?

lamixer, 2023-12-03:
Hello slipx06

I didn't manually set any of the units of measure, and they seem to be correct since I am using some of them with other cards and they seem to show correctly. Here's a list of most of the sensors:

```
number.foxess_max_charge_current
Max Charge Current (FoxESS)
	18.0 	min: 0	max: 50
step: 0.1
mode: box
unit_of_measurement: A
device_class: current
friendly_name: Max Charge Current (FoxESS)

number.foxess_max_discharge_current
Max Discharge Current (FoxESS)
	32.0 	min: 0	max: 50
step: 0.1
mode: box
unit_of_measurement: A
device_class: current
friendly_name: Max Discharge Current (FoxESS)

number.foxess_max_soc
Max SoC (FoxESS)
	98 	min: 10	max: 100
step: 1
mode: box
unit_of_measurement: %
device_class: battery
icon: mdi:battery-arrow-up
friendly_name: Max SoC (FoxESS)

number.foxess_min_soc
Min SoC (FoxESS)
	10 	min: 10	max: 100
step: 1
mode: box
unit_of_measurement: %
device_class: battery
icon: mdi:battery-arrow-down
friendly_name: Min SoC (FoxESS)

number.foxess_min_soc_on_grid
Min SoC (On Grid) (FoxESS)
	35 	min: 10	max: 100
step: 1
mode: box
unit_of_measurement: %
device_class: battery
icon: mdi:battery-arrow-down
friendly_name: Min SoC (On Grid) (FoxESS)

select.foxess_work_mode
Work Mode (FoxESS)
	Self Use 	options: Self Use, Feed-in First, Back-up
friendly_name: Work Mode (FoxESS)
sensor.foxess_ambtemp
Ambient Temp (FoxESS)
	37.6 	state_class: measurement
unit_of_measurement: °C
device_class: temperature
friendly_name: Ambient Temp (FoxESS)

sensor.foxess_bat_current
Battery Current (FoxESS)
	-1.2 	state_class: measurement
unit_of_measurement: A
device_class: current
friendly_name: Battery Current (FoxESS)
[... 408 lines cut ...]
sensor.foxess_total_yield_today
Yield Today (FoxESS)
	3.3 	state_class: total_increasing
unit_of_measurement: kWh
device_class: energy
icon: mdi:export
friendly_name: Yield Today (FoxESS)

sensor.foxess_total_yield_total
Yield Total (FoxESS)
	1554.9 	state_class: total
unit_of_measurement: kWh
device_class: energy
icon: mdi:export
friendly_name: Yield Total (FoxESS)

sensor.pv_power_foxess
PV Power (FoxESS)
	0.055 	state_class: measurement
unit_of_measurement: kW
device_class: power
icon: mdi:solar-power-variant-outline
friendly_name: PV Power (FoxESS)

```

slipx06, 2023-12-03:
Version 4.1.0 will handle automatic conversion of W, kW etc. I've only implemented this for solar but will enable for the other sections soon.

lamixer, 2023-12-03:
Oh, great! 

Then I guess if it's possible to have an inverter type 'foxess' and with those inverter statuses I listed previously, that would probably do it. Please let me know if you need any additional information.

I would be happy to share my configuration, though I'm not sure it is 100% correct. I could not get essential_power and nonessential_power to show me numbes that seemed correct without specifying them, but that could be due to quirks of my particular set up. You can see the entities that end in _watts are ones I had to convert from kW.

I'd be happy to hear from the expert if something doesn't look right.

```
type: custom:sunsynk-power-flow-card
cardstyle: full
panel_mode: true
card_width: 80%
show_solar: true
inverter:
  auto_scale: true
  colour: '#b6baa9'
  modern: true
battery:
  energy: 33000
  shutdown_soc: 35
  colour: pink
  full_capacity: 87
  max_power: 9000
  show_daily: true
  auto_scale: true
solar:
  show_daily: true
  mppts: 3
  auto_scale: false
  max_power: 5000
  pv1_name: 1 South F
  pv2_name: 2 South R
  pv3_name: 3 North
load:
  show_daily: false
  auto_scale: true
  additional_loads: 2
  load1_name: HW
  load1_icon: mdi:water-boiler
  load2_name: Kitchen
  load2_icon: mdi:stove
  show_aux: true
  aux_name: Climate
  aux_loads: 2
  aux_load1_name: Mitsu AC/Heat
  aux_load1_icon: mdi:heat-pump-outline
  aux_load2_name: Mirror Heater
  aux_load2_icon: mdi:mirror-rectangle
  essential_name: Main
grid:
  show_daily_buy: true
  show_daily_sell: true
  max_power: 12000
  auto_scale: true
  show_nonessential: true
  nonessential_name: EV
  nonessential_icon: mdi:ev-station
entities:
  use_timer_248: 'no'
  priority_load_243: 'no'
  inverter_voltage_154: sensor.emontx4_vrms
  inverter_current_164: sensor.foxess_rcurrent
  inverter_power_175: sensor.sensor_foxess_rpower_watts
  grid_connected_status_194: none
  inverter_status_59: sensor.foxess_inverter_fault_code
  day_battery_charge_70: sensor.foxess_battery_charge_today
  day_battery_discharge_71: sensor.foxess_battery_discharge_today
  battery_voltage_183: sensor.foxess_batvolt
  battery_soc_184: sensor.foxess_battery_soc
  battery_power_190: sensor.sensor_foxess_invbatpower_watts
  battery_current_191: sensor.foxess_invbatcurrent
  day_grid_import_76: sensor.foxess_grid_consumption_energy_today
  day_grid_export_77: sensor.foxess_feed_in_energy_today
  grid_power_169: sensor.sensor_foxess_load_power_watts
  grid_ct_power_172: sensor.sensor_foxess_grid_ct_watts
  day_load_energy_84: sensor.foxess_load_energy_today
  essential_power: sensor.essential_total_power
  nonessential_power: sensor.emontx4_p3
  aux_power_166: sensor.aux_total_power
  day_pv_energy_108: sensor.foxess_solar_energy_today
  pv1_power_186: sensor.sensor_foxess_pv1_watts
  pv2_power_187: sensor.sensor_foxess_pv2_watts
  pv3_power_188: sensor.sensor_foxess_pv3_watts
  pv1_voltage_109: sensor.foxess_pv1_voltage
  pv1_current_110: sensor.foxess_pv1_current
  pv2_voltage_111: sensor.foxess_pv2_voltage
  pv2_current_112: sensor.foxess_pv2_current
  pv3_voltage_113: sensor.foxess_pv3_voltage
  pv3_current_114: sensor.foxess_pv3_current
  nonessential_load1: sensor.emontx4_p3
  essential_load1: sensor.emontx4_p8
  essential_load2: sensor.kitchen_power
  aux_load1: sensor.emontx4_p12
  aux_load2: sensor.shlyclkrm_heater_power
  aux_load2_extra: sensor.shlycloakroom_temperature_2
  energy_cost_buy: sensor.octopus_energy_electricity_xxx_yyy_current_rate
  energy_cost_sell: >-
    sensor.octopus_energy_electricity_xxx_yyy_export_current_rate
```

and image attached of what it looks like (rainy day)!

[image]

lamixer, 2023-12-03:
... will have to wait until the sun comes up to confirm the solar numbers seem correct.

lamixer, 2023-12-03:
Not tested the inverter grid status yet, but it loaded up fine and pulled up the picture. Great! 

Also, I think I should be using autarky: energy like some of the other examples do, but honestly, I'm not too sure what that number should be for me. If it means 'self-sufficiency' then the lower number would be more accurate than the 100% I was seeing before. Now it's showing 44% but that seems high since I only generated 1.7 kW today.

slipx06, 2023-12-03:
Great. I prefer to use power to calculate autarky as it shows the value in real time.  However if you set it to energy It is calculated based on the formula below. 

```
        //Autarky in Percent = Home Production / Home Consumption
        //Ratio in Percent = Home Consumption / Home Production
        //let production_e = parseFloat(state_day_pv_energy.state) + parseFloat(state_day_battery_discharge.state);
        //let consumption_e = parseFloat(state_day_load_energy.state) + parseFloat(state_day_battery_charge.state);
```

lamixer, 2023-12-03:
Thanks for that, sorry I see it's in the docs.

lamixer, 2023-12-03:
Just switched off the grid to test. I correctly get the 

Note, in my example config I had grid_connected_status_194: none but obviously now I need grid_connected_status_194: sensor.foxess_inverter_state.

I think I should also use:

inverter_status_59: sensor.foxess_inverter_state

since the fault codes (I had assigned to that) don't make much sense and can output more than one at a time. So, please update the example config to use the state for _59.

I'm tired now! Should stop typing! Thank you.

lamixer, 2023-12-03:
It's showing actually as 'On Grid' not sure capitalisation counts. When the grid was down that worked though.

slipx06, 2023-12-04:
Thanks I've fixed the typo. It converts those status messages to lowercase.

lamixer, 2023-12-04:
Those strange results should be brief and wouldn't bother me. I don't see any other sensor that shows only grid on / off.

Regarding the W/kW auto-detection on the MPPTs ... works for each individual one, but the sum (auto-calculated by the card I guess) is only showing zero.

slipx06, 2023-12-04:
Thanks for confirming. I see what the issue is. Will fix in the next release

slipx06, 2023-12-04:
Please test https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.2.1

slipx06, 2023-12-04:
The latest version 4.3 should now work with all your kW sensors

lamixer, 2023-12-06:
Sorry for not being available to try that. I did now just update to the latest (4.3.1) and it all working well, including removing my conversions to watts. To that end, we should update the example config. You could replace the example config with this if you wish:

```
type: custom:sunsynk-power-flow-card
cardstyle: full
panel_mode: true
card_width: 80%
show_solar: true
inverter:
  auto_scale: true
  colour: '#b6baa9'
  modern: true
  model: foxess
  autarky: power
battery:
  energy: 33000
  shutdown_soc: 35
  colour: pink
  full_capacity: 87
  max_power: 9000
  show_daily: true
  auto_scale: true
solar:
  show_daily: true
  mppts: 3
  auto_scale: true
  max_power: 5000
  pv1_name: 1 South F
  pv2_name: 2 South R
  pv3_name: 3 North
load:
  show_daily: false
  auto_scale: true
  additional_loads: 2
  load1_name: HW
  load1_icon: mdi:water-boiler
  load2_name: Kitchen
  load2_icon: mdi:stove
  show_aux: true
  aux_name: Climate
  aux_loads: 2
  aux_load1_name: Mitsu AC/Heat
  aux_load1_icon: mdi:heat-pump-outline
  aux_load2_name: Mirror Heater
  aux_load2_icon: mdi:mirror-rectangle
  essential_name: Main
grid:
  show_daily_buy: true
  show_daily_sell: true
  max_power: 12000
  auto_scale: true
  show_nonessential: true
  nonessential_name: EV
  nonessential_icon: mdi:ev-station
entities:
  use_timer_248: 'no'
  priority_load_243: 'no'
  inverter_voltage_154: sensor.emontx4_vrms
  inverter_current_164: sensor.foxess_rcurrent
  inverter_power_175: sensor.foxess_rpower
  grid_connected_status_194: sensor.foxess_inverter_state
  inverter_status_59: sensor.foxess_inverter_state
  day_battery_charge_70: sensor.foxess_battery_charge_today
  day_battery_discharge_71: sensor.foxess_battery_discharge_today
  battery_voltage_183: sensor.foxess_batvolt
  battery_soc_184: sensor.foxess_battery_soc
  battery_power_190: sensor.foxess_invbatpower
  battery_current_191: sensor.foxess_invbatcurrent
  day_grid_import_76: sensor.foxess_grid_consumption_energy_today
  day_grid_export_77: sensor.foxess_feed_in_energy_today
  grid_power_169: sensor.foxess_load_power
  grid_ct_power_172: sensor.foxess_grid_ct
  day_load_energy_84: sensor.foxess_load_energy_today
  essential_power: sensor.essential_total_power
  nonessential_power: sensor.emontx4_p3
  aux_power_166: sensor.aux_total_power
  day_pv_energy_108: sensor.foxess_solar_energy_today
  pv1_power_186: sensor.foxess_pv1_power
  pv2_power_187: sensor.foxess_pv2_power
  pv3_power_188: sensor.foxess_pv3_power
  pv1_voltage_109: sensor.foxess_pv1_voltage
  pv1_current_110: sensor.foxess_pv1_current
  pv2_voltage_111: sensor.foxess_pv2_voltage
  pv2_current_112: sensor.foxess_pv2_current
  pv3_voltage_113: sensor.foxess_pv3_voltage
  pv3_current_114: sensor.foxess_pv3_current
  nonessential_load1: sensor.emontx4_p3
  essential_load1: sensor.emontx4_p8
  essential_load2: sensor.kitchen_power
  aux_load1: sensor.emontx4_p12
  aux_load2: sensor.shlyclkrm_heater_power
  aux_load2_extra: sensor.shlycloakroom_temperature_2
  energy_cost_buy: sensor.octopus_energy_electricity_xxx_yyy_current_rate
  energy_cost_sell: sensor.octopus_energy_electricity_xxx_yyy_export_current_rate
```