slipx06/sunsynk-power-flow-card #161: I have a sofar solar HYD-5000-ES. I can use this?

bobtailcarajas, 2023-12-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Best regards. I have a HYD-5000-ES solar sofar hybrid inverter and a SAJ self-consumption inverter. I have tried using this card but the solar energy from the solar sofar marks me in Kw and the SAJ marks me in W. Therefore, it can't add up well by being in different units. Solar soup apparently does not use negative values. It uses values of 0-65535 for some of its values. Could you integrate this inverter? Thank you.

### Expected behaviour

-

### Possible Solutions

-

### Mode

lite

### Context / Reason

-

slipx06, 2023-12-03:
I'm busy reworking the handling of different units i.e. W and kW. v4.1.0 includes an updated function that looks at the sensor's unit of measurement attribute. Its currently only activated for the solar elements. Please test.

slipx06, 2023-12-07:
This is resolved in release [4.3.0](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.3.0).

closed 2023-12-07 by slipx06