slipx06/sunsynk-power-flow-card #163: Add support for Fronius Inverter

slipx06, 2023-12-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Add support for Fronius Inverters

### Expected behaviour

Add inverter status codes and image

### Mode

Both

### Context / Reason

Add additional inverters

slipx06, 2023-12-05:
List of status codes 
[Fronius Status Codes.xlsx](https://github.com/slipx06/sunsynk-power-flow-card/files/13559464/Fronius.Status.Codes.xlsx)

wgumaa, 2023-12-13:
Working on it :)

wgumaa, 2023-12-13:
In addition, some of the sensors I used to get it to work have been made into templates. Here are some of the conversions: 

`    convert_grid_power:
      unique_id: convert_grid_power
      friendly_name: "Grid Power Watts"
      value_template: "{{ states('sensor.solarnet_power_grid') | float * 1000 | round(2) }}"
      
    convert_total_solar:
      unique_id: convert_total_solar
      friendly_name: "Total Solar Year KW"
      value_template: "{{ states('sensor.symo_advanced_15_0_3_m_1_energy_year') | float / 1000 | round(2) }}"
      
    convert_power_photovoltaics:
      unique_id: convert_power_photovoltaics
      friendly_name: "Power Photovoltaics W"
      value_template: "{{ states('sensor.solarnet_power_photovoltaics') | float(0) * 1000 | round(2) }}"
      
    convert_solar_day:
      unique_id: convert_solar_day
      friendly_name: "Solar Day W"
      value_template: "{{ states('sensor.symo_advanced_15_0_3_m_1_energy_days') | float * 1000 | round(2) }}"
      
    convert_charge_power:
      unique_id: convert_charge_power
      friendly_name: "Charge Power KW"
      value_template: "{{ (states('sensor.wattpilot_charging_energy') | float / 1000) | round(2) }}" 
      unit_of_measurement: kWh

    house_daily_load:
      unique_id: house_daily_load
      friendly_name: "House Daily Load"
      value_template: "{{ (states('sensor.daily_energy_import') | float + states('sensor.solarnet_energy_day') | float - states('sensor.convert_daily_energy_export') | float )}}" 
      unit_of_measurement: kWh

    essential_load:
      unique_id: essential_load
      friendly_name: "Essential Load"
      value_template: "{{ (states('sensor.symo_advanced_15_0_3_m_1_ac_power') | float(0) + states('sensor.convert_grid_power') | float )}}" 
      unit_of_measurement: kWh
      
    convert_daily_energy_export:
      unique_id: convert_daily_energy_export
      friendly_name: "Daily Energy Export KW"
      value_template: "{{ (states('sensor.daily_energy_export') | float / 1000) | round(2) }}"
      unit_of_measurement: kWh
      
    house_average_voltage:
      unique_id: house_average_voltage
      friendly_name: "House Average Voltage"
      value_template: "{{ ((states('sensor.smart_meter_ts_65a_3_voltage_phase_1') | float + states('sensor.smart_meter_ts_65a_3_voltage_phase_2') | float + states('sensor.smart_meter_ts_65a_3_voltage_phase_3') | float ) /3) | round(2) }}"
      `

slipx06, 2023-12-18:
You don't need to convert W to kW or kWh to Wh and visa versa anymore. If you have defined the correct unit of measurement and have `auto_scale` enabled it will calculate the conversion for you.

slipx06, 2023-12-20:
@wgumaa I've added support for the image in 4.7.0

wgumaa, 2023-12-20:
Thanks @slipx06. I am away for the holidays but when I get back I will finish the mapping and update you with the entities to include the fronius system.

wgumaa, 2024-03-01:
[Fronius Status Codes.xlsx](https://github.com/slipx06/sunsynk-power-flow-card/files/14458352/Fronius.Status.Codes.xlsx)

Here is the mapping I did the best I could. As you said too many codes!

wgumaa, 2024-03-03:
Here is the latest: 

[image]

I fixed the daily load but reinstating the template sensor: 

`house_daily_load:
  unique_id: house_daily_load
  friendly_name: "House Daily Load"
  value_template: "{{ (states('sensor.daily_energy_import') | float + states('sensor.solarnet_energy_day') | float - states('sensor.convert_daily_energy_export') | float )}}" 
  unit_of_measurement: kWh`

But the load going in is still wrong.

slipx06, 2024-03-03:
The load is being calculated base on your L1 +L2 +L3 sensors. 

[image]

It look like you have a sensor from the SolarNet Data Manager that you can use for  Essential, namely  `Power Load`

So try adding 

`essential_power: sensor.power_load` <----not sure the exact name

wgumaa, 2024-03-03:
Yes there is a sensor "sensor.solarnet_power_load" Yes but still showing a negative value. Shouldn't it be positive? 

[image]

Considering that the heatpump and charger are pulling?

wgumaa, 2024-03-05:
The last update broke the ratio and autarky

4.18.1 

[image]

wgumaa, 2024-03-05:
I see that but that is calculated and I can't change that unless I make a template sensor. How have you calculated the essential load?

wgumaa, 2024-03-05:
ok we have to figure a way to calculate the essential power. Using "sensor.power_load" is giving a negative value all the time. Is there a way to inverse that through a template?

slipx06, 2024-03-05:
Yes you can create a template sensor that simply multiplies by -1. I don't have an option in the card to invert the load at the moment, but maybe its something I can add in the future

```
inverted_power_load:
  unique_id: inverter_power_load
  friendly_name: "Inverter Power Load"
  value_template: "{{ (states('sensor.power_load') | float * -1)}}" 
  unit_of_measurement: W
```

slipx06, 2024-03-05:
I've added an option to invert the load power in https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.19.0

wgumaa, 2024-03-06:
The Inverse essential load seems to have done the trick! 

[image]

wgumaa, 2024-03-06:
Another bug appeared: when the system is running wholly on solar - not drawing anyhting from the grid, the Autarky and Ratio show 0. 

[image]

wgumaa, 2024-03-07:
Could be because the energy going out is negative? 

I noticed it disappears as soon as it goes in KW but still there when in W

slipx06, 2024-03-07:
Autarky only considers in your case the PV and Essential load. So there is something odd with the sensors. It wont work if your power to load is negative.  The Grid power has not influence on the calculation.

devilliersjohnny, 2024-03-10:
@wgumaa Hi I don't have a Fronius inverter and as such was wondering how you managed to integrate it into HA

slipx06, 2024-04-06:
@wgumaa I've gone though the list of status codes and these are my suggested mappings. These moslty map to error or fault conditions. I have confirmed here https://goodhewsolar.com.au/fronius-fault-codes/
I'm not seeing many "normal" or "standby" states. Can you check

<html

<body link="#0563C1" vlink="#954F72">

Code | Result | Status
-- | -- | --
  |   |  
101 | Grid voltage beyond permitted limits | Alarm
102 | AC voltage is too high | Alarm
103 | AC voltage is too low | Alarm
104 | Mains frequency beyond permitted limits | Alarm
105 | AC frequency is too high | Alarm
106 | AC frequency is too low | Alarm
107 | AC is outside the allowed thresholds | Alarm
108 | Independent operation detected | Alarm
112 | A residential current monitoring unit (RCMU) error has occurred | Fault
141 | Grid frequency increased above threshold | Alarm
  |   |  
301 | Overcurrent (AC) | Alarm
302 | Overcurrent (DC) | Alarm
303 | DC module overheating | Alarm
304 | AC module overheating | Alarm
305 | No power being fed in, despite closed relay | Normal
306 | PV output too low for feeding energy into the grid | Normal
307 | Low PV Voltage | Normal
308 | Intermediate circuit voltage too high | Alarm
309 | DC input voltage MPPT 1 too high | Alarm
311 | Polarity of DC strings reversed | Alarm
313 | DC input voltage MPPT 2 too high | Alarm
314 | Current sensor calibration timeout | Alarm
315 | AC current sensor error | Fault
316 | Interrupt Check fail | Fault
325 | Overtemperature in the connection area | Alarm
326 | Fan 1 error | Fault
327 | Fan 2 error | Fault
  |   |  
401 | No communication possible with the power stage set | Alarm
406 | AC module temperature sensor faulty (L1) | Fault
407 | AC module temperature sensor faulty (L2) | Fault
408 | DC component measured in the grid too high | Alarm
412 | Fixed voltage mode has been selected instead of MPP voltage mode |  Fault
415 | Safety cut out via option card or RECERBO has triggered |  Fault
416 | No communication possible between power stage set and control system | Fault
417 | Hardware ID problem | Fault
419 | Unique ID conflict | Fault
420 | No communication possible with the Hybrid manager | Fault
421 | HID range error | Fault
425 | No communication with the power stage set possible | Fault
426 | Possible hardware fault | Fault
427 | Possible hardware fault | Fault
428 | Possible hardware fault | Fault
431 | Software problem | Fault
436 | Functional incompatibility | Fault
437 | Power stage set problem | Fault
438 | Functional incompatibility | Fault
443 | Intermediate circuit voltage too low or asymmetric | Fault
445 | Compatability error | Fault
447 | Insulation fault | Fault
448 | Neutral conductor not connected | Alarm
450 | Guard cannot be found | Alarm
451 | Memory error detected | Fault
452 | Communication error between the processors | Fault
453 | Grid voltage and power stage set are incompatible | Fault
454 | Grid frequency and power stage set are incompatible | Fault
456 | Anti-islanding function is no longer implemented correctly | Alarm
457 | Grid relay sticking or the neutral conductor ground voltage is too high | Fault
458 | Error when recording the measuring signal | Fault
459 | Error when recording the measuring signal for insulation test | Fault
460 | Reference voltage source for the DSP is working out of tolerance | Fault
461 | Error in the DSP data memory | Fault
462 | Fault with the DC feed monitoring routine | Fault
463 | Reversed AC polarity | Alarm
474 | The RCMU sensor is faulty | Fault
475 | Insulation fault | Fault
476 | The driver supply voltage is too low | Alarm
479 | Intermediate circuit relay has switched off | Alarm
480 | Functional incompatibility | Alarm
481 | Functional incompatibility | Alarm
482 | The setup phase after the initial start-up was interrupted. | Alarm
483 | The voltage UDC fixed on MPP2 string is out of limits | Alarm
485 | The CAN transmit buffer is full | Alarm
489 | Permanent intermediate circuit capacitor overvoltage | Alarm
  |   |  
502 | Insulation fault on the solar panel | Fault
509 | No energy has been fed into the grid in the last 24 hours | Normal
515 | Unable to communicate with filter | Alarm
516 | Unable to communicate with storage unit | Fault
517 | Overheating which is causing power derating | Alarm
518 | Internal DSP fault | Fault
519 | Unable to communicate with storage unit | Alarm
520 | No energy has been fed into the grid by MPPT1 in the past 24 hours | Normal
522 | The DC is low on string 1 | Normal
523 | The DC is low in string 2 | Normal
528 | There is a functional incompatibility | Alarm
529 | There is a functional incompatibility | Alarm
560 | The grid frequency has become excessively high | Alarm
564 | There is a functional incompatibility | Alarm
566 | The arc detector has been deactivated | Alarm
567 | The GVDPR is active and the inverter will operate with reduced power   output | Alarm
  |   |  
601 | The CAN bus is full | Alarm
603 | The AC module sensor is faulty | Fault
604 | The DC module sensor is faulty | Fault
607 | RCMU fault | Fault
608 | There is a functional incompatibility | Fault
</body>
</html>