slipx06/sunsynk-power-flow-card #165: Ratio of house usage colour

Sboshoff76, 2023-12-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Not sure if it's just mine, but it seems that the ratio of power usage in the house icon is incorrect. In the attached example the solar to battery wattage is about 70:30 but the house icon colours seem to be showing 50:50
[image]

### Steps To Reproduce

N/A

### Expected behaviour

70:30

### Card Version

4.3.1

### Home Assistant Version

2023.10.3

Sboshoff76, 2023-12-11:
Sorry.. seems like it was not an internal issue

[image]

I currently have "use AC first" set, meaning all the grid will go into house and all the solar will go into battery, Yet the yellow house colour indicates otherwise.

slipx06, 2023-12-11:
Hi. The card only considers power from the battery and solar and then proportions it based on measured house load. Any shortfall is assumed to be supplied from the grid.  Since you have configured a different priority it wont accurately reflect  your use case  and you should disable the feature.

Sboshoff76, 2023-12-11:
Gotcha. Thanks for clearing it up

closed 2023-12-11 by slipx06