slipx06/sunsynk-power-flow-card #169: 4.4.3 made me turn on invert_grid

lamixer, 2023-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

need to use invert_grid = yes after 4.4.2 update

### Expected behaviour

previously did not use this flag

### Card Version

4.4.3

### Home Assistant Version

2023.12.1

### Configuration

```shell
Maybe it's correct! Just reporting the change I had to make when I installed this version.
```

slipx06, 2023-12-10:
Can you confirm that your grid sensor provides a negative number for grid import and positive number for grid export?

lamixer, 2023-12-10:
Hi...

sensor.foxess_grid_ct

is negative when pulling from the grid, positive when sending to grid. So, what you said = confirmed.

slipx06, 2023-12-10:
OK then the logic is correct 😄 and you need to set `invert_grid: true`

closed 2023-12-10 by slipx06

lamixer, 2023-12-10:
Great. So the the FoxESS config example is correct, could you please add that line to the config? Thanks!