slipx06/sunsynk-power-flow-card #170: House loads - option to subtract individual essential_loads

ragg987, 2023-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Currently load shows total power for all loads, even if individual loads are configured.

### Expected behaviour

Could you add an option to subtract the displayed individual loads from the total load. This would be a feature similar to the one provided in _Power Flow Plus Card_, where it is called "Subtract Individual".

i.e. displayed_load = total_load - essential_load1 - essential_load2...

### Mode

lite

### Context / Reason

Easy to visualize my core house loads vs e.g. my heat pump load.

slipx06, 2023-12-10:
At the moment there is no relationship between the additional essential loads and the total essential load. Just to be clear you are asking for an option to subtract the 866W from the 3.71kW and display in its place the 2.84kW in the example below

[image]

ragg987, 2023-12-10:
Hi yes exactly as per your example would be perfect. A nice to have rather than core.

lamixer, 2023-12-10:
Maybe these should be aux loads instead of essential loads? In that screenshot logically the 866 is part of the 3.71 kW. 

[image]

My 759 watts there is sensor.aux_total_power which is a min/max adding those two heating loads. And the 536 watts is sensor.essential_total_power which is a template sensor I made which takes my inverter's total load power and subtracts those aux loads and subtracts my car charger (which is a non-essential load).

I guess it makes sense to me to subtract non-essential or aux loads from the total the house is using, but not the essential loads.

But I agree with you that it would be great to have some built-in maths to make the power readings in the boxes read what one desires.

ragg987, 2023-12-10:
Good use of non-essential and aux loads (not used those yet as I was playing with Compact setup) however having the option with built-in maths would allow for more flexibility. Plus Compact permits 4 loads.