slipx06/sunsynk-power-flow-card #171: Dynamic Grid Colour for export, off grid, on grid

bazcurtis, 2023-12-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Static colour

### Expected behaviour

Do something similar to the solar lines when they get greyed out.

### Possible Solutions

Grey out the grid when not on the grid, red when importing, green when exporting (I know we can set the export colour). Add the mdi:transmission-tower-export for export and be able to set the colour. I would set that to green.

mdi:transmission-tower-off I would have this grey to match the lines
mdi:transmission-tower would be red as it is today

### Mode

Both

### Context / Reason

I love the way this card is so dynamic and the colours can be set to explain what is going on. My solar is green, battery yellow. The dynamic house icon makes this really easy to see.

Pho3niX90, 2023-12-11:
It might be usefull to maybe change the house icon with this as well. Since currently it's always `home-battery`, maybe should consider `home-lightning-bolt` for grid and pv `home-sound-in`

And sideline, maybe enable discussions for the repo.

slipx06, 2023-12-11:
All good ideas, thanks.

commit 574316a2 (2023-12-12): feat: add dynamic_icon and closes  #171
closed 2023-12-12 by slipx06