slipx06/sunsynk-power-flow-card #174: Visual editor is not supported for this configuration

HybridRCG, 2023-12-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

[image]

Visual editor is not supported for this configuration:
s.entries is not a function. (In 's.entries()', 's.entries' is undefined)
You can still edit your config in YAML.

### Steps To Reproduce

Add new Card on dashboard
No changes 
Cant edit in visual editor because of error

### Expected behaviour

Able to edit the config settings in the visual editor.

### Card Version

4.4.3

### Home Assistant Version

Core : 2023.12.1

slipx06, 2023-12-12:
Hi. Can you share your card config? Does this happen only when you add a new card or with existing cards as well?

HybridRCG, 2023-12-13:
Its the default card. Nothing changed...

type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
battery:
  energy: 0
  shutdown_soc: 20
  show_daily: true
solar:
  show_daily: true
  mppts: 2
load:
  show_daily: true
grid:
  show_daily_buy: true
  show_daily_sell: false
  show_nonessential: true
entities:
  use_timer_248: switch.sunsynk_toggle_system_timer
  priority_load_243: switch.sunsynk_toggle_priority_load
  inverter_voltage_154: sensor.sunsynk_inverter_voltage
  load_frequency_192: sensor.sunsynk_load_frequency
  inverter_current_164: sensor.sunsynk_inverter_current
  inverter_power_175: sensor.sunsynk_inverter_power
  grid_connected_status_194: binary_sensor.sunsynk_grid_connected_status
  inverter_status_59: sensor.sunsynk_overall_state
  day_battery_charge_70: sensor.sunsynk_day_battery_charge
  day_battery_discharge_71: sensor.sunsynk_day_battery_discharge
  battery_voltage_183: sensor.sunsynk_battery_voltage
  battery_soc_184: sensor.sunsynk_battery_soc
  battery_power_190: sensor.sunsynk_battery_power
  battery_current_191: sensor.sunsynk_battery_current
  grid_power_169: sensor.sunsynk_grid_power
  day_grid_import_76: sensor.sunsynk_day_grid_import
  day_grid_export_77: sensor.sunsynk_day_grid_export
  grid_ct_power_172: sensor.sunsynk_grid_ct_power
  day_load_energy_84: sensor.sunsynk_day_load_energy
  essential_power: none
  nonessential_power: none
  aux_power_166: sensor.sunsynk_aux_power
  day_pv_energy_108: sensor.sunsynk_day_pv_energy
  pv1_power_186: sensor.sunsynk_pv1_power
  pv2_power_187: sensor.sunsynk_pv2_power
  pv1_voltage_109: sensor.sunsynk_pv1_voltage
  pv1_current_110: sensor.sunsynk_pv1_current
  pv2_voltage_111: sensor.sunsynk_pv2_voltage
  pv2_current_112: sensor.sunsynk_pv2_current

HybridRCG, 2023-12-13:
It works no issues in yaml editing, but i guess the reason why you wanted to introduce the visual editor was for ease of use.
I have a sunsynk inverter with Solar Assistant integration and use the SA values where applicable.

slipx06, 2023-12-13:
What browser are you using?

HybridRCG, 2023-12-13:
I have tried, 
Safari, Chrome, Brave. 
Im on MAC , tried on IOS Safari aswell.

Pho3niX90, 2023-12-13:
> It works no issues in yaml editing, but i guess the reason why you wanted to introduce the visual editor was for ease of use. I have a sunsynk inverter with Solar Assistant integration and use the SA values where applicable. So it happens with a new card as well as old changed cards. Updated to latest version 4.5.1 , same problem.

Does it work for you in incognito mode?

HybridRCG, 2023-12-13:
[image]

Same error.

slipx06, 2023-12-13:
You should only have one instance (location). Did you ever manually install the card? Can you try delete all instances of the card and install via HACS

HybridRCG, 2023-12-13:
Ok

HybridRCG, 2023-12-13:
1. Removed Sunsynk-Power-Flow-Card in hacs.
2. checked all locations for Sunsynk-Power-Flow-Card , and removed in www.
3. checked dashboard resources , removed it there.
4. removed all cards for - Sunsynk-Power-Flow-Card
5. restart HA
6. Cleared cache
7. installed Sunsynk-Power-Flow-Card in HACS
8. checked in www/community for Sunsynk-Power-Flow-Card (Y)
9. added plan card : Sunsynk-Power-Flow-Card
10. same issue. :(

HybridRCG, 2023-12-13:
No the HA is on Intel nuc. Haos.Best RegardsRiaan Grobler

Are both of you using MacBook's?

Pho3niX90, 2023-12-13:
I have been completely unable to reproduce this my side. `s.entries` cannot be found in the compiled `js` file as well, so I am wondering if this is not perhaps a very niche error for the specific hardware configuration. 

Do you perhaps have more logs that could assist in the debug process? Either in the System -> Logs, as well as in developers tools (F12 for chromium based browsers)

HybridRCG, 2023-12-14:
I have tested this with a windows PC on Chrome. 
Gives the same result.  So the assumption is then safe to say that it is not browser related , but on the HA server.?
When searching for s.entries I cannot find anything in the logs on HA.

slipx06, 2023-12-14:
Can you share some Chrome console logs. 
F12 to open developer console
Access the visual editor and share the output.

ppmt, 2023-12-21:
Just wanted to add that I have the exact same issue. 

I am on Linux using Firefox 121. 

I just updated to the latest version of the card and cleared the cache (I think).

Problem is still here. Here is some logs when I try to add the card

```
Uncaught (in promise) TypeError: o.some is not a function
    updated card-mod.js:5
    promise callback*t.prototype.updated card-mod.js:5
    _$AE reactive-element.ts:1378
    performUpdate reactive-element.ts:1343
    scheduleUpdate reactive-element.ts:1261
    _$Ej reactive-element.ts:1233
    requestUpdate reactive-element.ts:1208
    set reactive-element.ts:726
    value hui-dialog-edit-card.ts:84
    c make-dialog-manager.ts:130
    h make-dialog-manager.ts:163
    r fire_event.ts:76
    o show-edit-card-dialog.ts:18
    value hui-dialog-create-card.ts:213
    handleEvent lit-html.ts:2018
    r fire_event.ts:76
    value hui-card-picker.ts:278
    handleEvent lit-html.ts:2018
    _$AI lit-html.ts:2007
    v lit-html.ts:1192
    g lit-html.ts:1553
    _$AI lit-html.ts:1384
    setValue async-directive.ts:366
    update until.ts:90
    promise callback*update until.ts:72
    _$AS directive.ts:134
    O lit-html.ts:1085
    _$AI lit-html.ts:1362
    v lit-html.ts:1192
    g lit-html.ts:1553
    _$AI lit-html.ts:1384
    T lit-html.ts:1618
    _$AI lit-html.ts:1403
    v lit-html.ts:1192
    g lit-html.ts:1553
    _$AI lit-html.ts:1384
    Z lit-html.ts:2183
    update lit-element.ts:165
    performUpdate reactive-element.ts:1329
    scheduleUpdate reactive-element.ts:1261
    _$Ej reactive-element.ts:1233
    requestUpdate reactive-element.ts:1208
    set reactive-element.ts:726
    value hui-card-picker.ts:234
    value hui-card-picker.ts:210
    _$AE reactive-element.ts:1376
    performUpdate reactive-element.ts:1343
    scheduleUpdate reactive-element.ts:1261
    _$Ej reactive-element.ts:1233
    requestUpdate reactive-element.ts:1208
    _$Eu reactive-element.ts:946
    _ reactive-element.ts:931
[card-mod.js:5:7507](https://haproxmox.ppmt.org/hacsfiles/lovelace-card-mod/card-mod.js?hacstag=190927524315)
Uncaught (in promise) TypeError: o.some is not a function
    updated card-mod.js:5
    promise callback*t.prototype.updated card-mod.js:5
    _$AE reactive-element.ts:1378
    performUpdate reactive-element.ts:1343
    scheduleUpdate reactive-element.ts:1261
    _$Ej reactive-element.ts:1233
    requestUpdate reactive-element.ts:1208
    set reactive-element.ts:726
    value hui-dialog-edit-card.ts:305
    handleEvent lit-html.ts:2018
    r fire_event.ts:76
    value hui-element-editor.ts:162
    promise callback*value hui-element-editor.ts:161
    value hui-element-editor.ts:397
    value hui-element-editor.ts:128
    value hui-element-editor.ts:122
    j lit-html.ts:1883
    _$AI lit-html.ts:1825
    v lit-html.ts:1192
    g lit-html.ts:1553
    _$AI lit-html.ts:1384
    Z lit-html.ts:2183
    update lit-element.ts:165
    performUpdate reactive-element.ts:1329
    scheduleUpdate reactive-element.ts:1261
    _$Ej reactive-element.ts:1233
    requestUpdate reactive-element.ts:1208
    set reactive-element.ts:726
    value hui-dialog-edit-card.ts:84
    c make-dialog-manager.ts:130
    h make-dialog-manager.ts:163
    r fire_event.ts:76
    o show-edit-card-dialog.ts:18
    value hui-dialog-create-card.ts:213
    handleEvent lit-html.ts:2018
    r fire_event.ts:76
    value hui-card-picker.ts:278
    handleEvent lit-html.ts:2018
    _$AI lit-html.ts:2007
    v lit-html.ts:1192
    g lit-html.ts:1553
[card-mod.js:5:7507](https://haproxmox.ppmt.org/hacsfiles/lovelace-card-mod/card-mod.js?hacstag=190927524315)
Unexpected value NaN parsing height attribute. [apexcharts-card.js:1:187097](https://haproxmox.ppmt.org/hacsfiles/apexcharts-card/apexcharts-card.js?hacstag=331701152204)
Unexpected value NaN parsing height attribute. [apexcharts-card.js:1:381775](https://haproxmox.ppmt.org/hacsfiles/apexcharts-card/apexcharts-card.js?hacstag=331701152204)
Uncaught (in promise) TypeError: o.some is not a function
    updated card-mod.js:5
    promise callback*t.prototype.updated card-mod.js:5
    _$AE reactive-element.ts:1378
    performUpdate reactive-element.ts:1343
    scheduleUpdate reactive-element.ts:1261
    _$Ej reactive-element.ts:1233
    requestUpdate reactive-element.ts:1208
    set reactive-element.ts:726
    value hass-base-mixin.ts:47
    value hass-base-mixin.ts:46
    value home-assistant.ts:150
    _$AE reactive-element.ts:1378
    performUpdate reactive-element.ts:1343
    scheduleUpdate reactive-element.ts:1261
    _$Ej reactive-element.ts:1233
    requestUpdate reactive-element.ts:1208
    set reactive-element.ts:726
    value hass-base-mixin.ts:56
    _updateHass context-mixin.ts:95
    hassConnected connection-mixin.ts:227
    s store.js:19
    b entities.js:74
    b entities.js:76
    _handleMessage connection.js:23
    _handleMessage connection.js:15
```

HybridRCG, 2023-12-21:
> @HybridRCG can you test with the latest release if this is still a problem

Hi. The problem is still there. Cleared the cache. New browser. Tried Chrome safari brave Firefox. No luck. 😩

slipx06, 2023-12-21:
@ppmt those errors seem to be related to card mod. Can you share the error that is printed at the bottom of the card editor

[image]

ppmt, 2023-12-21:
it is the same as @HybridRCG 

[image]

slipx06, 2023-12-21:
Can you try with a really simple card config

```
type: custom:sunsynk-power-flow-card-dev
cardstyle: compact
entities:
  pv1_power_186: sensor.sunsynk_pv1_power
  battery_soc_184: sensor.sunsynk_battery_soc
  battery_power_190: sensor.sunsynk_battery_power
  grid_ct_power_172: sensor.sunsynk_grid_ct_power
```

ppmt, 2023-12-21:
I get the same error straight away :(

slipx06, 2023-12-21:
It could be related to this https://github.com/flixlix/power-flow-card-plus/issues/441

ppmt, 2023-12-21:
Yes that did it. I removed card-mod and straight away I can access the UI...

Now just have to remember why I installed card-mod to start with...

HybridRCG, 2023-12-22:
> Great. @HybridRCG <https://github.com/HybridRCG> can you confirm the workaround.
> I see there is an open issue to track this thomasloven/lovelace-card-mod#321 <https://github.com/thomasloven/lovelace-card-mod/issues/321>

Hi 

I have uninstalled card-mod , cleared cache, tried re-installing the sunsynk card. but still have the same issue. :(

Kind Regards
Riaan Grobler

slipx06, 2023-12-22:
Ok. You'll need to share some browser console logs in order to understand what is going on.

HybridRCG, 2023-12-22:
Hi.. 

Found the culprit!!

I had browser-mode installed.
Removed it and started working…
Thanks for the help.

Kind Regards
Riaan Grobler

HybridRCG, 2023-12-22:
Adding back card-mod brings back the problem.

so removed card-mod , browser-mod for the card to work, 

Kind Regards
Riaan

slipx06, 2023-12-22:
Glad we figured this out. You'll need to keep an eye on on those two addon's for updates to hopefully fix this.

closed 2023-12-22 by slipx06