slipx06/sunsynk-power-flow-card #176: add SolaX Power to Inverter options

matomatusov, 2023-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hi, 
it would be possible to add SolaX power to Inverter options?

Thanks

### Expected behaviour

I don't have a description

### Card Version

4.5.1

### Home Assistant Version

2023.12.2

slipx06, 2023-12-14:
Hi. The card can be used with any inverter if you are able to map the required sensors.
What integration method are you using?

matomatusov, 2023-12-14:
Hi,
I am used integration method modbus RS485.

slipx06, 2023-12-14:
What is not working?

richsimpson, 2023-12-15:
I have it working on Solax, be great if it worked by default to make it simpler. I can share my code if you like? I think I have created a couple of the sensors manually actually. But majority will be there as a starter.

richsimpson, 2023-12-15:
[image]

type: custom:sunsynk-power-flow-card
cardstyle: compact
show_solar: true
show_grid: true
show_battery: true
battery:
  energy: 17600
  shutdown_soc: 11
  show_daily: true
  invert_power: true
  max_power: 3750
  auto_scale: false
  show_absolute: false
  hide_soc: false
solar:
  show_daily: true
  mppts: 1
  max_power: 7000
  display_mode: 2
  animation_speed: 9
  dynamic_colour: false
load:
  show_daily: true
  max_power: 15000
  show_aux: false
  show_nonessential: true
  additional_loads: 2
  load1_name: Water
  load2_name: EV
  load1_icon: mdi:thermometer-water
  load2_icon: mdi:ev-station
  animation_speed: 9
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  animation_speed: 9
  auto_scale: false
  export_colour:
    - 194
    - 6
    - 219
  dynamic_colour: true
  no_grid_colour:
    - 189
    - 188
    - 188
entities:
  inverter_voltage_154: sensor.solax_inverter_voltage
  inverter_current_164: sensor.solax_inverter_current
  inverter_power_175: sensor.solax_inverter_power
  radiator_temp_91: sensor.solax_inverter_temperature
  day_battery_charge_70: sensor.solax_battery_input_energy_today
  day_battery_discharge_71: sensor.solax_battery_output_energy_today
  battery_voltage_183: sensor.solax_battery_voltage_charge
  battery_soc_184: sensor.solax_battery_capacity
  battery_power_190: sensor.solax_battery_power_charge
  battery_current_191: sensor.solax_battery_current_charge
  battery_temp_182: sensor.solax_battery_temperature
  day_grid_import_76: sensor.solax_today_s_import_energy
  day_grid_export_77: sensor.solax_today_s_export_energy
  grid_power_169: sensor.solax_grid_export_import_sum2
  grid_ct_power_172: sensor.solax_grid_export_import_sum2
  day_load_energy_84: sensor.powerflow_today_house_load
  day_pv_energy_108: sensor.solax_today_s_solar_energy
  pv1_power_186: sensor.solax_pv_power_1
  pv1_voltage_109: sensor.solax_pv_voltage_1
  pv1_current_110: sensor.solax_pv_current_1
  remaining_solar: sensor.solcast_pv_forecast_forecast_remaining_today
  essential_load1: sensor.immersion_current_consumption
  essential_load1_extra: sensor.immersion_energy_day
  essential_load2: sensor.ev_power_consumption
  essential_load2_extra: sensor.ev_fast_charge_day
large_font: false
panel_mode: false
inverter:
  auto_scale: false
  modern: false

matomatusov, 2023-12-15:
Hi,
I don't know why the inverter doesn't want to show me the correct status.

[image]

Pho3niX90, 2023-12-15:
> Hi, I don't know why the inverter doesn't want to show me the correct status.
> 
> ![status](https://private-user-images.githubusercontent.com/129114957/290810265-dc7260bf-9022-4c8a-9ea4-185e7950696d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI2NDI2NTcsIm5iZiI6MTcwMjY0MjM1NywicGF0aCI6Ii8xMjkxMTQ5NTcvMjkwODEwMjY1LWRjNzI2MGJmLTkwMjItNGM4YS05ZWE0LTE4NWU3OTUwNjk2ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxNVQxMjEyMzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMGMxNTg1YjQ2YzA2ZDMxZWZhNTBhMTRmYTQ3MDY0NWI3MjU0ZDhmMTRhZmM4NmRjMmM4MGUwNDJjOGNkYmRjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.jRkefrZmL8rgWUjf-pF6NbOKWUEwu9NTT9ifd2SGDf0)

Do you have example data of what that sensor returns? Will probably need a code addition for this, as well as the solax inverter image

richsimpson, 2023-12-15:
There is a Run Mode with these options:

Normal Mode
Idle Mode
Checking

[image]

[image]
[image]

matomatusov, 2023-12-15:
[Pho3niX90](https://github.com/Pho3niX90)

this sensor is returning:

1.  Waiting
2.  Checking
3.  Normal
4.  Off
5.  Permanent Fault
6.  Updatingermanent Fault
7.  Updating

    EPS Check
       EPS Mode
       Self Test
        Idle
        Standby

Pho3niX90, 2023-12-15:
If you know how to change the `js` file without hacs, you can test the PR #179 to make sure everything was considered before the merge.

matomatusov, 2023-12-15:
Hi,
and where can the file be downloaded?

Thanks

Pho3niX90, 2023-12-15:
From here https://github.com/slipx06/sunsynk-power-flow-card/pull/179/files

matomatusov, 2023-12-15:
somehow the Solax doesn't appear there.

[image]

Pho3niX90, 2023-12-15:
Be sure to remove the both the `.js` and `.gz` file from the folder, before adding the new js, then you need to clear your browser cache, or, use incognito mode

matomatusov, 2023-12-15:
I don't know where I'm doing wrong, but it doesn't appear there. Solax is in the file.

[image]

Pho3niX90, 2023-12-15:
> I don't know where I'm doing wrong, but it doesn't appear there. Solax is in the file.
> 
> ![13](https://private-user-images.githubusercontent.com/129114957/290906882-d9c7438a-051d-48c3-87cb-4342c827e31c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI2NjYxNTgsIm5iZiI6MTcwMjY2NTg1OCwicGF0aCI6Ii8xMjkxMTQ5NTcvMjkwOTA2ODgyLWQ5Yzc0MzhhLTA1MWQtNDhjMy04N2NiLTQzNDJjODI3ZTMxYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxNVQxODQ0MThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ODI5YmFiMGZhMmY2MWNhMTc2N2M2YzMxM2Y0ODQzYTM5ODg0Nzk2MjkzNzE3YjY2OTJkYmVlNDUxZmM0YjMxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.7VN8EPUDvevw1hPk1AIVCd9gLopJNPTPUUYYCzkk8BU)

If using chrome, open dev tools (F12), right click on the reload button, and click hard reload. 

This is probably just a cache issue.

slipx06, 2023-12-15:
Thanks @Pho3niX90 tested and everything looks good

closed 2023-12-15 by slipx06

slipx06, 2023-12-15:
@matomatusov once you have tested please check that the provided example is still correct.

https://slipx06.github.io/sunsynk-power-flow-card/examples/solax.html

matomatusov, 2023-12-16:
Hi,
[slipx06](https://github.com/slipx06)

Wrong status is displayed. Inverter is status Idle.
[image]

slipx06, 2023-12-16:
HI. That is the correct mapping

The following states 'waiting', 'checking', 'idle mode', 'standby will all may to Standby.

```
solax: {
        normal: {states: ['normal mode'], color: 'green', message: localize('common.normal')},
        standby: {states: ['waiting', 'checking', 'idle mode', 'standby'], color: 'blue', message: localize('common.standby')},
        selftest: {states: ['self test'], color: 'yellow', message: localize('common.selftest')},
        offgrid: {states: ['off-grid waiting', 'off-grid', 'eps mode'], color: 'green', message: localize('common.offgrid')},
        fault: {states: ['permanent fault mode', 'fault'], color: 'red', message: localize('common.fault')},
        check: {states: ['checking', 'update mode', 'eps check mode'], color: 'orange', message: localize('common.check')},
    }
```

matomatusov, 2023-12-16:
Hi,
I was not expecting that.

Thanks

TomaszBon, 2024-01-10:
Hi, could anyone share the function for calculating the sensor.powerflow_today_house_load for entity 84?

matomatusov, 2024-01-11:
Hi,
SolaX does not have this entity.
It must be assembled from the following entities.
today_import_energy + today_solar_energy + battery_output - today_export_energy - battery_input

richsimpson, 2024-01-11:
> Hi, could anyone share the function for calculating the sensor.powerflow_today_house_load for entity 84?

I set up a sensor for this.
I reference this one in my Sunsynk Power Flow Card Configuration
day_load_energy_84: sensor.powerflow_today_house_load

And that is created as a sensor as follows:

      powerflow_today_house_load:
        friendly_name: "powerflow_today_house_load"
        unit_of_measurement: "kW"
        value_template: "{{ (states('sensor.solax_today_s_import_energy')| float(0) / 1 + states('sensor.solax_today_s_solar_energy')| float(0) / 1 + states('sensor.solax_battery_output_energy_today') | float(0) / 1) - states('sensor.solax_battery_input_energy_today') | float(0) / 1 - states('sensor.solax_today_s_export_energy') | float(0) / 1  | round(1) }}"

Formatted like this:
[image]

PR #179 (2023-12-15): Add 'Solax' to supported inverter types
## Description
This update expands the supported inverter types in the application by adding 'Solax' to the list. This enhancement improves the compatibility of the application with more inverter types and provides broader application usage for users with Solax inverters.

Inverter image:
[image]

## Related issues
fixes #176 

## Type of change

Please delete options that are not relevant.

- [X] New feature (non-breaking change which adds functionality)