slipx06/sunsynk-power-flow-card #178: Grid crossed out - but data working

richsimpson, 2023-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

really like the card, only bit I can't suss out is my grid shows as crossed out always. Its a Solax inerter setup and I have used and created various sensors to enable this

[image]

### Expected behaviour

Assume the grid needs a sensor to say its 'on' but I don't have that, so an option to just make the icon be standard grid connected icon?

### Card Version

4.5.2

### Home Assistant Version

2023.12.2

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: compact
show_solar: true
show_grid: true
show_battery: true
battery:
  energy: 17600
  shutdown_soc: 11
  show_daily: true
  invert_power: true
  max_power: 3800
solar:
  show_daily: true
  mppts: 1
  max_power: 7000
  display_mode: 2
  animation_speed: 9
load:
  show_daily: true
  max_power: 15000
  show_aux: false
  show_nonessential: true
  additional_loads: 2
  load1_name: Water
  load2_name: EV
  load1_icon: mdi:thermometer-water
  load2_icon: mdi:ev-station
  animation_speed: 9
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  animation_speed: 9
  auto_scale: false
  export_colour:
    - 194
    - 6
    - 219
  dynamic_colour: true
entities:
  inverter_voltage_154: sensor.solax_inverter_voltage
  inverter_current_164: sensor.solax_inverter_current
  inverter_power_175: sensor.solax_inverter_power
  radiator_temp_91: sensor.solax_inverter_temperature
  day_battery_charge_70: sensor.solax_battery_input_energy_today
  day_battery_discharge_71: sensor.solax_battery_output_energy_today
  battery_voltage_183: sensor.solax_battery_voltage_charge
  battery_soc_184: sensor.solax_battery_capacity
  battery_power_190: sensor.solax_battery_power_charge
  battery_current_191: sensor.solax_battery_current_charge
  battery_temp_182: sensor.solax_battery_temperature
  day_grid_import_76: sensor.solax_today_s_import_energy
  day_grid_export_77: sensor.solax_today_s_export_energy
  grid_power_169: sensor.solax_grid_export_import_sum2
  grid_ct_power_172: sensor.solax_grid_export_import_sum2
  day_load_energy_84: sensor.powerflow_today_house_load
  day_pv_energy_108: sensor.solax_today_s_solar_energy
  pv1_power_186: sensor.solax_pv_power_1
  pv1_voltage_109: sensor.solax_pv_voltage_1
  pv1_current_110: sensor.solax_pv_current_1
  remaining_solar: sensor.solcast_pv_forecast_forecast_remaining_today
  essential_load1: sensor.immersion_current_consumption
  essential_load1_extra: sensor.immersion_energy_day
  essential_load2: sensor.ev_power_consumption
  essential_load2_extra: sensor.ev_fast_charge_day
large_font: false
panel_mode: false
inverter:
  auto_scale: false
  modern: false
```

Pho3niX90, 2023-12-15:
This was fixed in `4.5.3`

closed 2023-12-15 by richsimpson

toleabivol, 2023-12-29:
With Deye this still happens

Pho3niX90, 2023-12-30:
> With Deye this still happens
> 
> ![image](https://private-user-images.githubusercontent.com/3755796/293361303-12cf4ec7-2374-4090-851c-291c69463512.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM5NzcyOTIsIm5iZiI6MTcwMzk3Njk5MiwicGF0aCI6Ii8zNzU1Nzk2LzI5MzM2MTMwMy0xMmNmNGVjNy0yMzc0LTQwOTAtODUxYy0yOTFjNjk0NjM1MTIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDIzMTIzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMzBUMjI1NjMyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTJjM2RlMjU1MTA2NzNhOTI3ZTJiNDRkMTcyNzI3ODg5MGVhYTIzZGQyNGYwOGM2YzNiZmM2ZWVkM2JjMzVjYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.qxBfLt13qgyts2BEMoPF2CzbqvMkCKhmtC0bwm2yPdI)
> 
> ```yaml
> square: false
> type: grid
> cards:
>   - type: custom:sunsynk-power-flow-card
>     cardstyle: full
>     show_solar: true
>     battery:
>       energy: 0
>       shutdown_soc: 14
>       show_daily: true
>       auto_scale: true
>     solar:
>       show_daily: true
>       mppts: 2
>       pv1_name: House
>       pv2_name: Garage
>       dynamic_colour: true
>       auto_scale: true
>     load:
>       show_daily: true
>       dynamic_icon: true
>       dynamic_colour: true
>       auto_scale: true
>       additional_loads: 2
>       load1_name: go-eCharger
>       show_aux: false
>       load2_name: go-eCharger
>       load1_icon: mdi:ev-station
>       load2_icon: mdi:ev-station
>     grid:
>       show_daily_buy: true
>       show_daily_sell: true
>       show_nonessential: true
>       auto_scale: true
>       export_colour:
>         - 25
>         - 210
>         - 71
>       additional_loads: 0
>     entities:
>       use_timer_248: switch.ss_use_timer
>       priority_load_243: switch.ss_priority_load
>       inverter_voltage_154: sensor.ss_inverter_voltage
>       load_frequency_192: sensor.ss_load_frequency
>       inverter_current_164: sensor.ss_inverter_current
>       inverter_power_175: sensor.ss_inverter_power
>       inverter_status_59: sensor.ss_overall_state
>       day_battery_charge_70: sensor.ss_day_battery_charge
>       day_battery_discharge_71: sensor.ss_day_battery_discharge
>       battery_voltage_183: sensor.ss_battery_voltage
>       battery_soc_184: sensor.ss_battery_soc
>       battery_power_190: sensor.ss_battery_power
>       battery_current_191: sensor.ss_battery_current
>       grid_power_169: sensor.ss_grid_power
>       day_grid_import_76: sensor.ss_day_grid_import
>       day_grid_export_77: sensor.ss_day_grid_export
>       grid_ct_power_172: sensor.ss_grid_ct_power
>       day_load_energy_84: sensor.ss_day_load_energy
>       essential_power: none
>       aux_power_166: sensor.sunsynk_aux_power
>       day_pv_energy_108: sensor.ss_day_pv_energy
>       pv1_power_186: sensor.ss_pv1_power
>       pv2_power_187: sensor.ss_pv2_power
>       pv1_voltage_109: sensor.ss_pv1_voltage
>       pv1_current_110: sensor.ss_pv1_current
>       pv2_current_112: sensor.deye_pv2_current
>       pv2_voltage_111: sensor.deye_pv2_voltage
>       essential_load1_extra: sensor.goecharger_go_echarger_p_all
>       essential_load2_extra: sensor.goecharger_go_echarger_current_session_charged_energy
>       aux_load1_extra: sensor.goecharger_go_echarger_p_all
>       aux_load2_extra: sensor.goecharger_go_echarger_current_session_charged_energy
>       aux_connected_status: sensor.goecharger_go_echarger_car_status
>       energy_cost_buy: '0.32'
>       energy_cost_sell: '0.082'
>       grid_connected_status_194: binary_sensor.ss_grid_connected
>     panel_mode: true
>     show_battery: true
>     show_grid: true
>     large_font: true
>     inverter:
>       three_phase: true
>       auto_scale: false
>       modern: true
>       autarky: 'no'
>       colour:
>         - 120
>         - 120
>         - 120
>       model: sunsynk
> columns: 1
> ```

Which version are you using?

toleabivol, 2023-12-31:
Latest: 4.9.1