slipx06/sunsynk-power-flow-card #184: hybrid mode

amgstone, 2023-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

when the hybrid mode is on, the battery is charged from the sun and the house is supplied from the network, then a separate drawing is needed on the house so that it is clear that the house is not supplied by the sun.

### Expected behaviour

when the hybrid mode is on, the battery is charged from the sun and the house is supplied from the network, then a separate drawing is needed on the house so that it is clear that the house is not supplied by the sun.

### Possible Solutions

when the hybrid mode is on, the battery is charged from the sun and the house is supplied from the network, then a separate drawing is needed on the house so that it is clear that the house is not supplied by the sun.

### Mode

lite

### Context / Reason

[image]

slipx06, 2023-12-19:
Hi. The implementation of the dynamic_icon and dynamic_colour is fairly simplistic. It only considers the essential_load and the battery and solar power. The card does not know what mode the inverter is running i.e. hybrid or if you are supplying loads in front of the inverter (non-essential).  I can perhaps build in another check that looks at the grid power i.e. PV power > essential_load power && battery power is < 5% of essential_power && grid power <5% of essential_power

If I look at your image you are supplying 1.6kW PV + 1.34kW Grid = 2.94kW supply but your load is only 1.1kW (Essential) + 968W (bat) = 2,06kW so you are missing some sensor data.

amgstone, 2023-12-19:
yes, I have 2 batteries, but 1 is connected to HA

slipx06, 2023-12-21:
I've added some additional checks for the next release to the dynamic_icon logic. It wont show the PV icon or the battery icon if the total_grid_power - Math.abs(nonessential) < 20

```
switch (true) {
            case pvPercentage_raw >= 100 && batteryPercentage_raw <= 5 && (total_grid_power - Math.abs(nonessential)) < 20 && config.load.dynamic_icon:
                essIcon = essPv;
                essIconSize = 1;
                break;
            case batteryPercentage_raw >= 100 && pvPercentage_raw <= 5 && (total_grid_power - Math.abs(nonessential)) < 20 && config.load.dynamic_icon:
                essIcon = essBat;
                essIconSize = 0;
                break;
            case pvPercentage_raw === 0 && batteryPercentage_raw === 0 && config.load.dynamic_icon:
                essIcon = essGrid;
                essIconSize = 0;
                break;
            default:
                essIcon = ess;
                essIconSize = 0;
                break;
        }
```

slipx06, 2023-12-21:
The change discussed in #189 should fix this

slipx06, 2023-12-21:
Fixed in https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.7.3

closed 2023-12-21 by slipx06