slipx06/sunsynk-power-flow-card #185: Dynamic colour for no-grid vs no power from grid

NemesisXB, 2023-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When enabling dynamic_colour, when I am not drawing power from the grid, the grid items colour changes to the no_grid_colour. This is as documented and expected.

### Expected behaviour

However, We have rolling blackouts. So multiple times a day where our grid power gets cut off. I want the no_grid_colour to be used when there is no grid (grid disconnected or off). And maybe have a seperate colour (no_grid_usage_colour) for when the grid is there but I am not drawing power from it.

I basically want the following:
No-grid: Red
Not drawing power from grid: green (this is good for me)
drawing from grid: orange (bad for me)

### Possible Solutions

Explained in the expected behaviour

### Mode

Both

### Context / Reason

Explained in the expected behaviour

slipx06, 2023-12-19:
Makes sense. Let me look into adding that option.

closed 2023-12-20 by slipx06

slipx06, 2023-12-20:
These are the options now. If export and no_grid are not set they will default to the import `colour:` set. If grid_off colour is not set it will follow import, export or grid off depending of the state of the grid

| Attribute             | Requirement | Default         | Description                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                |
|-----------------------|-------------|-----------------|------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|
| colour:               | Optional    | `'#5490c2'`     | Sets the colour of all the grid card objects. Hex codes (`'#66ff00'` etc) or names (`red`, `green`, `blue` etc).                                                                                                                                                                                                                                                                                                                                                                                                                                                                                          |
| export_colour:        | Optional    |                 | Sets the colour of all the grid card objects when exporting (selling) energy. Hex codes (`'#66ff00'` etc) or names (`red`, `green`, `blue` etc). If not set will use the `colour:` value defined above.                                                                                                                                                                                                                                                                                                                                                                                                                                                        |
| no_grid_colour:       | Optional    |                 | Sets the colour of all the grid card objects when there is no grid power. Hex codes (`'#66ff00'` etc) or names (`red`, `green`, `blue` etc). If not set will use the `colour:` value defined above total                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                |
| grid_off_colour:       | Optional   |                  | Sets the colour of the grid icon when the grid is disconnected.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  |

commit 316b74c5 (2023-12-20): feat: closes #185 other fixes and updates