slipx06/sunsynk-power-flow-card #186: Visual editor overwrites my custom yaml items

NemesisXB, 2023-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When using the visual editor, it overwrites any custom items I had in the yaml. For example when using the layout card with grid:
view_layout:
  grid-area: xxx

### Steps To Reproduce

Add custom items to yaml. Change settings with visual editor. Custom items disappear.

### Expected behaviour

The custom yaml items should not be removed.

### Card Version

4.6.1

### Home Assistant Version

2023.12

slipx06, 2023-12-19:
Hi. Yes unfortunately this is a know issue. For now you need to switch to the code editor and copy them before making changes using the visual editor. Once done you can paste them back.

Pho3niX90, 2023-12-20:
I have an idea what might be causing this. 

The editor is using the cards' specific config type. When a calue changes, it emits it in this specific type. We could apply a spread operator after the fact to make sure any missing attributes are retained. 

I will quickly have a look today.

slipx06, 2023-12-20:
Thank you.  As an example additional card attributes that are not included in the config that are overwritten/removed when using the visual editor are

```
view_layout:
  grid-area: flow
```
and

```
card_mod:
  style: |
    ha-card {
      --ha-card-background: transparent;
      box-shadow: none;
      border: 0px;
    }
```

Pho3niX90, 2023-12-20:
I added the spread operator. From my limited tests, it seems to do the job, however, I do not have any card mods so cannot really test if the functionality is present.

slipx06, 2023-12-20:
My commit did not fix this it results in the following error
[image]

Pho3niX90, 2023-12-20:
Will quickly have a look

Pho3niX90, 2023-12-20:
So #188 alone gives me the following. What other mods can I test?
https://streamable.com/sr872m

slipx06, 2023-12-20:
I found the issue. I had accidentally commented out some code. Pushing a fix now

slipx06, 2023-12-20:
can you test 4.7.2

NemesisXB, 2023-12-20:
> can you test 4.7.2

Seems to working for me.

PR #188 (2023-12-20): fix: #186
fixes: #186
commit fa00df88 (2023-12-20): Merge pull request #188 from Pho3niX90/bugfix/issue-186

fix: #186
closed 2023-12-20 by slipx06