slipx06/sunsynk-power-flow-card #189: entrance to the house

amgstone, 2023-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

when the battery is charging from the sun, it also shows the battery charging in the house, it is possible to somehow add settings so that the battery can be turned off so that it does not fall into the house as energy that goes into the house

### Expected behaviour

when the battery is charging from the sun, it also shows the battery charging in the house, it is possible to somehow add settings so that the battery can be turned off so that it does not fall into the house as energy that goes into the house

### Mode

lite

### Context / Reason

when the battery is charging from the sun, it also shows the battery charging in the house, it is possible to somehow add settings so that the battery can be turned off so that it does not fall into the house as energy that goes into the house
[image]

slipx06, 2023-12-21:
Hi. Yes I can fix that. The current logic assumes prioritized load, i.e. solar goes to load first, then battery. I've added a conditional check (see below) for the next release that evaluates the entity `priority_load_243: switch.toggle_priority_load` and adjusts available PV for the load. You can create a template switch that returns `on` and `off` and since you prioritize battery set it to false. 

```
       const pvPercentage_raw = total_pv === 0 
            ? 0 
            : priority === 'off' 
                ? battery_power > 0  //the battery is discharging
                    ? (total_pv / essential) * 100
                    : ((total_pv -  Math.abs(battery_power)) / essential) * 100 //pv goes to battery first
                : (total_pv / essential) * 100; //default assumes priority load
```

slipx06, 2023-12-21:
Fixed in https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.7.3

closed 2023-12-21 by slipx06