slipx06/sunsynk-power-flow-card #190: battery

amgstone, 2023-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I have an easun inverter, when the battery is not charged, the consumption from the battery reaches 100 watts. and at night it shows a mixed mode that consumption comes from both the battery and the network. perhaps it is worth doing as with solar panels, if the consumption is up to 100 watts, then show 0 consumption.

### Expected behaviour

I have an easun inverter, when the battery is not charged, the consumption from the battery reaches 100 watts. and at night it shows a mixed mode that consumption comes from both the battery and the network. perhaps it is worth doing as with solar panels, if the consumption is up to 100 watts, then show 0 consumption.
[image]

### Mode

lite

### Context / Reason

I have an easun inverter, when the battery is not charged, the consumption from the battery reaches 100 watts. and at night it shows a mixed mode that consumption comes from both the battery and the network. perhaps it is worth doing as with solar panels, if the consumption is up to 100 watts, then show 0 consumption.

slipx06, 2023-12-21:
I can look at increasing the thresholds for both the dynamic_colour and dynamic_icon. 
For dynamic colour I can ignore anything that is not > 2% of load
Similarly when both PV and bat are less than 5% of load assume supply from grid and show the correct icon. This is currently triggered when both  are 0%
It should address your issue. Will have to test it

amgstone, 2023-12-22:
super

commit 576d21a1 (2023-12-22): fix: closes #190 adjust dynamic_colour thresholds
closed 2023-12-22 by slipx06