slipx06/sunsynk-power-flow-card #194: Entity to Grid - Export Control User Limit

matomatusov, 2023-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

addition to the Grid

### Expected behaviour

Hi,
would it be possible to add an entity to the Grid section?
it would be about Export Control User Limit (W). 

[image]

Thanks :)

### Mode

lite

### Context / Reason

It would be good information about setting up export to the Grid.

slipx06, 2023-12-27:
The equivalent Synsynk setting is "max sell power". Will need to see if there is space for this on the card

jzielke84, 2024-02-05:
@slipx06 Just to be sure before I open a new issue: Is this also required if I'm injecting power back into the grid? This is my card configuration:

```
yaml
type: custom:sunsynk-power-flow-card
cardstyle: compact
show_solar: true
battery:
  energy: 0
  shutdown_soc: 20
  show_daily: false
solar:
  show_daily: false
  mppts: 1
  max_power: 800
  dynamic_colour: true
  pv1_name: Solix
  auto_scale: true
load:
  show_daily: false
  auto_scale: true
  dynamic_colour: true
  dynamic_icon: true
grid:
  show_daily_buy: false
  show_daily_sell: false
  energy_cost_decimals: 0
  auto_scale: true
entities:
  pv1_power_186: sensor.solix_power
  essential_power: sensor.energie_eigenverbruch
  grid_voltage: sensor.bitshake_smartmeterreader_sml_volt_p2
  grid_ct_power_172: sensor.netz_nettobezug
  solar_sell_247: sensor.solix_netzeinspeisung
large_font: true
panel_mode: true
show_grid: true
show_battery: false
decimal_places: 2
inverter:
  three_phase: false
  auto_scale: true
title: Stromverbrauch
title_size: '1'
```

The calculations are fine but the grid should also show a value for injecting back into the grid. Like a dot flowing from right to left. This is my current output:

[image]

slipx06, 2024-02-05:
Hi. No its not required. The card will take care of that if your grid sensor i.e. `grid_ct_power_172: sensor.netz_nettobezug` reports positive power for import and negative power for export.

jzielke84, 2024-02-05:
> Hi. No its not required. The card will take care of that if your grid sensor i.e. `grid_ct_power_172: sensor.netz_nettobezug` reports positive power for import and negative power for export.

Ah ok, thank you for clarifying this one.

closed 2024-04-06 by slipx06

slipx06, 2024-04-06:
@matomatusov finally got around to adding this feature request to https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.29.0

matomatusov, 2024-04-06:
HI, 
I updated and I can't find max_sell_power in the entites.
Is it in the endites?

slipx06, 2024-04-06:
Sorry I have not added it to the visual editor yet. Will push a fix. For now you can add it under the entities: using the code editor
``
entities:
   max_sell_power: number.sunsynk_max_sell_power
```

matomatusov, 2024-04-06:
I have it there thanks
[image]

slipx06, 2024-04-06:
Its added to https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.29.1

slipx06, 2024-04-06:
Fixed in 4.29.1

[image]

matomatusov, 2024-04-06:
I confirm it's in the visual editor (max_sell_power). 
The export is also correct.

commit 6f8e159a (2024-04-06): feat: add "max sell power"/ Export Control User Limit" entity. closes #194