slipx06/sunsynk-power-flow-card #195: Support multiple inverter in a setup

elearningdienst, 2023-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I see one of my inverter in the card

### Expected behaviour

I want to see my second inverter in the same card

### Possible Solutions

In multi inverter settings it should be possible to add additionals entities from another inverter.  

### Mode

lite

### Context / Reason

As i have a huge roof, I have two goodwe inverters. I can't see both inverter in one card.

Pho3niX90, 2023-12-28:
> Hi. Currently the card only supports a single inverter. For now you can create a second card for you second inverter or create template sensors and use those.

Perhaps, the sensors for the inverters could accept an array for inverters to accomplish this. 

Example:
```yaml
inverter_current_164:
 - sensor.inverter1_current
 - sensor.inverter2_current
 ```
 
 And then just sum/avg them where applicable. 2 cards might give duplicate values for PV and battery

slipx06, 2023-12-28:
Sounds like a good solution but I have no idea how multiple inverters report sensor values i.e. sum, separate etc. @elearningdienst are you running your inverters in parallel? How is the data presented on the master?

codifierr, 2024-01-10:
@slipx06 I also have similar requirement. I have two Deye inverters of 6KW. One master, one slave and they report separate values therefore if we have options to sum the values then the card can show the total.
Or if we get feature of multiple inverter then it will be great.

slipx06, 2024-01-11:
Hi. Unfortunately I don't think it would be possible to show two inverters on one card (space issues) so my suggestion is that you create separate cards for the primary and secondary inverters. With the additional inverter there are also more MPPTS so creating a combined card could still present some challenges. Has anyone attempted to create template sensors that sum your values and use this as input into a single view card?

Dubists, 2024-01-14:
I use HA helpers to combine multiple inverter and battery values. t
These helpers are than used as inputs for the power flow card and also for statistic purposes.

Roving-Ronin, 2024-02-05:
@slipx06 Yes this is exactly what we are doing with the Huawei inverters and WLCRS integration (that provides access to Huawei in HA).

If you can assist with creating g that Huawei example doco page I'll post up the yaml for this with Huawei. 
In short though you'll find inverters work in a cascade with the primary inverter managing the overall installation (as the power / consumption meter that is then use to allow device doing things like controlling maximum power kW bro g exported to the grid is attacked to the primary inverter). However within HA the inverters / batteries end up being seen as seperate devices with their own input / active power / daily kWh generated today etc sensors. That is where we then use the 'derived sensors' to create new sensors that provide things like the input power of all the PV strings / Active Power of all inverters / Grid imported or exported today etc (using energy_meter instead of utility_meter, as it also calculates costs / compensation for those kWh used/exported).

https://github.com/slipx06/sunsynk-power-flow-card/issues/135

slipx06, 2024-03-17:
I'm moving this to a discussion. Until there is a wide version of the card you need to follow the suggestion above and create template entities