slipx06/sunsynk-power-flow-card #196: Sunsynk settings for when using the Aux port as an output

johnmsole, 2023-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Currently I can add a options to change the Aux port from Generator to Aux (excess solar) output but I cant then change the associated parameters

### Expected behaviour

Pl add the additional features to access the other Aux port settings>
-Solar Power,
-Aux Load OFF Batt and
-Aux Load ON Batt

### Mode

full

### Context / Reason

When adjusting the settings from Gen input to Aux output (Excess solar) I find it necessary to often change the settings based on various external factors

slipx06, 2023-12-27:
Hi. I'm not sure I understand this request. Are you looking for the ability to be able to toggle these settings from the card?
If so can you share the sensor/switch that you have that lets you do this?

johnmsole, 2023-12-27:
Firstly thank you for a phenomenal flow card.

Pl cancel the request as this is not a power flow card related but more the
modbus integration which allows setting features

J

closed 2023-12-27 by slipx06