slipx06/sunsynk-power-flow-card #208: Not showing 3rd additional non-essential load

duanemck, 2024-01-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When configuring `grid.additional_loads` to `3`, only the first 2 loads are displayed. I have configured a name and icons for the 3rd load and I have set `use_timer_248` to `false`.

[image]

### Steps To Reproduce

1. In `grid` config:
```
              additional_loads: 3
              load1_name: Geyser
              load2_name: Pool
              load3_name: Oven
              load1_icon: boiler
              load2_icon: pump
              load3_icon: oven
```
2. In `entities` config:
```
entities:
              use_timer_248: false
              priority_load_243: false
              ...
              non_essential_load1: sensor.geyser_real_time_usage
              non_essential_load2: sensor.pool_real_time_usage
              non_essential_load3: sensor.oven_real_time_usage
```

### Expected behaviour

See all 3 additional loads (geyser/pool/oven)

### Card Version

4.9.1

### Home Assistant Version

2024.1.1

### Configuration

```shell
Relevant snippets under "Steps to Reproduce"
```

slipx06, 2024-01-08:
Hi. Thanks I will update the documentation. Please add

```
battery:
  hide_soc: true 
```

slipx06, 2024-01-10:
Hi. The load3 icon only accepts an mdi icon and not the presets

closed 2024-01-11 by slipx06