slipx06/sunsynk-power-flow-card #212: Time Of Use Tabel

Mights001, 2024-01-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hi, can i activated somewhere this Datapoints in the Card? 

[image]

### Expected behaviour

.

### Card Version

4.10.1

### Home Assistant Version

2024.1.3

slipx06, 2024-01-17:
Its not displayed on the card, but you can create a separate card. I have two examples you can refer to here https://github.com/slipx06/Sunsynk-Home-Assistant-Dash

Example 1
[image]

Example 2
[image]

Mights001, 2024-01-17:
Thank you very much

closed 2024-01-17 by Mights001