slipx06/sunsynk-power-flow-card #217: Solar Left Today not displayed

pipip, 2024-01-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

No matter on the mode (lite, compact, full) the value for solar "left today" is not displayed

[image]

### Expected behaviour

solar "left today" should be displayed in any mode

### Card Version

1.19.1

### Home Assistant Version

2024.1.3

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: compact
show_solar: true
show_battery: false
inverter:
  model: RCT
solar:
  show_daily: true
  mppts: 1
  pv1_name: Roof
  max_power: 4000
load:
  show_daily: true
  additional_loads: 2
  load1_name: House
  load1_icon: mdi:home
  load2_name: Heatpump
  load2_icon: mdi:heat-pump-outline
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
entities:
  inverter_voltage_154: sensor.stromzaehler_haus_channel_a_voltage
  load_frequency_192: sensor.rct_power_storage_grid_frequency
  inverter_status_59: sensor.rct_power_storage_faults
  inverter_power_175: sensor.rct_power_storage_inverter_ac_power
  grid_power_169: sensor.stromzaehler_sml_power_cur
  day_grid_import_76: sensor.stromzaehler_sml_in_daily
  day_grid_export_77: sensor.stromzaehler_sml_out_daily
  grid_ct_power_172: sensor.stromzaehler_sml_power_cur
  day_load_energy_84: sensor.stromzaehler_overall_energy_daily
  essential_power: sensor.stromzaehler_overall_power
  essential_load1: sensor.stromzaehler_house_power
  essential_load2: sensor.stromzaehler_heatpump_power
  essential_load1_extra: sensor.stromzaehler_house_energy_daily
  essential_load2_extra: sensor.stromzaehler_heatpump_energy_daily
  day_pv_energy_108: sensor.rct_inverter_energy_production_daily
  pv_total: sensor.rct_power_storage_inverter_ac_power
  pv1_power_186: sensor.rct_power_storage_all_generators_power
  pv1_voltage_109: sensor.rct_power_storage_generator_a_voltage
  radiator_temp_91: sensor.rct_power_storage_undefinedtype_singleton_2
  dc_transformer_temp_90: sensor.rct_power_storage_undefinedtype_singleton
  remaining_solar: sensor.solcast_pv_forecast_forecast_today
  energy_cost_buy: input_number.grid_energy_cost_buy
  energy_cost_sell: input_number.grid_energy_cost_sell
  environment_temp: sensor.wp_aussen_temperatur
large_font: true
panel_mode: true
```

slipx06, 2024-01-21:
By default only the daily total is shown. You need to add `display_mode: 2` to your solar section. 

```
solar:
  show_daily: true
  mppts: 1
  pv1_name: Roof
  max_power: 4000
  display_mode: 2
```

pipip, 2024-01-21:
thanks, sorry I missed this option.

closed 2024-01-21 by pipip