slipx06/sunsynk-power-flow-card #218: esential load  - 4 loads dont work. max 2

cipriancu, 2024-01-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

on card style = compact and lite essential load support 4 individual loads 
on card style = Full essential load support just 2 individual loads
on card style = compact and lite non-essential loads are not visible 
 

### Steps To Reproduce

on card style = compact and lite essential load support 4 individual loads 
on card style = Full essential load support just 2 individual loads
on card style = compact and lite non-essential loads are not visible 

### Expected behaviour

on card style = compact and lite essential load support 4 individual loads 
on card style = Full essential load support just 2 individual loads
on card style = compact and lite non-essential loads are not visible 

### Card Version

4.10.2

### Home Assistant Version

Core 2024.1.5 Supervisor 2023.12.1 Operating System 11.4 Frontend 20240104.0

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
battery:
  energy: 0
  shutdown_soc: 20
  show_daily: true
  show_remaining_energy: true
  invert_power: true
  hide_soc: false
  show_absolute: false
  auto_scale: false
  max_power: 2500
solar:
  show_daily: true
  mppts: 2
  max_power: 6000
  auto_scale: true
  dynamic_colour: true
  colour:
    - 106
    - 210
    - 137
load:
  show_daily: true
  essential_name: ESENTIAL
  additional_loads: 4
  load1_name: Living
  load2_name: Net
  load3_name: Parter
  load4_name: Etaj
  load1_icon: mdi:television
  load2_icon: mdi:ip-outline
  load3_icon: mdi:lightbulb-on-30
  load4_icon: mdi:lightbulb-on-90
  animation_speed: 2
  auto_scale: false
  dynamic_colour: true
  dynamic_icon: true
  show_aux: false
  aux_name: Rest
  show_daily_aux: false
grid:
  show_daily_buy: true
  show_daily_sell: false
  show_nonessential: true
  nonessential_name: Non esential
  nonessential_icon: mdi:account
  load1_name: '1'
entities:
  use_timer_248: switch.sunsynk_toggle_system_timer
  priority_load_243: switch.sunsynk_toggle_priority_load
  inverter_voltage_154: sensor.inv_hv2310015703_grid_ab_voltage
  load_frequency_192: sensor.inv_hv2310015703_grid_frequency
  inverter_current_164: sensor.inv_hv2310015703_phase_a_current
  inverter_power_175: sensor.inv_hv2310015703_hv2310015703_realtime_power
  grid_connected_status_194: binary_sensor.sunsynk_grid_connected_status
  inverter_status_59: sensor.sunsynk_overall_state
  day_battery_charge_70: sensor.battery_1_charging_capacity
  day_battery_discharge_71: sensor.battery_1_discharging_capacity
  battery_voltage_183: sensor.battery_1_battery_voltage
  battery_soc_184: sensor.battery_1_battery_state_of_charge_soc
  battery_power_190: sensor.battery_1_charge_discharge_power
  battery_current_191: sensor.sunsynk_battery_current
  grid_power_169: sensor.sunsynk_grid_power
  day_grid_import_76: sensor.sunsynk_day_grid_import
  day_grid_export_77: sensor.sunsynk_day_grid_export
  grid_ct_power_172: sensor.sunsynk_grid_ct_power
  day_load_energy_84: sensor.sunsynk_day_load_energy
  essential_power: none
  nonessential_power: none
  aux_power_166: sensor.sunsynk_aux_power
  day_pv_energy_108: sensor.inv_hv2310015703_yield_today
  pv1_power_186: sensor.putere_panouri_solare_pv1
  pv2_power_187: sensor.putere_panouri_solare_pv2
  pv1_voltage_109: sensor.inv_hv2310015703_pv1_input_voltage
  pv1_current_110: sensor.inv_hv2310015703_pv1_input_current
  pv2_voltage_111: sensor.inv_hv2310015703_pv2_input_voltage
  pv2_current_112: sensor.inv_hv2310015703_pv2_input_current
  battery_rated_capacity: sensor.battery_1_battery_state_of_health_soh
  battery_status: sensor.battery_1_battery_state_of_health_soh
  essential_load1: sensor.tasmota_spm_activepower_0
  essential_load2: sensor.tasmota_spm_activepower_1
  essential_load3: sensor.tasmota_spm_activepower_2
  essential_load4: sensor.tasmota_spm_activepower_3
  pv_total: sensor.putere_totala_mppt
  radiator_temp_91: sensor.inv_hv2310015703_inverter_internal_temperature
  inverter_current_L2: sensor.inv_hv2310015703_phase_b_current
  inverter_current_L3: sensor.inv_hv2310015703_phase_c_current
  inverter_voltage_L2: sensor.inv_hv2310015703_grid_bc_voltage
  inverter_voltage_L3: sensor.inv_hv2310015703_grid_ca_voltage
  total_pv_generation: sensor.putere_totala_panouri
inverter:
  three_phase: true
  auto_scale: true
  modern: false
  autarky: energy
  model: huawei
show_battery: true
large_font: true
panel_mode: true
show_grid: true
```

slipx06, 2024-01-22:
Hi. I'm assuming you are asking to display 4 essential loads on the full card? 
The current 2 load limit is purely a design decision due to space constraints. Let me see what I can do

cipriancu, 2024-01-22:
Yup. either 4 cards of essential on full view or non-essential visible in lite or compact

slipx06, 2024-01-22:
Non-essential is only available on the full card.

Roving-Ronin, 2024-02-05:
@cipriancu 

Just wondering what you using to get Huawei data? Looking at the entity sensors you have (or don't have) might I suggest taking a look at the WLCRS integration for Huawei Solar devices.

Also have a look at this thread https://github.com/slipx06/sunsynk-power-flow-card/issues/135 where we have been working on adding support for Huawei inverters (and batteries etc) and mapping out all the entities to use and (just wait for for @slipx06 to add Doco example page for Huawei ... I can't work out how to...) so we can paste L1 / M1 inverters yaml to paste into a card and use. And also the yaml files for adding the extra sensors needed for this card (and/or if you have multiple inverters).

cipriancu, 2024-02-05:
Roving 
Thank you for reaching out and sharing information about the WLCRS integration for Huawei Solar devices. Currently, I am using the Fusion Solar integration (https://github.com/tijsverkoyen/HomeAssistant-FusionSolar).
I am indeed interested in migrating to Huawei Solar and have recently acquired a Modbus Gateway. However, this transition will require some time for setup and testing. I appreciate your suggestion and will certainly explore the WLCRS integration further.
C

cipriancu, 2024-02-06:
> @cipriancu
> What's your Huawei setup, number and type of inverters and battery ?
>
> I'm guessing M1 3 phase and a 5kWH LUNA ESS Battery ?
> Do you also have DTSU-666H power meter and sDongle (USB Fast Ethernet / WiFi)?
>
> What gateway are you referring too? If it's an M1 inverter they come with a bundled sDongle that, apart from connecting it to FusionSolar portal also allow connecting it using the WLCRS integration.
>
> BTW the FusionSolar integration is rather 'limited', only pulling data iirc very 5min and dependant upon the cloud, versus using Modbus local polling updating every 30 seconds (& giving HA services to control exports/battery charges).
>
> If you want to send me info and also the other sensors, what they do and what you're trying to achieve, I can see about giving a hand.

Roving 

 I have a SUN2000-6KTL-M1 inverter with a 5kWH LUNA ESS BatteryI have the power meter but I'm not using it yet  ( For the next 4 months I will be off the grid until the power company comes and connects me) the model is indeed DTSU-666H. To be able to work off-grid I also have a backup box which is feeding one essential phase 
To measure the power consumption I use sonoff spm-main + modules relay https://sonoff.tech/product/diy-smart-switches/spm-main-spm-4relay/ with tasmota flashed on it 
For the essential phase, I use 4 relays (2 for lights, one for internet and the last for other stuff) On the main load, I will have another 12 relays 
Related dongle, I have a SDongleA-05 USB to ethernet and wifi 
My plan to add in home assistant the inverter + the heat pump + ventilation units I bought  this Modbus gateway: https://www.cdebyte.com/products/NB144S
I turned it on when it arrived but that was it. Need some time to configure it 
 There is one more issue, the Raspberry Pi that hosts the homeassstant due to power fluctuation have the USB ports broken

C

Roving-Ronin, 2024-02-06:
Whilst I run 6 different RPi's I've found once you start getting a bunch of sensors and wanting to get into leveraging HA, your better off with something like a Lenovo M700 Tiny PC (16GB RAM and toss in a 1TB SSD), used no SSD for -USD$100-120 off eBay. Chalk and cheese difference in performance between the two.

Re that gateway appliance it looks like it only has 1 RS485 port, so you'll need to make sure the heat pump and ventilation units ( air conditioners ?) are capable of supporting RS485 cascading / daisy chaining (with COM IDs assigned to each device).
Even with the getaway your still best to use the sDongle for the Huawei, as with Modbus you can only have one master controller. If you use the sDongle it will allow putting the Huawei inverter(s), LUNA battery, DTSU-666H and optimisers (of your choice have any) all online and connecting it to FusionSolar (for Huawei reporting, firmware updates, remote support and warranty) as well as connecting to HA using WLCRS. If you try use the other gateway it will not be able to do any of that, only connect to HA via WLCRS.

Does the Sonoff support 3 phase? Trying to work it out and looks like a single point of failure, with all wiring and loads (limited to 4400W per module... issue for the heat pump / aircon?) for the house going through it versus solutions that use CT clamps to monitor each circuit (like IoTaWatt or the less capable Shelly EM3).

Given you can't fit every sensor/circuit on the Sunsynk card, it would be worthwhile starting to think what sensors/circuits you can combine into groups. ie All Lights / All GPO (PowerPoints) / All AirCons + Heat Pump etc. Using HA these can all be summed up so you can stick the AirCon / Heat Pump in Non-Essential part of the card (can get 2 entities showing there), stick the Lights and GPO in Essential section (can get up to 4 entities in there) and then use AUX devices section (if your not running a backup generator you need to monitor) to put another two entities in the card (personally my biggest essential kWh users are home network gear, switches/routers/cctv/wifi/cooling fans, and then the servers/nas so I use AUX device section to see the Active Power for these groups, then the total active power for both groups and then the daily energy (kWh) for them.

BTW can I ask your country location, 4 months seems a long time to get a grid connection?

cipriancu, 2024-02-06:
Yup I think în the end im going to give up to raspberry pi and I'm switch to an intel NUC that I have it. The gateway have 4 rs485 or 4 rs232 depends of how you configure it and I hope that each can act as master. Related sonoff is not 3 phases is just single phase but that's OK . Only the hob and the charger for EV will use 3 phase the rest of the stuff are single phase. Each relay of sonoff module can take 20 amps so each circuit can handle 4.4kw  but as I told you I will have 16 circuits 4 on essential and 12 on non essential. Non essentials are split per 2x3 phases differential circuit breakers . So will be 2 circuits on one phase.  I'm tempted to get some sort of measuremet tool to install it at the output of differentials Or there's the option to create a numeric sensor that will be a sum of individual sonoffs  
I'm in Romania.  I had some issues with paperwork and although I have the solar panels since November I managed to place the installation order for the grid last week. And when I placed the order they told me 3-4 months. And that will be just to consume power from the grid. Once I have that I have to place another request to be able to inject power into the grid. 
I haven't moved yet to the new home that's why I'm not having as much time as I want to do all the configurations. Currently I'm focusing on installing lamps sockets and breakers.  C