slipx06/sunsynk-power-flow-card #227: Entities not available on card config

bensebborn, 2024-02-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I’m trying to add an extra load onto the card (my Zappi EV charger)

as we’re using 3 phase I only have 3 separate CT power entities, so I’ve created my own sensor which combines the three values

this shows up fine on my dashboard and I can see the value in the debugging tools

however when going to the card config, it doesn’t show up in the drop-down to select from. 

Is there a certain config I need for the custom sensor template for it to show up?

### Steps To Reproduce

[Platform: sensor
 Sensors: 
:
        friendly_name: Total Zappi Power Usage
        unit_of_measurement: W
        device_class: power
        icon: mdi:flash
        state_class: measurement
        value_template: >
          {{ [ states('sensor.myenergi_zappi_19556152_internal_load_ct1')|float(0) , states('sensor.myenergi_zappi_19556152_internal_load_ct2')|float(0) , states('sensor.myenergi_zappi_19556152_internal_load_ct3')|float(0)] | sum }}
]

### Expected behaviour

Custom entity above shows in the card config to map to load entity

### Card Version

4.10.5

### Home Assistant Version

2024.1.6

Roving-Ronin, 2024-02-05:
his isn't a bug, please read the documentation on defining the entities.

For example on the non essential loads just use one of them for the EV charger sensor you have.

non_essential_load2: sensor.total_zappi_power_usage

Id also suggest adding a unique ID to your sensor. ie

unique_id: total_zappi_power_usage

slipx06, 2024-02-05:
Hi. Which entity are you trying to map this to? I cant see why it would not show up. The only filtering on the drop down list is the device class but you can still just paste in your template sensor or switch to the code editor and add it

closed 2024-02-20 by slipx06