slipx06/sunsynk-power-flow-card #230: Amend the "Time to 100% charge" Logic

Sboshoff76, 2024-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

It takes the current charge rate and does a simple math equation to work out how much time it'll take to get to 100% SOC.

### Expected behaviour

What makes this invalid is that as the batteries get closer to 100% SOC the BMS limits the amount of amps, ergo power going into the battery.

### Possible Solutions

For my lux SNA5000 inverter I have a max charge current sensor in HA. I have 4 batteries so my value fluctuates as the SOC changes as follows:

200A  0%-79%
160A  80%-89%
  70A  90%-94%
  35A  95%-99%
   0A  100%

Could the time to full charge equation logic be programmed to NOT simply use the current charging rate but to rather to use the current charging rate up to the 80% mark and then add that to the max charging rate's time for each of the remaining 3 charging profiles (160A, 70A, 30A in my example). Hope I'm explaining this clearly enough.

I understand this could be a rather complicated math equation to setup and will have to be done differently for each inverter but this will give a much more accurate indication of when the batteries would be full.

### Mode

lite

### Context / Reason

More accurate full charge ETA prediction.

slipx06, 2024-02-09:
Hi
The formula to charge to 100% is shown below.  `battery_power` is related to charging current (P= V x A). Maybe I'm not understanding the request but as you decrease the charging current you will decrease the power going into the battery and this will increase total seconds to charge

` total seconds = ((((100 - parseInt(state_battery_soc.state)) / 100) * battery_energy) / battery_power) * 60 * 60 * -1;`

Sboshoff76, 2024-02-09:
Yeah thanks for the formula. So that would be 100% spot on if the battery was able to charge up to 100% using the current power going into the battery. Thing is the BMS doesn't allow full power all the way to 100% and starts limiting the power going into the battery when it reaches certain thresholds (like my table above). So to get a more accurate figure you need to allow for this limiting action.

I'll try and come up with a formula of my own use case to give you an example.

slipx06, 2024-02-09:
An example would be good because I would assume that the sensor that measures battery_power would take this reduction into consideration but it sounds like the value your inverter is reporting is not accurate?

wacalitz, 2024-02-10:
I'm also trying to figure out why you would say using 100% won't give an accurate indication.   

Do you not perhaps mean you just want to rather see the charge time remaining to those BMS step downs before you see the calc to 100%?  I have something similar setup in a markdown card and calculations in my config but is based on my timer SOC's so if the current SOC is below the current timer target SOC it will give me the time to reach the SOC target and once reached it will give me the time to 100%.  As the card calculation is based on 100% this way I can see how long until I reach my SOC target and how long until I reach 100%.  

Is this not perhaps what you are trying to achive?

Overall the time to 100% should effectivly be the same regarless of the BMS stepdowns.

Pho3niX90, 2024-02-19:
Whilst inverters do have a different charge ratio by SOC, the BMS is the one ultimately in control of the real charging. To account for every single inverter and BMS combo, is just impossible. The reward vs overhead might just not be worth it

Pho3niX90, 2024-02-19:
I did some quick equations with your specific provided charge currents for the brackets. The variance would be 20mins (77m vs 57m) for a 10kWh battery, and 10min for 5kWh.