slipx06/sunsynk-power-flow-card #231: Conditional icons and colors

wendan20, 2024-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

At the moment the essential load icon can dynamicly change color to the source of the power. Also the color of the grid part can change color between asking and returning power to the grid. I use the same colors as in the energy dashboard in HA.

### Expected behaviour

I would have hoped that the same conditional formating would also be available for the non essential icon. So I can see if for example the grid or the solar is powering the non essentials. Also the colors of the battery could change when on charging or discharging the battery.

### Possible Solutions

My request is:
1. Make it possible to dynamically change the icon of the non essential load in the same way as the essential load icon.
2. Make it possible to dynamically change the color of the battery part of the card for charging and discharging.

### Mode

lite

### Context / Reason

By adding these additions the card would even be better readable and also for my wife.... ;-)

Thanks for all the great work!

slipx06, 2024-02-12:
Hi. 
I will look at adding an option to specify separate colours for battery charge and discharge.
Adding a dynamic icon for non-essentials is not simple as some inverters are not hybrid and can only power these from the grid. In addition there are setting that prioritize load vs battery etc and things start getting complicated.

slipx06, 2024-02-14:
I've add the ability to display separate colours for charging and discharging the battery in https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.14.0

closed 2024-02-14 by slipx06

wendan20, 2024-02-14:
> I've add the ability to display separate colours for charging and discharging the battery in https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v4.14.0

Very happy with that and also very fast. Thank you very much!

wendan20, 2024-02-14:
> Hi. I will look at adding an option to specify separate colours for battery charge and discharge. Adding a dynamic icon for non-essentials is not simple as some inverters are not hybrid and can only [...]

Irt to the non essential load. Could it be an option to give people with or without a hybrid inverter a choice (switch?) so they can choose to make use of the dynamic coloring if they have a hybrid? I cannot oversee the implications, but just trying to think with the project. Allready very happy with the colors for the battery.

mirams, 2024-02-14:
Would be nice if the battery did the same as the essential load in terms of turning the colour of what power source is charging it (grid/solar).

slipx06, 2024-02-15:
> Would be nice if the battery did the same as the essential load in terms of turning the colour of what power source is charging it (grid/solar).
Good idea. Busy working on it
[image]